Koenkk/zigbee2mqtt #9219: ERROR: Template variable warning: dict object has no element Undefined when rendering

KitHubek, 2021-10-18:
from closed: https://github.com/Koenkk/zigbee2mqtt/issues/6987

As topic. Log below:

Logger: homeassistant.helpers.template
Source: helpers/template.py:1612
First occurred: 09:49:27 (550 occurrences)
Last logged: 23:15:27

Template variable warning: dict object has no element Undefined when rendering '{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}'

Drafteed, 2021-11-10:
@Koenkk after [this change](https://github.com/Koenkk/zigbee2mqtt/commit/ca7bb0cd4d734c4a9c2f9e80f5f00eb3820c0cee), HA started spamming an error into the logs about HomeKit.

```
First occurred: 20:31:22 (4 occurrences)
Last logged: 20:31:22

Error while processing state change for climate.siterwell_trv
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 269, in _async_state_change_dispatcher
    hass.async_run_hass_job(job, event)
  File "/usr/src/homeassistant/homeassistant/core.py", line 433, in async_run_hass_job
    hassjob.target(*args)
  File "/usr/src/homeassistant/homeassistant/components/homekit/accessories.py", line 389, in async_update_event_state_callback
    self.async_update_state_callback(event.data.get("new_state"))
  File "/usr/src/homeassistant/homeassistant/components/homekit/accessories.py", line 411, in async_update_state_callback
    self.async_update_state(new_state)
  File "/usr/src/homeassistant/homeassistant/components/homekit/type_thermostats.py", line 444, in async_update_state
    self._async_update_state(new_state)
  File "/usr/src/homeassistant/homeassistant/components/homekit/type_thermostats.py", line 466, in _async_update_state
    homekit_hvac_action = HC_HASS_TO_HOMEKIT_ACTION[hvac_action]
KeyError: 'None' 
```

Koenkk, 2021-11-10:
@Drafteed looks like a HA bug to me, HA should be able to handle `null` values properl.

Drafteed, 2021-11-10:
@Koenkk The issue was created there earlier. It would be nice if you left feedback there.
https://github.com/home-assistant/core/issues/59330

closed 2021-12-19 by github-actions[bot]