Koenkk/zigbee2mqtt #9235: Some Develco Frient SMSZB-120 Smoke detectors Fail to correctly pair

Veldkornet, 2021-10-19:
## What happened
I have 4x Develco Frient SMSZB-120 Smoke detectors.

2x pair and work correctly.
2x par, but don't pull all of the sensors

## What did you expect to happen
That the sensor should look like the below:
[image]
```
{
    "battery": 100,
    "battery_low": false,
    "current_firmware": "3.4.25",
    "enrolled": true,
    "fault": false,
    "ias_cie_address": "0x0012ab002000abcd",
    "linkquality": 255,
    "max_duration": 240,
    "reliability": "no_fault_detected",
    "restore_reports": true,
    "smoke": false,
    "supervision_reports": true,
    "temperature": 21.06,
    "test": false,
    "update": {
        "state": "idle"
    },
    "update_available": false,
    "voltage": 3000,
    "zone_id": 23
}
```

However, it instead loos like this:
[image]
```
{
    "battery_low": false,
    "linkquality": 255,
    "restore_reports": true,
    "smoke": false,
    "supervision_reports": true,
    "test": false,
    "battery": null,
    "fault": null,
    "max_duration": null,
    "reliability": null,
    "temperature": null
}
```

## How to reproduce it (minimal and precise)
Add a SMSZB-120 Smoke detector

## Debug info
Zigbee2MQTT version: 1.21.2
Adapter hardware: Conbee II
Adapter firmware version: 0x26720700

```
Info Device '0x0012ab002000abcd' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0012ab002000abcd","ieee_address":"0x0012ab002000abcd"},"type":"device_joined"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0012ab002000abcd/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0012ab002000abcd"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0012ab002000abcd","sw_version":"Zigbee2MQTT 1.21.2"},"device_class":"temperature","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x0012ab002000abcd","name":"0x0012ab002000abcd temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0x0012ab002000abcd","unique_id":"0x0012ab002000abcd_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0012ab002000abcd/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0012ab002000abcd"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0012ab002000abcd","sw_version":"Zigbee2MQTT 1.21.2"},"device_class":"battery","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x0012ab002000abcd","name":"0x0012ab002000abcd battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x0012ab002000abcd","unique_id":"0x0012ab002000abcd_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Info MQTT publish: topic 'homeassistant/binary_sensor/0x0012ab002000abcd/smoke/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0012ab002000abcd"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0012ab002000abcd","sw_version":"Zigbee2MQTT 1.21.2"},"device_class":"smoke","json_attributes_topic":"zigbee2mqtt/0x0012ab002000abcd","name":"0x0012ab002000abcd smoke","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x0012ab002000abcd","unique_id":"0x0012ab002000abcd_smoke_zigbee2mqtt","value_template":"{{ value_json.smoke }}"}'
Info MQTT publish: topic 'homeassistant/binary_sensor/0x0012ab002000abcd/battery_low/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0012ab002000abcd"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0012ab002000abcd","sw_version":"Zigbee2MQTT 1.21.2"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/0x0012ab002000abcd","name":"0x0012ab002000abcd battery low","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x0012ab002000abcd","unique_id":"0x0012ab002000abcd_battery_low_zigbee2mqtt","value_template":"{{ value_json.battery_low }}"}'
Info MQTT publish: topic 'homeassistant/binary_sensor/0x0012ab002000abcd/test/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0012ab002000abcd"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0012ab002000abcd","sw_version":"Zigbee2MQTT 1.21.2"},"json_attributes_topic":"zigbee2mqtt/0x0012ab002000abcd","name":"0x0012ab002000abcd test","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x0012ab002000abcd","unique_id":"0x0012ab002000abcd_test_zigbee2mqtt","value_template":"{{ value_json.test }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0012ab002000abcd/max_duration/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0012ab002000abcd"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0012ab002000abcd","sw_version":"Zigbee2MQTT 1.21.2"},"enabled_by_default":false,"json_attributes_topic":"zigbee2mqtt/0x0012ab002000abcd","name":"0x0012ab002000abcd max duration","state_topic":"zigbee2mqtt/0x0012ab002000abcd","unique_id":"0x0012ab002000abcd_max_duration_zigbee2mqtt","unit_of_measurement":"s","value_template":"{{ value_json.max_duration }}"}'
Info MQTT publish: topic 'homeassistant/number/0x0012ab002000abcd/max_duration/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x0012ab002000abcd/set/max_duration","device":{"identifiers":["zigbee2mqtt_0x0012ab002000abcd"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0012ab002000abcd","sw_version":"Zigbee2MQTT 1.21.2"},"json_attributes_topic":"zigbee2mqtt/0x0012ab002000abcd","max":600,"min":-65535,"name":"0x0012ab002000abcd max duration","state_topic":"zigbee2mqtt/0x0012ab002000abcd","unique_id":"0x0012ab002000abcd_max_duration_zigbee2mqtt","value_template":"{{ value_json.max_duration }}"}'
Info MQTT publish: topic 'homeassistant/switch/0x0012ab002000abcd/switch_alarm/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x0012ab002000abcd/set/alarm","device":{"identifiers":["zigbee2mqtt_0x0012ab002000abcd"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0012ab002000abcd","sw_version":"Zigbee2MQTT 1.21.2"},"json_attributes_topic":"zigbee2mqtt/0x0012ab002000abcd","name":"0x0012ab002000abcd alarm","payload_off":"OFF","payload_on":"START","state_topic":"zigbee2mqtt/0x0012ab002000abcd","unique_id":"0x0012ab002000abcd_switch_alarm_zigbee2mqtt","value_template":"{{ value_json.alarm }}"}'
Info MQTT publish: topic 'homeassistant/binary_sensor/0x0012ab002000abcd/fault/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0012ab002000abcd"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0012ab002000abcd","sw_version":"Zigbee2MQTT 1.21.2"},"json_attributes_topic":"zigbee2mqtt/0x0012ab002000abcd","name":"0x0012ab002000abcd fault","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x0012ab002000abcd","unique_id":"0x0012ab002000abcd_fault_zigbee2mqtt","value_template":"{{ value_json.fault }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0012ab002000abcd/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0012ab002000abcd"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0012ab002000abcd","sw_version":"Zigbee2MQTT 1.21.2"},"enabled_by_default":false,"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x0012ab002000abcd","name":"0x0012ab002000abcd linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x0012ab002000abcd","unique_id":"0x0012ab002000abcd_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0012ab002000abcd/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0012ab002000abcd"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0012ab002000abcd","sw_version":"Zigbee2MQTT 1.21.2"},"enabled_by_default":false," [... cut ...]
Error Failed to configure '0x0012ab002000abcd', attempt 2 (Error: Bind 0x0012ab002000abcd/35 genBinaryInput from '0x00212effff053a17/1' failed (Error: status: 140) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:757:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:500:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:153:13) at Configure.configure (/app/lib/extension/configure.ts:119:13) at Configure.onMQTTMessage_ (/app/lib/extension/configure.ts:60:21))
```

jamesonuk, 2021-11-01:
Just got myself one of these and I am experiencing the same issue (again with Conbee II).
I had several issues with it failing at interview (but updating to latest Conbee firmware seems to have fixed that).
Also had it drop off and return saying unsupported device and then when repairing I get.
```
Error Failed to configure '0x123', attempt 3 (Error: Bind 0x123/35 genBasic from '0xabc/1' failed (Error: status: 140) 
  at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:757:19) 
  at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:500:13) 
  at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:9) 
  at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:153:13) 
  at Configure.configure (/app/lib/extension/configure.ts:119:13))
```

jamesonuk, 2021-11-01:
OK I have done some further digging and there was a change to the herdsman Develco script since the last release which for this device adds `0x8010` and `0x8020` to
```
await endpoint.read('genBasic', [0x8000, 0x8010, 0x8020], options);
```

updating that script does seem to remove the error above but I am not seeing temperature, reliability etc. as it shows in the OP

The state is now showing up as
```
{
    "battery": 100,
    "battery_low": false,
    "fault": null,
    "linkquality": 255,
    "max_duration": 240,
    "reliability": null,
    "restore_reports": true,
    "smoke": false,
    "supervision_reports": true,
    "temperature": null,
    "test": false,
    "update": {
        "state": "idle"
    },
    "update_available": false,
    "voltage": 3000
}
```

So looks more like a working sensor except I am missing
* current_firmware
* enrolled
* ias_cie_address
* zone_id

and the following aren't getting populated
* fault
* reliability
* temperature

Veldkornet, 2021-11-01:
One of the two just decided to start working out of the blue... no idea why, they've both been sitting 10cm away from the ConBee the whole time while I try figure out what's wrong. 

I saw some small changes for this device that are coming in the next release, so I'm hoping that will help things when it's released.

jamesonuk, 2021-11-01:
> One of the two just decided to start working out of the blue... no idea why, they've both been sitting 10cm away from the ConBee the whole time while I try figure out what's wrong.

I have had it go the other way though.   It had paired and seemed OK but then the next day it randomly appeared as an unsupported device

Veldkornet, 2021-11-01:
Uff, 1.22.0-2 giving errors on one of my existing ones now:

```
Zigbee2MQTT:error 2021-11-01 22:48:39: Failed to configure 'Keuken Rookmelder', attempt 1 (Error: Bind 0x0015bc0031009605/35 genBinaryInput from '0x00212effff053a17/1' failed (Error: status: 140)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:757:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:500:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:165:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))
```

jamesonuk, 2021-11-09:
I have seen the same issue and still no temperature readings.
I don't know enough about Zigbee to understand but here is the co-ordinator / bind tabs of a little Sonoff temperature sensor I have

[image]
[image]

But for the Frient HESZB-120 (Heat Sensor which appears to be just about identical to the SMSZB-120 and both show the same) I have multiple endpoints with temperature being under 38.   On the bind tab if I select endpoint 38 only Identity shows up

[image]
[image]

I don't have enough knowledge to know whether this helps, is a red-herring or meaningless but thought I would post here in case someone who does understand these things is able to spot an issue

sjabby, 2021-11-10:
Have similar problem, i got 4 devices. 3 of them works fine. 1 is acting up and not reporting temperature etc.
Logfile:

Zigbee2MQTT:error 2021-11-10 20:53:22: Failed to configure '0x0015bc003100a73f', attempt 1 (Error: Bind 0x0015bc003100a73f/35 ssIasZone from '0x00212effff0676ee/1' failed (waiting for response TIMEOUT),
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:757:19),
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:500:13),
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:9),
    at Configure.configure (/app/lib/extension/configure.ts:115:13)),
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:165:13),
Zigbee2MQTT:info  2021-11-10 20:53:29: Configuring '0x0015bc003100a73f',
Zigbee2MQTT:debug 2021-11-10 20:53:29: Received Zigbee message from '0x0015bc003100a73f', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 35 with groupID null,
Zigbee2MQTT:info  2021-11-10 20:53:29: MQTT publish: topic 'zigbee2mqtt/0x0015bc003100a73f', payload '{"alarm":null,"battery":100,"battery_low":false,"fault":null,"linkquality":207,"max_duration":null,"reliability":null,"smoke":null,"temperature":null,"test":null,"update_available":null,"voltage":3000}',
Zigbee2MQTT:debug 2021-11-10 20:53:29: Received Zigbee message from '0x0015bc003100a73f', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryAlarmState":0,"batteryVoltage":30}' from endpoint 35 with groupID null,
Zigbee2MQTT:debug 2021-11-10 20:53:29: No converter available for 'SMSZB-120' with cluster 'genPollCtrl' and type 'commandCheckIn' and data '{}',
Zigbee2MQTT:debug 2021-11-10 20:53:30: Received Zigbee message from '0x0015bc003100a73f', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":48}' from endpoint 35 with groupID null,
Zigbee2MQTT:info  2021-11-10 20:53:30: MQTT publish: topic 'zigbee2mqtt/0x0015bc003100a73f', payload '{"alarm":null,"battery":100,"battery_low":false,"fault":null,"linkquality":199,"max_duration":null,"reliability":null,"restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":null,"test":false,"update_available":null,"voltage":3000}',
Zigbee2MQTT:info  2021-11-10 20:52:16: Device '0x0015bc003100a73f' joined,
Zigbee2MQTT:info  2021-11-10 20:52:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc003100a73f","ieee_address":"0x0015bc003100a73f"},"type":"device_joined"}',
Zigbee2MQTT:info  2021-11-10 20:52:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0015bc003100a73f"},"type":"device_connected"}',
Zigbee2MQTT:info  2021-11-10 20:52:16: Starting interview of '0x0015bc003100a73f',
Zigbee2MQTT:info  2021-11-10 20:52:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x0015bc003100a73f"},"type":"pairing"}',
Zigbee2MQTT:info  2021-11-10 20:52:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc003100a73f","ieee_address":"0x0015bc003100a73f","status":"started"},"type":"device_interview"}',
Zigbee2MQTT:debug 2021-11-10 20:52:19: Received Zigbee message from '0x0015bc003100a73f', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":48}' from endpoint 35 with groupID null,
Zigbee2MQTT:debug 2021-11-10 20:52:19: Skipping message, definition is undefined and still interviewing,
Zigbee2MQTT:debug 2021-11-10 20:52:19: Received Zigbee message from '0x0015bc003100a73f', type 'readResponse', cluster 'genBasic', data '{"modelId":"SMSZB-120"}' from endpoint 35 with groupID null,
Zigbee2MQTT:debug 2021-11-10 20:52:19: Received Zigbee message from '0x0015bc003100a73f', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"Develco Products A/S"}' from endpoint 35 with groupID null,
Zigbee2MQTT:debug 2021-11-10 20:52:19: Received Zigbee message from '0x0015bc003100a73f', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 35 with groupID null,
Zigbee2MQTT:debug 2021-11-10 20:52:20: Received Zigbee message from '0x0015bc003100a73f', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 35 with groupID null,
Zigbee2MQTT:debug 2021-11-10 20:52:20: Skipping message, definition is undefined and still interviewing,
Zigbee2MQTT:debug 2021-11-10 20:52:20: Received Zigbee message from '0x0015bc003100a73f', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 35 with groupID null,
Zigbee2MQTT:debug 2021-11-10 20:52:20: Received Zigbee message from '0x0015bc003100a73f', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20210622 08:49"}' from endpoint 35 with groupID null,
Zigbee2MQTT:debug 2021-11-10 20:52:20: Received Zigbee message from '0x0015bc003100a73f', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 38 with groupID null,
Zigbee2MQTT:debug 2021-11-10 20:52:21: Received Zigbee message from '0x0015bc003100a73f', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 38 with groupID null,
Zigbee2MQTT:debug 2021-11-10 20:52:21: Skipping message, definition is undefined and still interviewing,
[... cut ...]
Zigbee2MQTT:debug 2021-11-10 20:52:21: Received MQTT message on 'homeassistant/sensor/0x0015bc003100a73f/update_state/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0015bc003100a73f"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0015bc003100a73f","sw_version":"Zigbee2MQTT 1.22.0"},"enabled_by_default":false,"icon":"mdi:update","json_attributes_topic":"zigbee2mqtt/0x0015bc003100a73f","name":"0x0015bc003100a73f update state","state_topic":"zigbee2mqtt/0x0015bc003100a73f","unique_id":"0x0015bc003100a73f_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'

Inrego, 2021-11-15:
I just got one of these, and am also seeing the same issue :/

PacDeSaintes, 2021-12-03:
Hello friends,
I just received 2 and had the same problem to get temperature value.

jkaberg, 2021-12-16:
I have the Develco (none Frient version) and some them also have problems with temperature.

When trying to reconfigure (the none OK one):
```Failed to configure '0x0015bc003100d36f', attempt 4 (Error: Bind 0x0015bc003100d36f/35 genBasic from '0x00212effff04246b/1' failed (Error: status: 140) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:753:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:517:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:165:13) at Configure.configure (/app/lib/extension/configure.ts:115:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21))```

[image]

erazor666, 2021-12-27:
`Zigbee2MQTT:error 2021-12-27 13:38:36: Failed to configure 'smokea_1etg', attempt 1 (Error: Bind 0x0015bc00310061c8/35 genBasic from '0x00212effff06d6b3/1' failed (Error: status: 140)
    at DeconzAdapter.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:753:19)
    at Endpoint.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:517:13)
    at Object.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:9)
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/develco.js:165:13)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:115:13))
Zigbee2MQTT:error 2021-12-27 13:39:23: Failed to configure 'smokea_kjeller', attempt 1 (Error: Bind 0x0015bc00310061e4/35 genPowerCfg from '0x00212effff06d6b3/1' failed (waiting for response TIMEOUT)
    at DeconzAdapter.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:753:19)
    at runNextTicks (internal/process/task_queues.js:60:5)
    at processTimers (internal/timers.js:497:9)
    at Endpoint.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:517:13)
    at Object.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:9)
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/develco.js:165:13)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:115:13))
`

Same issue here, not reporting temperature, reporting voltage and battery however.
[image]

Conbee II stick, when using Deconz with same stick and same smoke detector it works and reports temperature.

[image]

bouyssic, 2022-01-10:
Hello team,

Just got 2 of those for home and I'm experiencing the same with both:
```
Error` Failed to configure 'Living Room Fire Alarm', attempt 4 (Error: Bind 0x0015bc003100ad92/35 genBasic from '0x00212effff0661ad/1' failed (Error: status: 140) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:753:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:517:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:165:13) at Configure.configure (/app/lib/extension/configure.ts:115:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21))
```

As per other users I'm using a conbeeII with deconz adapter. I'll keep digging see if I can find something else.

```
Zigbee2MQTT version 1.22.2 commit: unknown
Coordinator type ConBee2/RaspBee2 
Coordinator revision 0x26660700
Frontend version 0.6.67
```

Full debug of device joining:
```
Info Device '0x0015bc003100ae23' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc003100ae23","ieee_address":"0x0015bc003100ae23"},"type":"device_joined"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0015bc003100ae23"},"type":"device_connected"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0015bc003100ae23/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0015bc003100ae23"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0015bc003100ae23"},"device_class":"temperature","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x0015bc003100ae23","name":"0x0015bc003100ae23 temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0x0015bc003100ae23","unique_id":"0x0015bc003100ae23_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0015bc003100ae23/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0015bc003100ae23"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0015bc003100ae23"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x0015bc003100ae23","name":"0x0015bc003100ae23 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x0015bc003100ae23","unique_id":"0x0015bc003100ae23_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Info MQTT publish: topic 'homeassistant/binary_sensor/0x0015bc003100ae23/smoke/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0015bc003100ae23"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0015bc003100ae23"},"device_class":"smoke","json_attributes_topic":"zigbee2mqtt/0x0015bc003100ae23","name":"0x0015bc003100ae23 smoke","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x0015bc003100ae23","unique_id":"0x0015bc003100ae23_smoke_zigbee2mqtt","value_template":"{{ value_json.smoke }}"}'
Info MQTT publish: topic 'homeassistant/binary_sensor/0x0015bc003100ae23/battery_low/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0015bc003100ae23"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0015bc003100ae23"},"device_class":"battery","entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x0015bc003100ae23","name":"0x0015bc003100ae23 battery low","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x0015bc003100ae23","unique_id":"0x0015bc003100ae23_battery_low_zigbee2mqtt","value_template":"{{ value_json.battery_low }}"}'
Info MQTT publish: topic 'homeassistant/binary_sensor/0x0015bc003100ae23/test/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0015bc003100ae23"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0015bc003100ae23"},"entity_category":"diagnostic","icon":"mdi:test-tube","json_attributes_topic":"zigbee2mqtt/0x0015bc003100ae23","name":"0x0015bc003100ae23 test","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x0015bc003100ae23","unique_id":"0x0015bc003100ae23_test_zigbee2mqtt","value_template":"{{ value_json.test }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0015bc003100ae23/max_duration/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0015bc003100ae23"],"manufacturer":"Develco","model":"Smoke detector with siren (SMSZB-120)","name":"0x0015bc003100ae23"},"enabled_by_default":false,"json_attributes_topic":"zigbee2mqtt/0x0015bc003100ae23","name":"0x0015bc003100ae23 max duration","state_topic":"zigbee2mqtt/0x0015bc003100ae23","unique_id":"0x0015bc003100ae23_max_duration_zigbee2mqtt","unit_of_measurement":"s","value_template":"{{ value_json.max_duration }}"}'
Info MQTT publish: topic 'homeassistant/number/0x0015bc003100ae23/max_duration/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x0015bc003100ae23/set/max_durati [... cut ...]
Info MQTT publish: topic 'zigbee2mqtt/0x0015bc003100ae23', payload '{"alarm":null,"battery":null,"battery_low":false,"enrolled":true,"fault":null,"ias_cie_address":"0x00212effff0661ad","linkquality":255,"max_duration":null,"reliability":null,"restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":null,"test":false,"update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from '0x0015bc003100ae23', type 'read', cluster 'genTime', data '["time"]' from endpoint 35 with groupID null
```

erazor666, 2022-01-15:
Came across a pretty detailed manual for these smoke detectors, maybe it can help https://www.develcoproducts.com/media/1742/smszb-120-technical-manual-smoke-alarm.pdf

chino-lu, 2022-01-16:
this helped me to get the temperature using a Conbee II stick: https://community.home-assistant.io/t/strange-device-pairing-with-zigbee2mqtt/376945/2

Veldkornet, 2022-01-16:
> this helped me to get the temperature using a Conbee II stick: https://community.home-assistant.io/t/strange-device-pairing-with-zigbee2mqtt/376945/2

That helps get the temperature indeed 👍🏼 But there's still more missing.

chino-lu, 2022-01-16:
as already posted on home-assistant: 
after a manual request of the ias parameters I got the following:
[image]

the status tab looks like the following:
`{
    "battery": 100,
    "battery_low": false,
    "enrolled": true,
    "fault": false,
    "ias_cie_address": "0x00212effff071fba",
    "last_seen": "2022-01-16T09:29:16+01:00",
    "linkquality": 255,
    "max_duration": 3,
    "reliability": "no_fault_detected",
    "restore_reports": true,
    "smoke": false,
    "supervision_reports": true,
    "temperature": 18.5,
    "test": false,
    "update": {
        "state": "idle"
    },
    "update_available": false,
    "voltage": 3000,
    "zone_id": 23,
    "alarm": null
}`

all I did was reading the following parameters manually (well not all I did... but during trial and error, this was the step which made them visible):
[image]

NeuerUser, 2022-01-20:
> as already posted on home-assistant: after a manual request of the ias parameters I got the following: [image]
>
> the status tab looks like the following: `{ "battery": 100, "battery_low": false, [...]
Same problem here with 2 new frient detectors. Reading these parameters, however, did not change anything. Still missing many values. :(

chino-lu, 2022-01-20:
> Same problem here with 2 new frient detectors. Reading these parameters, however, did not change anything. Still missing all these values. :(
> 
> > all I did was reading the following parameters manually (well not all I did... but during trial and error, this was the step which made them visible): [image]

what was the reply you got when clicking the green reading button?

NeuerUser, 2022-01-20:
[image]
No error. Everything reported as ok. But still:
[image]

chino-lu, 2022-01-20:
ok, this means that this was not the part of my trial and error which fixed it... 

something else I tried was the reporting like this:
[image]

but as stated before, this ends in an error

NeuerUser, 2022-01-20:
This also gives an error on my side and no change on the dashboard. I am not on the edge version though:
[image]

PacDeSaintes, 2022-01-29:
Still no change with this Edge config :

[image]

Veldkornet, 2022-02-01:
I decided to switch from the **ConBee II** to the **SONOFF Zigbee 3.0 USB Dongle Plus**; which resulted in re-pairing everything.

With the SONOFF, all 4x of my smoke detectors seem to have paired correctly. Also the ones which didn't work previously now have all of their attributes.

Maybe it's just coincidence, or maybe it's a problem with the ConBee II?

[image]

erazor666, 2022-02-03:
I also switched to a SONOFF Zigbee 3.0 USB Dongle Plus, I still cant seem to get the smoke detectors to connect properly. Using the 20220103 coordinator firmware

`Zigbee2MQTT:info  2022-02-03 11:42:38: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-02-03.11-42-34' filename: log.txt
Zigbee2MQTT:info  2022-02-03 11:42:38: Starting Zigbee2MQTT version 1.23.0 (commit #6ea61bae)
Zigbee2MQTT:info  2022-02-03 11:42:38: Starting zigbee-herdsman (0.14.10)
Zigbee2MQTT:info  2022-02-03 11:43:18: zigbee-herdsman started (restored)
Zigbee2MQTT:info  2022-02-03 11:43:18: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220103,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2022-02-03 11:43:18: Currently 0 devices are joined:
Zigbee2MQTT:warn  2022-02-03 11:43:18: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2022-02-03 11:43:18: Allowing new devices to join.
Zigbee2MQTT:warn  2022-02-03 11:43:18: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2022-02-03 11:43:18: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-02-03 11:43:19: Connecting to MQTT server at mqtt://10.0.0.9
Zigbee2MQTT:info  2022-02-03 11:43:19: Connected to MQTT server
Zigbee2MQTT:info  2022-02-03 11:43:19: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2022-02-03 11:43:19: Started frontend on port 0.0.0.0:8080
Zigbee2MQTT:info  2022-02-03 11:47:17: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-02-03 11:47:18: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"tmei6-1"}'
Zigbee2MQTT:info  2022-02-03 11:47:18: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-02-03 11:47:18: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":false},"status":"ok"}'
Zigbee2MQTT:info  2022-02-03 11:47:20: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-02-03 11:47:20: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"tmei6-2"}'
Zigbee2MQTT:info  2022-02-03 11:49:16: Device '0x0015bc00310061e4' joined
Zigbee2MQTT:info  2022-02-03 11:49:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00310061e4","ieee_address":"0x0015bc00310061e4"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-02-03 11:49:16: Starting interview of '0x0015bc00310061e4'
Zigbee2MQTT:info  2022-02-03 11:49:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00310061e4","ieee_address":"0x0015bc00310061e4","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-02-03 11:49:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00310061e4","ieee_address":"0x0015bc00310061e4"},"type":"device_announce"}'
Zigbee2MQTT:info  2022-02-03 11:49:20: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-02-03 11:49:20: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":false},"status":"ok"}'
Zigbee2MQTT:info  2022-02-03 11:49:20: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-02-03 11:49:20: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":false},"status":"ok"}'
Zigbee2MQTT:info  2022-02-03 11:49:35: Successfully interviewed '0x0015bc00310061e4', device has successfully been paired
Zigbee2MQTT:info  2022-02-03 11:49:35: Device '0x0015bc00310061e4' is supported, identified as: Develco Smoke detector with siren (SMSZB-120)
Zigbee2MQTT:info  2022-02-03 11:49:35: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smoke detector with siren","exposes":[{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Indicates whether the device detected smoke","name":"smoke","property":"smoke","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Indicates if the battery of this device is almost empty","name":"battery_low","property":"battery_low","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Indicates whether the device is being tested","name":"test","property":"test","type":"binary","value_off":false,"value_on":true},{"access":7,"description":"Duration of Siren","name":"max_duration","property":"max_duration","type":"numeric","unit":"s","value_max":600,"value_min":0},{"access":2,"description":"Manual Start of Siren","name":"alarm","property":"alarm","type":"binary","value_off":"OFF","value_on":"START"},{"access":1,"description":"Indicates reason if any fault","name":"reliability","property":"reliability","type":"enum","values":["no_fault_detected","unreliable_other","process_error"]},{"access":1,"description":"Indicates whether the device are in fault state","name":"fault","property":"fault","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"SMSZB [... cut ...]
    at processTimers (internal/timers.js:497:7))
Zigbee2MQTT:info  2022-02-03 11:50:18: Configuring '0x0015bc00310061e4'
Zigbee2MQTT:error 2022-02-03 11:50:28: Failed to configure '0x0015bc00310061e4', attempt 3 (Error: Bind 0x0015bc00310061e4/35 genPowerCfg from '0x00124b0024bc493f/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))
`

Veldkornet, 2022-02-07:
Arg, out of the blue, one of mine stopped working again:

```
error 2022-02-07 12:39:36: Exception while calling fromZigbee converter: Cannot set property 'softwareBuildID' of undefined}
warn  2022-02-07 12:41:33: Device 'Keuken Rookmelder' left the network
error 2022-02-07 12:41:53: Failed to configure 'Keuken Rookmelder', attempt 1 (Error: Bind 0x0015bc0031009605/35 genPowerCfg from '0x00124b00258d4fb3/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7))
error 2022-02-07 12:42:03: Failed to interview 'Keuken Rookmelder', device has not successfully been paired
error 2022-02-07 12:42:08: Exception while calling fromZigbee converter: Cannot set property 'softwareBuildID' of undefined}
error 2022-02-07 12:42:09: Exception while calling fromZigbee converter: Cannot set property 'softwareBuildID' of undefined}
```

erazor666, 2022-02-10:
This just keeps on giving:

`Zigbee2MQTT:info  2022-02-10 13:05:38: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-02-10.13-05-35' filename: log.txt
Zigbee2MQTT:info  2022-02-10 13:05:39: Starting Zigbee2MQTT version 1.23.0 (commit #6ea61bae)
Zigbee2MQTT:info  2022-02-10 13:05:39: Starting zigbee-herdsman (0.14.10)
Zigbee2MQTT:info  2022-02-10 13:05:51: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-02-10 13:05:51: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220103,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2022-02-10 13:05:51: Currently 0 devices are joined:
Zigbee2MQTT:warn  2022-02-10 13:05:51: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2022-02-10 13:05:51: Allowing new devices to join.
Zigbee2MQTT:warn  2022-02-10 13:05:51: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2022-02-10 13:05:51: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-02-10 13:05:52: Connecting to MQTT server at mqtt://10.0.0.9
Zigbee2MQTT:info  2022-02-10 13:05:52: Connected to MQTT server
Zigbee2MQTT:info  2022-02-10 13:05:52: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2022-02-10 13:05:52: Started frontend on port 0.0.0.0:8080
Zigbee2MQTT:info  2022-02-10 13:09:32: Device '0x0015bc00310061c8' joined
Zigbee2MQTT:info  2022-02-10 13:09:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00310061c8","ieee_address":"0x0015bc00310061c8"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-02-10 13:09:32: Starting interview of '0x0015bc00310061c8'
Zigbee2MQTT:info  2022-02-10 13:09:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00310061c8","ieee_address":"0x0015bc00310061c8","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-02-10 13:09:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00310061c8","ieee_address":"0x0015bc00310061c8"},"type":"device_announce"}'
Zigbee2MQTT:error 2022-02-10 13:11:50: Failed to interview '0x0015bc00310061c8', device has not successfully been paired
Zigbee2MQTT:info  2022-02-10 13:11:50: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00310061c8","ieee_address":"0x0015bc00310061c8","status":"failed"},"type":"device_interview"}'
(node:12053) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
(node:12053) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:12053) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
`

erazor666, 2022-02-10:
[image]

Just fired up Deconz for shit and giggles. Works instantly.

TheNetStriker, 2022-02-16:
I just bought an Frient (Develco) SMSZB-120 smoke detector (Hardware version 5.0.2) and tried for several hours to pair the device, but the device is not recognized by zigbee2mqtt at all. I've tried with multiple batteries and with the device near the coordinator and I did reset the device multiple times, but nothing worked so far.

Here are the details of my system:
zigbee2mqtt: Version 1.23.0 running on Raspberry Pi 4
Coordinator: Texas Instruments LAUNCHXL-CC1352P-2
Coordinator Firmware: zStack3x0 20211217

Sadly there is no information in the log regarding the pairing. I also sniffed the Zigbee traffic. I can see the beacon requests of the device, but no beacon response from the coordinator. Other devices are pairing without any problems.

closed 2022-03-27 by github-actions[bot]

boelle, 2023-03-30:
still a problem, and its random

got 2 of them, one works out the box... the other  i have given the 15 sec reset... tried battery out....

[image]

the working one

boelle, 2023-03-30:
@Koenkk any idea why we have this issue?

crivchri, 2023-05-24:
Hi @Koenkk, I am having this same issue.
Do you have any idea?
Thanks and kind regards

crivchri, 2023-05-24:
> https://community.home-assistant.io/t/strange-device-pairing-with-zigbee2mqtt/376945/2

I found out how to fix this, by manually adding three 'Reporting' rows (see picture below).

[image]