Koenkk/zigbee2mqtt #9262: Please help to support Aqara wireless remote switch E1.

z-x-f, 2021-10-21:
## data/database.db entry of the device
<pre><code>
{"id":43,"type":"Router","ieeeAddr":"0x54ef44100027abd6","nwkAddr":59728,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.switch.b3n01","epList":[1,2,3,21,41,42,43,51,52,53,61,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2,3,4,5,6,9,1794,2820],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.switch.b3n01","manufacturerName":"LUMI","powerSource":1,"zclVersion":3,"appVersion":23,"stackVersion":2,"hwVersion":1,"dateCode":"04-16-2021"}},"aqaraOpple":{"attributes":{"229":{"type":"Buffer","data":[9,0,9,0,0,4,37,224,231,37,60,214,163,37,82,6,221,37,117,31,24,37,71,32,206,37,33,49,161,37,27,114,53,37,27,100,231,37,10]},"247":{"type":"Buffer","data":[100,16,0,101,16,0,102,16,0,3,40,21,152,57,0,216,6,59,149,57,175,12,148,55,150,57,109,151,18,69,151,57,0,0,0,0,5,33,8,0,154,32,0,8,33,23,1,9,33,0,10,11,32,0,10,33,214,163,12,32,1]},"512":1,"513":0,"515":0}},"genOnOff":{"attributes":{"245":65622016,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"aqaraOpple":{"attributes":{"512":1}},"genOnOff":{"attributes":{"245":65622016,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"aqaraOpple":{"attributes":{"512":1}},"genOnOff":{"attributes":{"245":65622016,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"epId":21,"inClusterList":[],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[],"configuredReportings":[],"meta":{}},"41":{"epId":41,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"42":{"epId":42,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"43":{"epId":43,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"51":{"epId":51,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":2}}},"binds":[],"configuredReportings":[],"meta":{}},"52":{"epId":52,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":2}}},"binds":[],"configuredReportings":[],"meta":{}},"53":{"epId":53,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":2}}},"binds":[],"configuredReportings":[],"meta":{}},"61":{"epId":61,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":2}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":23,"stackVersion":2,"hwVersion":1,"dateCode":"04-16-2021","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2114939527},"lastSeen":1634830713649}
</code></pre>
Log after pairing:
<pre><code>
Debug Received Zigbee message from '0x54ef441000311cdb', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.remote.acn004"}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0x54ef441000311cdb', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0x54ef441000311cdb', type 'attributeReport', cluster 'genBasic', data '{"appVersion":6,"modelId":"lumi.remote.acn004"}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0x54ef441000311cdb', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0x54ef441000311cdb', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0x54ef441000311cdb', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0x54ef441000311cdb', type 'readResponse', cluster 'genBasic', data '{"hwVersion":0}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0x54ef441000311cdb', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0x54ef441000311cdb', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Info Successfully interviewed '0x54ef441000311cdb', device has successfully been paired
Warning Device '0x54ef441000311cdb' with Zigbee model 'lumi.remote.acn004' and manufacturer name 'LUMI' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
</code></pre>
I tried to add the converter and re-pair the device. After pairing, I press any wireless switch button, there is no response in z2m. Only after pressing the button 5 times quickly, I will get this in log: 

<code>Warning Received message from unsupported device with Zigbee model'lumi. remote.acn004' and manufacturer name'LUMI'</code>

I uploaded the sniffed file , hope it helps.
[wxkg17lm.zip](https://github.com/Koenkk/zigbee2mqtt/files/7391241/wxkg17lm.zip)

z-x-f, 2021-11-01:
UPDATE: According to the method provided here, I use the new compiled firmware to try, the E1 wireless switch can work normally.
https://github.com/Koenkk/zigbee2mqtt/issues/9274#issuecomment-955803433

Koenkk, 2021-11-01:
@z-x-f great, I will check if it is possible to integrate this in the default firmware.

Koenkk, 2021-11-02:
can you test with fws from https://github.com/Koenkk/zigbee2mqtt/issues/9274#issuecomment-958146099

z-x-f, 2021-11-03:
@Koenkk I use another usb stick to test your firmware and it works normally.

Koenkk, 2021-11-03:
@z-x-f what `configure()` did you use? Did you manage to get battery reporting working?

z-x-f, 2021-11-03:
@Koenkk battery I only used fz.aqara_opple_report, and then I can get the voltage normally.

Koenkk, 2021-11-03:
@z-x-f so no `configure()` needed at all?

z-x-f, 2021-11-03:
@Koenkk Yes, I tested it last night, I can test it again tonight and give you feedback

z-x-f, 2021-11-03:
@Koenkk Hi, KK, This is the converter I use. One problem is that the action is always displayed as the button just triggered，How can I solve it
```js
    {
        zigbeeModel: ['lumi.remote.acn004'],
        model: 'WXKG17LM',
        vendor: 'Xiaomi',
        description: 'Aqara wireless remote switch E1 (double rocker)',
        fromZigbee: [fz.xiaomi_multistate_action, fz.aqara_opple_report],
        toZigbee: [],
        exposes: [e.battery(), e.battery_voltage(),
            e.action(['single_left', 'single_right'])],
        meta: {battery: {voltageToPercentage: '3V_2500'}, multiEndpoint: true},
        endpoint: (device) => {
            return {left: 1, right: 2};
        },
        onEvent: preventReset,
    },
```
[image]
<pre>
Info Successfully interviewed '0x54ef441000311cdb', device has successfully been paired
Info Device '0x54ef441000311cdb' is supported, identified as: Xiaomi Aqara wireless remote switch E1 (double rocker) (WXKG17LM)
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara wireless remote switch E1 (double rocker)","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["single_left","single_right"]},{"access":1,"description":"Voltage of the battery in millivolts","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"WXKG17LM","options":[],"supports_ota":false,"vendor":"Xiaomi"},"friendly_name":"0x54ef441000311cdb","ieee_address":"0x54ef441000311cdb","status":"successful","supported":true},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Aqara wireless remote switch E1 (double rocker)","friendly_name":"0x54ef441000311cdb","model":"WXKG17LM","supported":true,"vendor":"Xiaomi"},"type":"pairing"}'
Debug Received MQTT message on 'zigbee2mqtt/bridge/config/device_options' with data '{"friendly_name":"0x54ef441000311cdb","options":{"retain":true}}'
Info Changed device specific options of '0x54ef441000311cdb' ({"retain":true})
Debug Received Zigbee message from '0x54ef441000311cdb', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,170,11,3,40,25,4,33,0,0,5,33,1,0,8,33,6,0,10,33,0,0,12,32,30,100,32,1],"type":"Buffer"}}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x54ef441000311cdb', payload '{"battery":92,"linkquality":135,"voltage":2986}'
</pre>

z-x-f, 2021-11-03:
Ah, the cluster 'aqaraOpple' seems to be reported once when it is first paired, and then when the button is pressed five times quickly, but I haven't found it actively reporting.
I couldn't find a way to get voltage from other places.

Koenkk, 2021-11-03:
Great, Please create a pr?

Note: `{battery: {voltageToPercentage: '3V_2500'}` -> this can be removed, it is not used in `fz.aqara_opple_report`

z-x-f, 2021-11-03:
Okay, no problem. 
Is there a way to solve the problem that the trigger action always exists in the <code>action</code>?

Koenkk, 2021-11-03:
If you are using the home assistant integration this is expected, see https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty

z-x-f, 2021-11-03:
Aha, my problem is exactly the opposite. It will not become empty after triggering. For example, now my action is always single_left or single_right.

z-x-f, 2021-11-04:
Someone has already created a PR for this device, and he has done better.

closed 2021-11-04 by Koenkk