Koenkk/zigbee2mqtt #9403: z2m groups are not discovered by HA

FlexxFR, 2021-10-29:
## What happened
Created z2m groups using HA add-on webinterface.
Checked the groups.yaml, it;s properly updated. I believe my config is according https://www.zigbee2mqtt.io/information/groups.html

## What did you expect to happen
Groups to discovered as entity after reloading add-on. (or a even restart)
According the info I believe these groups should be auto-discovered.
But whatever I try, I don't see any groups in HA.

## How to reproduce it (minimal and precise)
Create a group via web GUI, that results into this: 

'100':
  friendly_name: z2m_verlichting_eettafel
  devices:
    - 0xccccccfffeed97fb/1
    - 0x14b457fffe34c5d5/1
    - 0xec1bbdfffe2f3691/1

No errors notices during the creation. Group is not discovered from HA. (not after reloading/restarting/rebooting)

## Debug info
Zigbee2MQTT version: Zigbee2MQTT version 1.18.1 (commit #1.18.1) / zigbee-herdsman (0.13.71)
Adapter hardware: slae.sh_cc2652rb_stick
Adapter firmware version: ?

Koenkk, 2021-10-29:
You are using an old version which does not support this yet, update to the latest.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

closed 2021-10-29 by Koenkk

FlexxFR, 2021-10-29:
Oh man, how could I missed that...?? Already a while ago.... thans for the fast reply, I was really stuck searching in a complete different direction....

Update was a "bumby ride" but I managed to restore the old config for the new version... :)
And after the update the groups appeared as expected.