Koenkk/zigbee2mqtt #9409: Support Moes/Zonnsmart TV01-ZG thermostat (TRV)

slydiman, 2021-10-30:
I saw the same device as Zonnsmart TV01-ZG, but I bought it as Moes here https://aliexpress.ru/item/1005003136452052.html

Manufacturer: _TZE200_e9ba97vf
Model: TS0601
Application Version: 69

It is very close to https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html and converters are the same.
But the main problem is that this device does not report the current temperature (local_temperature) itself without a secret command during initialization. Note the current temperature is updated immediately in the Tuya app via Tuya gateway.

Here is the  .pcapng dump of the peering process using the Tuya gateway http://slydiman.me/_t/tuya_moes_trv.zip
I used the following filter `zbee_nwk.addr eq 0x4b6b and zbee_nwk.addr eq 0x0000`.

As a workaround it is possible to set the local_temperature_calibration each 15 minutes to force update local_temperature.

Hope it is possible to add the complete support of this device.

insipiens, 2021-11-01:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3185

slydiman, 2021-11-01:
> Koenkk/zigbee-herdsman-converters#3185
```
Zigbee2MQTT version
1.21.2-dev commit: 109c62b
Coordinator type
zStack30x
Coordinator revision
20210212
Frontend version
0.6.30
```
Note the commit 109c62b was today.

I see the device in the list, but
```
Friendly name: 0x84fd27fffe2debac
Last seen: N/A
Device type: Unknown
Zigbee Model: <empty>
Zigbee Manufacturer: <empty>
Support status: Not supported
```

Log:
```
info  2021-11-01 21:30:46: Zigbee: allowing new devices to join.
info  2021-11-01 21:30:46: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"rb2zj-13"}'
info  2021-11-01 21:32:25: Device '0x84fd27fffe2debac' joined
info  2021-11-01 21:32:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84fd27fffe2debac","ieee_address":"0x84fd27fffe2debac"},"type":"device_joined"}'
info  2021-11-01 21:32:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x84fd27fffe2debac"},"type":"device_connected"}'
info  2021-11-01 21:32:25: Starting interview of '0x84fd27fffe2debac'
info  2021-11-01 21:32:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84fd27fffe2debac","ieee_address":"0x84fd27fffe2debac","status":"started"},"type":"device_interview"}'
info  2021-11-01 21:32:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x84fd27fffe2debac"},"type":"pairing"}'
debug 2021-11-01 21:34:01: Saving state to file /config/zigbee2mqtt/state.json
error 2021-11-01 21:34:44: Failed to interview '0x84fd27fffe2debac', device has not successfully been paired
```

It worked with custom converters but the problem was that the device did not report local_temperature. I updated local_temperature_calibration to force update local_temperature.

slydiman, 2021-11-02:
> [Koenkk/zigbee-herdsman-converters#3185](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3185)

Something is broken in the interview process. Device is joined but the interview failed. I used the  latest dev version. USB stick is CC2538 ver 20201010.

slydiman, 2021-11-02:
> [Koenkk/zigbee-herdsman-converters#3185](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3185)

Right the converter is ok. It worked before with a custom converter too.

As I mentioned in the initial post the main problem is that this device does not report the current temperature (local_temperature) itself without a secret command during initialization. Note the current temperature is updated immediately in the Tuya app via Tuya gateway.

insipiens, 2021-11-02:
The Tuya hub only uses zigbee, so not secret but undiscovered….

slydiman, 2021-11-02:
I meant a secret/uncommon/special zigbee command/attribute.
Therefore I have attached the .pcapng dump of the peering process using the Tuya gateway which may help to discover it.

insipiens, 2021-11-02:
A better way would be use the ‘how to add a tuya device’ and chuck a bag of ice over the device. You should then get some ‘secret’ messages. 😃

drdownload, 2021-11-02:
Look at #8522 thats the same device. we've found a way to update the temperature

insipiens, 2021-11-02:
The unfortunate thing about Tuya devices is they allow their clients to choose the DPs.

thiscantbeserious, 2021-11-22:
It seems like that this also leads to the valve not adjusting itself correctly when connected to zigbee2mqtt for a longer amount of time - not even by using the buttons on the device itself.

thiscantbeserious, 2021-11-25:
[image]

Here's a picture of the thing in its parts - so far hardware-wise everything seems solid and easily repairable. No glue used, not even on the screen-layers ...

Only thing there is that there might be somewhat a maintenance mode that turns the valve position more in because I had to apply some force opening up the case ...

thiscantbeserious, 2021-11-29:
See here - https://github.com/Koenkk/zigbee2mqtt/issues/9858#issuecomment-981719178

Datapoint 115 could actually the right thing to call - with the payload I mentioned there. This was seemingly triggered when I opened the App. The screen didn't turn on but the temperature updated in the App view there.

I didn't log the response yet tought.

closed 2022-01-06 by github-actions[bot]

Tommixoft, 2023-01-24:
Well looks it still not fixed.. sad.. usless e-waste without temp reporting.

srett, 2025-01-28:
@drdownload @thiscantbeserious @Tommixoft 

Anyone here still using this TRV?

I never had this problem since I started using it about a year ago. However, now with the update to z2m 2.0, I face this very same issue: local_temperature never (or almost never) updates. A quick git blame of `tuya.ts` doesn't reveal any recent changes regarding the TRV directly, so I wonder if something in the many helper functions changed. But before losing to much time on tracing this I thought I might try to figure out if anyone else is experiencing this. Maybe this is just coincidence and my device went bad.

Attached is a screenshot from HA, you see the slight interruption where I updated z2m and then it stopped working and only updated erratically once a day or so.

[image]

srett, 2025-01-28:
@thiscantbeserious thanks for getting back to me. I added these values, however the way I interpret #24198 this would only really matter for older setups, my install isn't even a year old. I added the values anyway and restarted, but unfortunately no change. :-(
The TRV works in general, I can adjust the heating setpoint, but the local temperature only changes once in a honeymoon.
What's also odd to me is that the original post here mentions adjusting `local_temperature_calibration` as a workaround makes the TRV update the current temperature reading, which doesn't work for me either. I have an automation in HA that would adjust the `local_temperature_calibration` every 5 minutes anyways using an external thermometer, so this is what I've basically been doing already for months. Curiously, updating `local_temperature_calibration` now lights up the TRVs display every time. It definitely didn't do that before, I would have noticed that at night and gone crazy.

I'll experiment a bit more and also plan to downgrade to pre-2.0 again to double-check when I have a bit more free time. Any additional comments/ideas welcome though.

thiscantbeserious, 2025-01-28:
Aaaah yea I remember now.

I long since switched to node-red for automation and have implemented a routine that does regular readouts via an mqtt command thats why its working for me.

You do have to send an update command to get it updated.

I'm currently not at home but will have a look later. I used a subflow for it.

thiscantbeserious, 2025-01-28:
Apparently I forgot to share my findings back in 2021 - wow ... I'm still using them as of today.

Here you go - you need to send this via MQTT to get the current local temperature reported back

every time you want to read it out you'll have to send it, for each thermostat.

`
zigbee2mqtt/your_thermostat_id/set: {
  "online":"ON"
}
`

srett, 2025-01-28:
Thanks! That does the trick! So the real mystery seems to be why it even worked at all for me before, without any hacks. Even before I added an automation to adjust the temp offset, the value would update two or three times an hour before the z2m update. Weird, but this is much better now anyways, I can get much more fine-grained data :)