Koenkk/zigbee2mqtt #9410: Danfoss Ally no reaction after update the external_measured_room_sensor

snp88, 2021-10-30:
## What happened
I update external temperature sensor value to trigger the heat bellow occupied_heating_setpoint
new value show in sensor field 
## What did you expect to happen
Valve should be openned and heat should start
## How to reproduce it (minimal and precise)
setup value of external_measured_room_sensor bellow occupied_heating_setpoint
[image]

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

Scalabanana, 2021-10-30:
I can reproduce this. Not sure though, if "local_temperature" is supposed to be overridden by external sensors value or if the TRV handles the temperature internally. Anyway valve position and "heat_required" are no changing, so I guess it really is an issue.

snp88, 2021-10-30:
the value should be overwritten by the value from the external sensor. otherwise I don't see the point.

rindlerblabla, 2021-10-30:
According to this, https://community.home-assistant.io/t/danfoss-ally-trv-working-with-remote-temp-sensor/276686/25 , Danfoss ally uses an average of the internal and an external sensor, so the value of the internal sensor is not overwritten.

rindlerblabla, 2021-10-30:
[image]
Like here, the external temperature is 20.11, local temperature 25.32, set point 22 and still the valve is 91% open.

Cherry-Dude, 2021-10-31:
> According to this, https://community.home-assistant.io/t/danfoss-ally-trv-working-with-remote-temp-sensor/276686/25 , Danfoss ally uses an average of the internal and an external sensor, so the value of the internal sensor is not overwritten.

That is certainly my experience, but perhaps others have found a workaround? Boy is it difficult to get a TRV that simply accepts external temperature....

Cherry-Dude, 2021-12-03:
There is an upcoming firmware which should add a radiator covered function (in relation to an also soon to be released external Danfoss sensor). This should allow the Ally to only use external sensor temps. Perhaps this can be incorporated?

closed 2022-01-12 by github-actions[bot]