Koenkk/zigbee2mqtt #9411: Setting window_open_external resetting immediately

Scalabanana, 2021-10-30:
## What happened
I tried to set window_open_external to "true" but another mqtt message immediately after sets it back to "false".
`Info MQTT publish: topic 'zigbee2mqtt/Thermostat Gästezimmer', payload '{"algorithm_scale_factor":5,"battery":65,"day_of_week":"tuesday","external_measured_room_sensor":1800,"heat_available":true,"heat_required":false,"keypad_lockout":"unlock","linkquality":93,"load_estimate":-1,"local_temperature":21.7,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":21,"pi_heating_demand":1,"running_state":"idle","setpoint_change_source":"externally","system_mode":"heat","thermostat_vertical_orientation":false,"trigger_time":1026,"update":{"state":"idle"},"update_available":false,"viewing_direction":false,"window_open_external":"true","window_open_internal":"quarantine"}'
Info MQTT publish: topic 'zigbee2mqtt/Thermostat Gästezimmer', payload '{"algorithm_scale_factor":5,"battery":65,"day_of_week":"tuesday","external_measured_room_sensor":1800,"heat_available":true,"heat_required":false,"keypad_lockout":"unlock","linkquality":96,"load_estimate":-1,"local_temperature":21.7,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":21,"pi_heating_demand":1,"running_state":"idle","setpoint_change_source":"externally","system_mode":"heat","thermostat_vertical_orientation":false,"trigger_time":1026,"update":{"state":"idle"},"update_available":false,"viewing_direction":false,"window_open_external":false,"window_open_internal":"quarantine"}'`
## What did you expect to happen
"window_open_external" set to true to change state of "window_open_internal".
## How to reproduce it (minimal and precise)
topic: zigbee2mqtt/FRIENDLY_NAME/set
payload: {"window_open_external": "true"}
## Debug info
Zigbee2MQTT version:
Adapter hardware: cc2652rb
Adapter firmware version: 20210120

robertalexa, 2021-11-03:
@Scalabanana I am unable to reproduce this at my end (I have added support for window_open_external). 

Just a few things I want to double check with you.

Changing this can take a few seconds, not necessarily instant, keep your eyes on the log in the meantime. This is because the internal PID of the TRV has to make deductions as to how it needs to behave as a response to your command

It is very possible that you get messages from the device that came as part of the reporting, coninciding you asking to set window_open_external true, but they might not old data before the TRV got a chance to apply your request. That is not to say that the device is not doing it.

Here is what I would suggest you try. Set window_open_external true. After that, interrogate it on the `get` topic and see what value you are getting.

window_open_internal will change to `open` when external is TRUE and back to `quarantine` when external gets set to FALSE. After this quarantine period it should go back to `closed`.

You can also try to use the Frontend as well do run the action. If the switch stays in the True position (doesn't flip back), your command was successful. You should also hear your TRV close (if the TRV was open before due to heating demand)

[image]

Log of me setting it to `FALSE`
```
MQTT publish: topic 'zigbee2mqtt/Rob's Office TRV', payload '{"algorithm_scale_factor":1,"battery":41,"day_of_week":"wednesday","external_measured_room_sensor":-8000,"heat_available":true,"heat_required":false,"keypad_lockout":"unlock","last_seen":1635933871185,"linkquality":63,"load_estimate":-8000,"local_temperature":20.42,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":22,"pi_heating_demand":0,"running_state":"idle","system_mode":"heat","thermostat_vertical_orientation":true,"trigger_time":660,"viewing_direction":1,"window_open_external":false,"window_open_internal":"external_open"}'
```

Log of me setting is to `TRUE` - not how many reporting commands arrived before the TRV reacted to window external open
```
 MQTT publish: topic 'zigbee2mqtt/Rob's Office TRV', payload '{"algorithm_scale_factor":1,"battery":41,"day_of_week":"wednesday","external_measured_room_sensor":-8000,"heat_available":true,"heat_required":true,"keypad_lockout":"unlock","last_seen":1635933932150,"linkquality":57,"load_estimate":-8000,"local_temperature":20.31,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":22,"pi_heating_demand":99,"running_state":"heat","system_mode":"heat","thermostat_vertical_orientation":true,"trigger_time":660,"viewing_direction":1,"window_open_external":false,"window_open_internal":"quarantine"}'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office TRV', payload '{"algorithm_scale_factor":1,"battery":41,"day_of_week":"wednesday","external_measured_room_sensor":-8000,"heat_available":true,"heat_required":true,"keypad_lockout":"unlock","last_seen":1635933983015,"linkquality":57,"load_estimate":-8000,"local_temperature":20.31,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":22,"pi_heating_demand":99,"running_state":"heat","system_mode":"heat","thermostat_vertical_orientation":true,"trigger_time":660,"viewing_direction":1,"window_open_external":true,"window_open_internal":"quarantine"}'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office TRV', payload '{"algorithm_scale_factor":1,"battery":41,"day_of_week":"wednesday","external_measured_room_sensor":-8000,"heat_available":true,"heat_required":true,"keypad_lockout":"unlock","last_seen":1635933991155,"linkquality":60,"load_estimate":-8000,"local_temperature":20.31,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":22,"pi_heating_demand":99,"running_state":"heat","system_mode":"heat","thermostat_vertical_orientation":true,"trigger_time":660,"viewing_direction":1,"window_open_external":true,"window_open_internal":"quarantine"}'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office TRV', payload '{"algorithm_scale_factor":1,"battery":41,"day_of_week":"wednesday","external_measured_room_sensor":-8000,"heat_available":true,"heat_required":true,"keypad_lockout":"unlock","last_seen":1635933991201,"linkquality":60,"load_estimate":-8000,"local_temperature":20.25,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":22,"pi_heating_demand":89,"running_state":"heat","system_mode":"heat","thermostat_vertical_orientation":true,"trigger_time":660,"viewing_direction":1,"window_open_external":true,"window_open_internal":"quarantine"}'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office TRV', payload '{"algorithm_scale_factor":1,"battery":41,"day_of_week":"wednesday","external_measured_room_sensor":-8000,"heat_available":true,"heat_required":true,"keypad_lockout":"unlock","last_seen":1635934051173,"linkquality":60,"load_estimate":869,"local_temperature":20.25,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":22,"pi_heating_demand":89,"running_state":"heat","system_mode":"heat","thermostat_vertical_orientation":true,"trigger_time":660,"viewing_direction":1,"window_open_external":true,"window_open_internal":"quarantine"}'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office TRV', payload '{"algorithm_scale_factor":1,"battery":41,"day_of_week":"wednesday","external_measured_room_sensor":-8000,"heat_available":true,"heat_required":true,"keypad_lockout":"unlock","last_seen":1635934051210,"linkquality":57,"load_ [... cut ...]
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office TRV', payload '{"algorithm_scale_factor":1,"battery":41,"day_of_week":"wednesday","external_measured_room_sensor":-8000,"heat_available":true,"heat_required":false,"keypad_lockout":"unlock","last_seen":1635934111253,"linkquality":60,"load_estimate":869,"local_temperature":20.15,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":22,"pi_heating_demand":0,"running_state":"idle","system_mode":"heat","thermostat_vertical_orientation":true,"trigger_time":660,"viewing_direction":1,"window_open_external":true,"window_open_internal":"open"}'
```

Scalabanana, 2021-11-03:
@robertalexa 
Thanks for your detailed reply. I will check it out. Perhaps I was too impatient.
Today I received 3 more of these TRVs so I can also try if this was a specific issue.
I will report back.

rindlerblabla, 2021-11-03:
I have also also pair of those Ally thermostats and mine behave just like @robertalexa is describing.

Scalabanana, 2021-11-03:
Well first of all thanks again for your advices. It's working like a charm, now!!!

Using the frontend kept the switch in the position I intended and the logs showed several reportings as you described. Using the switch in the device overview in homeassistant gave me the same results. So I tried to use the mqtt service again and I had the same issue as I stated above. I don't really know why.

I ended up not using the mqtt service but "turn_on" command in my automations to handle the switch as what it is... a switch. 
I think I was not considering this option as I was using some MOES and DECT TRVs before and they didn't have switch entities.

Well I am happy now and love those TRVs for their capability compared to others.

closed 2021-11-03 by Scalabanana

robertalexa, 2021-11-03:
@Scalabanana I am glad all is well.

yeah no need to use mqtt direct, when I worked on these TRVs my aim was to have everything directly functional from HA, for convenience.

I think I know why your mqtt command was failing, do you mind doing a test?
Instead of sending:
```
{"window_open_external": "true"}
```
Can you try...
```
{"window_open_external": true}
```

Thanks

Scalabanana, 2021-11-03:
> @Scalabanana I am glad all is well.
>
> yeah no need to use mqtt direct, when I worked on these TRVs my aim was to have everything directly functional from HA, for convenience.
>
> I think I know why your [...]

Yep that's it. 
Could have found this myself. I am always confused if I have to set anything in " " or not. I guess I tried it with quotation marks first, saw the switch moving on and off again and thought that must be correct. And the logs don't show a difference either. 😉

robertalexa, 2021-11-03:
It would be because the value is of type boolean (true/false) rather than string ("text", "true", "false"). Shame on me for not noticing earlier haha :)

Glad everything is good.

Scalabanana, 2021-11-03:
Haha, no worries.
Now I just have to figure out which algorithm_scale_factor is best for my usecase.
And what is trigger_time used for? Couldn't figure it out yet.

robertalexa, 2021-11-03:
`trigger_time` works in conjunction with the internal schedule (which is 100% not worth using), so you don't need to worry about it at all. I have exposed it for the sake of completion, not because it actually is useful.

As to the algorithm, here is what the specification says:
[image]

I am not entirely sure if it means what it seems to mean. That on top of the fact that Hive (Hive TRV is the same as Danfoss Ally) do not even mention such a feature, I have figured not to fiddle with it myself.

That being said, all my TRVs have scale factor set to 1, which is also the default value after factory reseting the devices. The TRVs react quick, and the battery life has been veeery good, despite the trvs doing multiple operations (in my daily schedule) over winter. So with this, my personal advice would be to not worry about it either :) Just enjoy your TRVs haha

Scalabanana, 2021-11-03:
Perfect, I won't worry about those.
Honestly these TRVs are sooo much better than those MOES I had before. They had some relly annoyingly slow behaviour of changing the valve position. Danfoss Ally sure is a higher price tag but so much more useful.