Koenkk/zigbee2mqtt #9416: Cable outlet connected LeGrand 064879

MysterePMS, 2021-10-30:
Hello,
I have just purchased "Cable outlet connected Legrand 064879"
This one has the following functions:
- ON / OFF
- Instantaneous power measurement and daily and monthly consumption monitoring.
- 4-order and 6-order pilot wire control.
According to my research, this one works under Deconz with the "mode": comfort / comfort-1 / comfort-2 / eco / frost protection / off. [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/883#issuecomment-562977413](url)

Thanks for all your hard work.

## Information about the device + link

1. [https://www.legrand.fr/pro/catalogue/43076-sortie-de-cable/sortie-de-cable-connectee-pour-installation-with-netatmo-compatible-fil-pilote-3000w-on-off-et-mesure-conso-blanc](url)

2. [https://zigbee.blakadder.com/Legrand_064879.html](url)

## data/database.db entry of the device
```
{"id":39,"type":"Router","ieeeAddr":"0x000474000011ea72","nwkAddr":27650,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" Cable outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1,"inClusterList":[0,3,4,6,5,64513,2820,64576],"outClusterList":[0,64513,25],"clusters":{"genBasic":{"attributes":{"modelId":" Cable outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":1,"zclVersion":2,"appVersion":0,"stackVersion":66,"hwVersion":1,"dateCode":" \u000020210617\u0000\u0000\u0000\u0000\u0000","swBuildId":"0008"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":66,"hwVersion":1,"dateCode":" \u000020210617\u0000\u0000\u0000\u0000\u0000","swBuildId":"0008","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1635602232443}
{"id":40,"type":"Router","ieeeAddr":"0x000474000011ea46","nwkAddr":22379,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" Cable outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1,"inClusterList":[0,3,4,6,5,64513,2820,64576],"outClusterList":[0,64513,25],"clusters":{"genBasic":{"attributes":{"modelId":" Cable outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":1,"zclVersion":2,"appVersion":0,"stackVersion":66,"hwVersion":1,"dateCode":" \u000020210617\u0000\u0000\u0000\u0000\u0000","swBuildId":"0008"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":66,"hwVersion":1,"dateCode":" \u000020210617\u0000\u0000\u0000\u0000\u0000","swBuildId":"0008","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1635604042481}
{"id":41,"type":"Router","ieeeAddr":"0x000474000011e96d","nwkAddr":25176,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" Cable outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1,"inClusterList":[0,3,4,6,5,64513,2820,64576],"outClusterList":[0,64513,25],"clusters":{"genBasic":{"attributes":{"modelId":" Cable outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":1,"zclVersion":2,"appVersion":0,"stackVersion":66,"hwVersion":1,"dateCode":" \u000020210617\u0000\u0000\u0000\u0000\u0000","swBuildId":"0008"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":66,"hwVersion":1,"dateCode":" \u000020210617\u0000\u0000\u0000\u0000\u0000","swBuildId":"0008","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1635608579867}
```

MysterePMS, 2021-11-03:
Hello,
Is this included in the new update or not yet?
cordially

MysterePMS, 2021-12-04:
will it be compatible one day?

silversmarty, 2022-01-05:
Hi @MysterePMS ,
I'm about to buy this device : did you succeed into making it work with z2m ? on Blackadder it says it's a standard zibee device.
In particular, do you manage power consumption?

Thanks

MysterePMS, 2022-01-05:
Hello Silversmarty,
No to date I was able to put this one on Z2M but it is not recognized.

closed 2022-01-13 by github-actions[bot]

FlyingDomotic, 2022-12-14:
Looks like being the same as Legrand 628618, but with only 16A instead of 30A. You may perhaps try to modify Legrand device code to create a new definition by copying 628618 one?