Koenkk/zigbee2mqtt #9426: Phillips Hue bulb not installing its firmware

tmugan, 2021-10-31:
Bug report?
- If applicable, provide steps how to reproduce the problem.
- Provide the herdsman debug logging: https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

-->

## What happened
My Philips Hue bulb detects an OTA upgrade on Z2M and seems to install it correctly but it continuously reports that there is an OTA available. 

This is what is being reported in Z2M...
Manufacturer | Model | Firmware build date | Firmware version
Philips | 8718696449691 | 20170908 | 1.29.0_r21169

I note from Phillips firmware release notes https://www.philips-hue.com/en-au/support/release-notes/lamps
that the Software version: 1.29.0_r21169 has a date stamp of November 14, 2017 (not the 8th Sept reported by my bulb)

When I check the log during the OTA enquiry, I see that there is indeed a fileversion difference being identified/reported...

Debug Device 'Dining Room Lamp 2' requested OTA
Debug Got OTA request '{"fieldControl":0,"manufacturerCode":4107,"imageType":268,"fileVersion":16781568}'
Debug Is new image available for '0x0017880104188122', current '{"fieldControl":0,"manufacturerCode":4107,"imageType":268,"fileVersion":16781568}', latest meta '{"fileVersion":16785664,"fileSize":267644,"url":"http://fds.dc1.philips.com/firmware/ZGB_100B_010C/16785664/100B-010C-01002100-ConfLight-Lamps_0012.zigbee","sha512":"efa1c94c60896468624c3f699bc2a1986c4667e5fe6512ecd7db722799be898f634144825035280e62becda576d8571fc22a421e553adc57b454f099c479fa93"}'
Debug Update available for '0x0017880104188122': YES

## What did you expect to happen

Upon finishing the update, it seems to be successful but the  veriosn do not change and the process will repeat if I check OTA updates again.  I have tried many times with the same outcome each time.
I've also tried after a factory reset of the Philips Hue bulb (with the Alexa Plus and using the serial number method through the Alexa Android app).

## How to reproduce it (minimal and precise)

It seems to be stuck because the firmware being installed is not updating the file version correctly and so it keeps incorrectly thinking it needs to get the same firmware each time.

## Debug info

Adapter hardware: CC2531

Zigbee2MQTT version
1.21.2 commit: unknown
Coordinator type
zStack12
Coordinator revision
20180507
Frontend version
0.6.12

tmugan, 2021-11-06:
If anyone has any ideas what is going on here, I'd appreciate any advice or direction.

I'm guessing that the either Phillips site is misreporting their firmware fileversion or that Z2M is somehow retaining the old fileversion data.

Nauttor, 2021-11-08:
Hi tmugan, 

I can confirm I'm experiencing exactly the same behaviour when trying to upgrade my Philip Led Stripes. 

Find attached the log after the install is completed. The upgrade was completed, but if you see the log, it was updates from the old version to the same version, very strange. 

zigbee2mqtt/bridge/logging {"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{\"data\":{\"from\":{\"date_code\":\"20170323\",\"software_build_id\":\"1.24.0_r20239\"},\"id\":\"TiraLedCocina\",\"to\":{\"date_code\":\"20170323\",\"software_build_id\":\"1.24.0_r20239\"}},\"status\":\"ok\",\"transaction\":\"dlkds-1\"}'"}

The same happen, every time I check for OTA, Z2M says there is a new version. BTW I'm using the last version 1.22.

Other weird not related issue I'm having is that the webUI buttons on the dashboard view is not responsive, I need to reload the page to see the change of status when i click any action like ON/OFF.

Nauttor, 2021-11-08:
@tmugan I think you need to take a look at #8757 as is the same issue we are experiencing. Apparently hue devices require a stepped upgrade if the version is too old.

tmugan, 2021-11-09:
> @tmugan I think you need to take a look at #8757 as is the same issue we are experiencing. Apparently hue devices require a stepped upgrade if the version is too old.

Thanks @Nauttor That does look like exactly the issue.  My option there seems to be to install a totally separate older version of Z2M and Mosquitto on my macbook, temporarily borrow the Zigbee USB adaptor and then try to install an older version of the Hue bulb firmware.  

I appreciate the tip off.

Is this likely to be included in the main codebase in future, I wonder?

closed 2021-12-17 by github-actions[bot]