Koenkk/zigbee2mqtt #9427: Error while installing

RCraveiro, 2021-10-31:
I've searched for this issue i cannot find any help.
I'm trying to install the zig2bee on Win10 machine, when i run "npm ci" i get this errors:

`PS E:\Pessoal\zigbee2mqtt> npm ci
npm WARN old lockfile
npm WARN old lockfile The npm-shrinkwrap.json file was created with an old version of npm,
npm WARN old lockfile so supplemental metadata must be fetched from the registry.
npm WARN old lockfile
npm WARN old lockfile This is a one-time fix-up, please be patient...
npm WARN old lockfile
npm ERR! code 1
npm ERR! path E:\Pessoal\zigbee2mqtt\node_modules\zigbee-herdsman\node_modules\@serialport\bindings
npm ERR! command failed
npm ERR! command C:\WINDOWS\system32\cmd.exe /d /s /c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
npm ERR! Building the projects in this solution one at a time. To enable parallel build, please add the "-m" switch.
npm ERR!   serialport.cpp
npm ERR! C:\Users\RCraveiro\AppData\Local\node-gyp\Cache\16.13.0\include\node\node.h(63,10): fatal error C1083: Cannot open include file: 'v8.h': No such file or directory [E:\Pessoal\zigbee2mqtt\node_modules\zigbee-herdsman\node_modules\@serialport\bindings\build\bindings.vcxproj]
npm ERR! prebuild-install WARN install No prebuilt binaries found (target=16.13.0 runtime=node arch=x64 libc= platform=win32)
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@8.2.0
npm ERR! gyp info using node@16.13.0 | win32 | x64
npm ERR! gyp info find Python using Python version 3.9.7 found at "C:\Users\RCraveiro\AppData\Local\Programs\Python\Python39\python.exe"
npm ERR! gyp info find VS using VS2019 (16.11.31624.102) found at:
npm ERR! gyp info find VS "C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise"
npm ERR! gyp info find VS run with --verbose for detailed information
npm ERR! gyp info spawn C:\Users\RCraveiro\AppData\Local\Programs\Python\Python39\python.exe
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   'C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\gyp\\gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'msvs',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   'E:\\Pessoal\\zigbee2mqtt\\node_modules\\zigbee-herdsman\\node_modules\\@serialport\\bindings\\build\\config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   'C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   'C:\\Users\\RCraveiro\\AppData\\Local\\node-gyp\\Cache\\16.13.0\\include\\node\\common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=C:\\Users\\RCraveiro\\AppData\\Local\\node-gyp\\Cache\\16.13.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=C:\\\\Users\\\\RCraveiro\\\\AppData\\\\Local\\\\node-gyp\\\\Cache\\\\16.13.0\\\\<(target_arch)\\\\node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=E:\\Pessoal\\zigbee2mqtt\\node_modules\\zigbee-herdsman\\node_modules\\@serialport\\bindings',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'E:\\Pessoal\\zigbee2mqtt\\node_modules\\zigbee-herdsman\\node_modules\\@serialport\\bindings\\build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\MSBuild\Current\Bin\MSBuild.exe
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   'build/binding.sln',
npm ERR! gyp info spawn args   '/clp:Verbosity=minimal',
npm ERR! gyp info spawn args   '/nologo',
npm ERR! gyp info spawn args   '/p:Configuration=Release;Platform=x64'
npm ERR! gyp info spawn args ]
npm ERR! gyp ERR! build error
npm ERR! gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\MSBuild\Current\Bin\MSBuild.exe` failed with exit code: 1
npm ERR! gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:194:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:390:28)
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)
npm ERR! gyp ERR! System Windows_NT 10.0.19043
npm ERR! gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd E:\Pessoal\zigbee2mqtt\node_modules\zigbee-herdsman\node_modules\@serialport\bindings
npm ERR! gyp ERR! node -v v16.13.0
npm ERR! gyp ERR! node-gyp -v v8.2.0
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in:`

Log file can be found here https://pastebin.com/mxL0RaQi.
I have python 3.9 and node 16.13

Thanks for the help

Faustino-ai, 2021-11-02:
Ya I am having the same issue as well.

Faustino-ai, 2021-11-02:
Actually, by downgrading my node version to 14.xx, this error is gone. However it was met with another error when I tried to run npm start.

closed 2021-12-10 by github-actions[bot]

Jon811, 2022-03-05:
Every time i run update there is:
```
npm WARN old lockfile
npm WARN old lockfile The npm-shrinkwrap.json file was created with an old version of npm,
npm WARN old lockfile so supplemental metadata must be fetched from the registry.
npm WARN old lockfile
npm WARN old lockfile This is a one-time fix-up, please be patient...
npm WARN old lockfile

```
It takes a lot time, then looks like running, but there are problems with permissions. Only after `chmod -R 777 /opt/zigbee2mqtt` it works fine.