Koenkk/zigbee2mqtt #9428: Philips Hue GU10 White Ambiance 5047131P6 / 9290019533

Fluke242, 2021-10-31:
Hi, I'm new to z2m and got several products already connected. Unfortunately, above mentioned GU10 is marked "unsupported". I'v already found instructions about the mentioned 9290019533 but that's referring to 5309030P9 not mine being  5047131P6: https://githubmemory.com/repo/Koenkk/zigbee2mqtt/issues/8502
Any efforts by resetting the bulb, updating FW via direct-connect via BT and trying to adjust above mentioned instructions to 5047131P6 didn't help.

## Information about the device + link
..

## data/database.db entry of the device
..

danieledwardgeorgehitchcock, 2021-11-01:
Hi,

I have added your device fingerprint to the converters. Once the above PR is merged, you can switch to the Dev / Edge branch for device support or wait for a month for the release to Production

closed 2021-11-01 by Koenkk

Fluke242, 2021-11-02:
Thanks all!