Koenkk/zigbee2mqtt #9437: Philips motion sensor model SML003 configure error

bobreim, 2021-11-01:
I am trying to add support for the new model of the Philips motion sensor (SML003).
Using the configuration from SML002 seems to work except for the configuration. It gives the following error:

MQTT:info  2021-11-01 09:31:46: Successfully configured '<device_id>'
Zigbee2MQTT:error 2021-11-01 09:31:46: Failed to configure '<device_id>', attempt 3 (TypeError: Cannot read property 'configureKey' of undefined
    at Configure.configure (/app/lib/extension/configure.js:138:69))

Adding the key meta with value { configureKey: 1} to the definition works. But I am not sure why and whether this the right approach.

danieledwardgeorgehitchcock, 2021-11-01:
The configureKey shouldn't need to be added as it is done automatically.

Can you share your external converter / device code and database.db entry?

bobreim, 2021-11-01:
entry in database.db:

`{"id":5,"type":"EndDevice","ieeeAddr":"0x001788010bcc7ec5","nwkAddr":17493,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Battery","modelId":"SML003","epList":[2],"endpoints":{"2":{"profId":260,"epId":2,"devId":263,"inClusterList":[0,1,3,1030,1024,1026],"outClusterList":[25,0,3,6],"clusters":{"genBasic":{"attributes":{"modelId":"SML003","manufacturerName":"Signify Netherlands B.V.","powerSource":3,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20210830","swBuildId":"2.53.6"}},"genPowerCfg":{"attributes":{"61441":3205,"61442":3177,"61443":3207,"batteryPercentageRemaining":200}},"msOccupancySensing":{"attributes":{"48":2,"64":2,"65":6,"66":0,"67":6,"68":3,"occupancy":0,"pirOToUDelay":0}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":22115}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2056}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e332c2","endpointID":1},{"cluster":1024,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e332c2","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e332c2","endpointID":1},{"cluster":1030,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e332c2","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1030,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100},{"cluster":1024,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":5}],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20210830","swBuildId":"2.53.6","zclVersion":2,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1635756582177}`

external converter 9290030675.js:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['SML003'],
    model: '9290030675',
    vendor: 'Philips',
    description: 'Hue motion sensor',
    fromZigbee: [fz.battery, fz.occupancy, fz.temperature, fz.occupancy_timeout, fz.illuminance,
        fz.hue_motion_sensitivity, fz.hue_motion_led_indication],
    exposes: [e.temperature(), e.occupancy(), e.battery(), e.illuminance_lux(), e.illuminance(),
        exposes.enum('motion_sensitivity', ea.ALL, ['low', 'medium', 'high']),
        exposes.binary('led_indication', ea.ALL, true, false).withDescription('Blink green LED on motion detection'),
        exposes.numeric('occupancy_timeout', ea.ALL).withUnit('second').withValueMin(0).withValueMax(65535)],
    toZigbee: [tz.occupancy_timeout, tz.hue_motion_sensitivity, tz.hue_motion_led_indication],
    endpoint: (device) => {
        return {
            'default': 2, // default
            'ep1': 1,
            'ep2': 2, // e.g. for write to msOccupancySensing
        };
    },
    meta: { configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(2);
        const binds = ['genPowerCfg', 'msIlluminanceMeasurement', 'msTemperatureMeasurement', 'msOccupancySensing'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.batteryPercentageRemaining(endpoint);
        await reporting.occupancy(endpoint);
        await reporting.temperature(endpoint);
        await reporting.illuminance(endpoint);
        // read occupancy_timeout and motion_sensitivity
        await endpoint.read('msOccupancySensing', ['pirOToUDelay']);
        await endpoint.read('msOccupancySensing', [48], {manufacturerCode: 4107});
    }
};

module.exports = definition;`

closed 2021-12-09 by github-actions[bot]