Koenkk/zigbee2mqtt #9439: Ikea TRADFRI open/close remote (E1766) not working after firmware update

Heisenberg2980, 2021-11-01:
## What happened
After updating the firmware of the Ikea TRADFRI open/close remote (E1766) the device doesn´t work anymore. The pairing process works and zigbee2mqtt shows the device is in the network but the button doesn´t work. I have sniffed the traffic and after the pairing process is complete, no more messages are received from the device even after pressing any of the buttons (up or down).

## What did you expect to happen
Device should send open/close commands when pressing the buttons. It was working fine before updating the firmware.

## How to reproduce it (minimal and precise)
Update device firmware and repair the device

## Debug info
Zigbee2MQTT version: 1.21.2-1
Adapter hardware: CC2652R 
Adapter firmware version: Z-Stack_3.x.0

sjorge, 2021-11-01:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/7716 it's a new issue with the latest firmware sadly.

Heisenberg2980, 2021-11-01:
> Duplicate of #7716 it's a new issue with the latest firmware sadly.

Thanks @sjorge, before opening this issue I searched for it but I couldn´t find it because #7716 doesn´t mention this device (E1766). I guess this issue is affecting a lot (or maybe all) of Ikea devices.

I will close this issue to keep track of the updates in #7716

closed 2021-11-01 by Heisenberg2980

eloo, 2021-11-02:
@Heisenberg2980 nice workaround.
i can confirm that i was able to get a E1812 working with your hints. 
it took one or 3 tries but it seems to work properly!

Thanks a lot!

Theoreet, 2021-11-03:
> {"from": "<device friendly name>", "to": "coordinator"}

tried this serveral time, but i wont bind:
`{
  "message": {
    "cluster": "genOnOff",
    "cluster": "genLevelCtrl",
    "from": "1Knop",
    "to": "Coordinator"
  },
  "type": "device_bind_failed"`

I tried pressing the shortcut button ( E1812) every second, but stil this error. 

The log error tells me: 
Failed to bind cluster 'genOnOff' from '1Knop' to 'Coordinator' (Error: Bind 0x0c4314fffeacac43/1 genOnOff from '0x00124b002291d816/1' failed (AREQ - ZDO - bindRsp after 10000ms))

Z2M is reccieving the battery and linkquality, but the action is "null" and stays there

Theoreet, 2021-11-03:
> > {"from": "", "to": "coordinator"}
> 
> tried this serveral time, but i wont bind: `{ "message": { "cluster": "genOnOff", "cluster": "genLevelCtrl", "from": "1Knop", "to": "Coordinator" }, "type": "device_bind_failed"`
> 
> I tried pressing the shortcut button ( E1812) every second, but stil this error.
> 
> The log error tells me: `Failed to bind cluster 'genOnOff' from '1Knop' to 'Coordinator' (Error: Bind 0x0c4314fffeacac43/1 genOnOff from '0x00124b002291d816/1' failed (AREQ - ZDO - bindRsp after 10000ms))`
> 
> Z2M is reccieving the battery and linkquality, but the action is "null" and stays there

Tried it 5times, and it's working when you use this  payload {"from": "<device friendly name>", "to": "coordinator"} directly when it enters the network ... 

I'am happy

eloo, 2021-11-03:
Maybe this workaround should be added to the documentation somewhere?

JXGA, 2021-11-07:
> tried this serveral time, but i wont bind: `{ "message": { "cluster": "genOnOff", "cluster": "genLevelCtrl", "from": "1Knop", "to": "Coordinator" }, "type": "device_bind_failed"`
>
> I tried pressing [...]

Did you try to do this after removing the battery? The device could be in a sleep mode, same happened to mine. 

Thanks to @Heisenberg2980 for the assist!

r4ian, 2021-11-09:
@Heisenberg2980 Wish I found your workaround earlier. I've unpaired it (E1812) and now it doesn't even wanna pair again. It just won't show up in the logs, won't start the interview. 

Maybe you have any idea how to unf*ck this. (also tried restoring an older backup but it just won't show up).

r4ian, 2021-11-09:
Also found something weird with this button thingy, when I long-press the pairing button of it, the also IKEA lightbulb turns on and starts to pulse. Is this a feature ? 😮

Heisenberg2980, 2021-11-09:
@r4ian the first thing I would try is replacing the battery. At least in my case after so many trial and error the battery was dead, it had enough energy to make the led blink (what confused me even more) but not enough energy to start the pairing process.

I just saw your next message, if other device like the Ikea bulb react to the pairing you could try to unplug every other Ikea device you have as the button might be finding and pairing with the Ikea device before it discovers your coordinator.

McMornan, 2021-11-30:
> Just in case anyone is having the same issue, a workaround to make the E1766 to send messages again to z2m is to bind it to the coordinator after pairing by sending the mqtt message `zigbee2mqtt/bridge/request/device/bind` with payload `{"from": "<device friendly name>", "to": "coordinator"}`

You made my day ! After this months long discussion I was just about to throw them into the thrash ... what shall I say ? It works :)

darekpro, 2021-11-30:
> Just in case anyone is having the same issue, a workaround to make the E1766 to send messages again to z2m is to bind it to the coordinator after pairing by sending the mqtt message `zigbee2mqtt/bridge/request/device/bind` with payload `{"from": "<device friendly name>", "to": "coordinator"}`

Someone can explain me how to this? thank you.

kovacsbalu, 2021-12-10:
@darekpro use an mqtt publisher like mosquitto_pub
`mosquitto_pub -t zigbee2mqtt/bridge/request/device/bind -m '{"from": "<device friendly name>", "to": "Coordinator"}'`

For me the coordinator name starts with capital 'C'

cameronjust, 2022-04-22:
Thanks to all for posting this fix. Two of my three E1812s dropped off my network.

Can confirm this fix works and as with @kovacsbalu my coordinator starts with a capital C as well.

Do we know what causes it yet and maybe how to avoid it in the future?