Koenkk/zigbee2mqtt #9440: Can't add "devices:" to fix the "Deprecated click event." Xiaomi button WXKG11LM

zweetkonijn, 2021-11-01:
## What happened
I tried to add "devices" to the configuartion.yaml file as stated on this page: [Xiaomi WXKG11LM](https://www.zigbee2mqtt.io/devices/WXKG11LM.html)

But the configuration validation gives me the following error: 
`Integration error: devices - Integration 'devices' not found.`

ybychenok, 2021-11-01:
You need to add that to `devices.yaml`, located under `zigbee2mqtt/devices.yaml`

zweetkonijn, 2021-11-01:
ybychenok's reply fixed my problem.

closed 2021-11-01 by zweetkonijn

dannykellett, 2022-02-28:
Thanks but I think it should specify. It's infers that the update should go in configuration
Same here: https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration