Koenkk/zigbee2mqtt #9441: After upgrading to 1.21.2 from 1.17.2 zigbee2mqtt will not start due to pan id collision

tigertail100, 2021-11-01:
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

euzu, 2021-11-02:
I had the same problem after an update. I had to disconnect all sockets (zigbee repeater) from power. Only then I could restart zigbee2mqtt. After that I plugged the sockets back in and everything worked.

bartplessers, 2021-11-02:
Had same problem one.
For me what helped: wrapping the zigbee stick in alu foil so no interference with existing mesh was possible. Z2m started, than I removed foil and everything worked as designed

tigertail100, 2021-11-02:
Hi Bart, thank you for your suggestion, I built a fresh z2m on a nuc using
the same CC2538 and it worked, however only if I  used port:
/dev/serial.by-id/xxxxxxxxxxxxxxxxxxxx
I then used the same method with the Pi4 z2m and it resolved it .

Regards

Ian

closed 2021-12-10 by github-actions[bot]