Koenkk/zigbee2mqtt #9447: Support for lumi.curtain.acn002 - Aqara Roller Shade Driver E1 (Europe-version?)

wisekki, 2021-11-01:
Finally I received the Aqara Roller Shade Driver E1, but it seems this one doesn't work out of the box with zigbee2mqtt.
I'm using zigbee2mqtt 1.22.0-dev. The device joins the network and is interviewed as:

`ZNJLBL01LM - Xiaomi Aqara roller shade companion E1 (EndDevice)`

Problem is that it doesn't react to anything and keeps flooding the logs:

`'{"battery":null,"last_seen":"2021-11-01T20:07:44+02:00","linkquality":48,"position":0,"running":false,"update_available":null}'`

Only thing that changes is the link quality, nothing else.

On the device back it says Model: RSD-M01 and it's manufactured last month 10/2021. So I think this would be a new device perhaps but it joins as ZNJLBL01LM. What can I do?

When googling RSD-M01 the first result goes to Aqara's Europe site of Roller Shade Driver E1.

## Information about the device + link
Aqara Roller Shade Driver E1
Model: RSD-M01
[https://www.aqara.com/eu/product/roller-shade-driver-e1](url)

## data/database.db entry of the device
> {"id":12,"type":"EndDevice","ieeeAddr":"0x54ef4410002e066a","nwkAddr":62470,"manufId":4447,"manufName":"LUMI","powerSource":"Unknown","modelId":"lumi.curtain.acn002","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2,3,4,5,6,9,13,19,258],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.curtain.acn002","manufacturerName":"LUMI","powerSource":0,"zclVersion":3,"appVersion":25,"stackVersion":2,"hwVersion":1,"dateCode":"05-18-2021"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":25,"stackVersion":2,"hwVersion":1,"dateCode":"05-18-2021","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1635787960439}

AndyKassell, 2021-11-05:
Hi,

I have same issue with same device. It is a supported device but it just doesn't work correctly. I have debug logs turned on also and see it does an OTA request, but yes it is constantly sending this in.

```
Info MQTT publish: topic 'zigbee2mqtt/Bed 3 Blind', payload 
'{"battery":null,"last_seen":1636097972251,"linkquality":54,"position":20,"running":false,"update_available":null}'
Debug Received Zigbee message from 'Bed 3 Blind', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":3608,"imageType":14976,"manufacturerCode":4447}' from endpoint 1 with groupID 0
Debug Device 'Bed 3 Blind' requested OTA
```
Whilst the device does not report battery or current position correctly I can control the blind using the controls, however they do not update to reflect the current position you need to do a manual refresh. I have linked a video review below which shows the exact issue starting for the 8 minute mark. 

[Aqara Blind Driver Unbox + Setup in Home Assistant without a hub](https://youtu.be/pnkJZoIHZwc?t=481)

SchoutenHans, 2021-11-05:
Hi,

I have same issue with this device. (Model: RSD-M01). Also same problems mentioned by Andy Kassell.

HenriMatthijssen, 2021-11-08:
Same here. It would also be nice to see the battery status.

Koenkk, 2021-11-08:
Similar to the E1 contact sensor, it might be that this sensor requires a small change in the coordinator firmware to work:

CC2531/CC2530 fws: https://github.com/Koenkk/zigbee2mqtt/issues/9274#issuecomment-959505298
CC2652/CC1352: https://github.com/Koenkk/zigbee2mqtt/issues/9274#issuecomment-958146099

daenny, 2021-11-10:
Thanks @Koenkk! I had the same problem.

I just tested with latest 1.22.1 HassIO addon and your patch and it works!

[image]

Except battery reporting is not working. Do you have any idea how to fix that?

Koenkk, 2021-11-11:
Could you check if battery works with the following report:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.curtain.acn002'],
    model: 'ZNJLBL01LM',
    description: 'Aqara roller shade companion E1',
    vendor: 'Xiaomi',
    fromZigbee: [fz.xiaomi_curtain_position, fz.battery, fz.cover_position_tilt, fz.ignore_basic_report, fz.xiaomi_curtain_options, fz.aqara_opple_report],
    toZigbee: [tz.xiaomi_curtain_position_state, tz.xiaomi_curtain_options],
    exposes: [e.cover_position().setAccess('state', ea.ALL), e.battery()],
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

See https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#instructions how to use it, make sure you are on z2m 1.22.0

daenny, 2021-11-11:
Thanks for your help! I get the following issue:

```
[23:00:48] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.0 start
> node index.js
(node:1445) UnhandledPromiseRejectionWarning: ReferenceError: ota is not defined
(Use `node --trace-warnings ...` to show where the warning was created)
(node:1445) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:1445) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

I went on and removed the line with `ota` and I got some state update with battery :tada: :
```
Zigbee2MQTT:info  2021-11-11 23:03:22: MQTT publish: topic 'zigbee2mqtt/Entrance Roller Rechts', payload '{"battery":0,"last_seen":"2021-11-11T22:03:22.425Z","linkquality":102,"position":100,"running":false,"voltage":1297}
```

And the state is shown as:
```
{
    "battery": 0,
    "last_seen": "2021-11-11T22:03:22.425Z",
    "linkquality": 102,
    "position": 100,
    "running": false,
    "voltage": 1297
}
```

However, I think the battery which is included is a LiPO, which should have ~3.7V-4.2V. But it's definitely an improvement.

Koenkk, 2021-11-12:
And with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.curtain.acn002'],
    model: 'ZNJLBL01LM',
    description: 'Aqara roller shade companion E1',
    vendor: 'Xiaomi',
    fromZigbee: [fz.xiaomi_curtain_position, fz.battery, fz.cover_position_tilt, fz.ignore_basic_report, fz.xiaomi_curtain_options],
    toZigbee: [tz.xiaomi_curtain_position_state, tz.xiaomi_curtain_options],
    exposes: [e.cover_position().setAccess('state', ea.ALL), e.battery()],
    ota: ota.zigbeeOTA,
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
    },
};

module.exports = definition;
```

??

HenriMatthijssen, 2021-11-13:
Tried it out with patched firmware for CC2652R and above change and this is definitely an improvement. Progress is instantly shown now. However I have inverted my roller and this is not taken into account yet (position should be (100 - position)).

Battery still shows N/A. I also see battery not reported in the logging:

`info  2021-11-13 13:19:31: MQTT publish: topic 'zigbee2mqtt/Rolgordijn Keuken', payload '{"linkquality":99,"position":86,"running":false}'
`

Koenkk, 2021-11-14:
@HenriMatthijssen 

- Does it show battery?
- How did you invert the roller? (through `configuration.yaml`?)

HenriMatthijssen, 2021-11-14:
See above, It shows N/A for Battery.

I have inverted the roller on the device itself (pressing reset button 3 times in a row).

Koenkk, 2021-11-14:
@HenriMatthijssen 

- does battery reporting work with https://github.com/Koenkk/zigbee2mqtt/issues/9447#issuecomment-966546707 ?
- you have to make z2m aware that your covers is inverted, see the invert_cover option: https://www.zigbee2mqtt.io/devices/ZNJLBL01LM.html#options

SchoutenHans, 2021-11-14:
@Koenkk ,
I have two Aqara Roler Shade drivers E1.  I'm running the docker-compose version 1.22 on a Raspberry Pi 4.
Also I  installed the CC2652/CC1352: #9274 (comment) Firmware.

My Rolers working fine.  The position is updated after you modify the position.
The battery value is in the payload. 
Zigbee2MQTT:info  2021-11-14 17:47:28: MQTT publish: topic 'zigbee2mqtt/Blind-3', payload '{"battery":0,"last_seen":1636908448351,"linkquality":72,"position":0,"running":false,"voltage":1297}'

The voltage reported is 1297mv.  The Battery stay's at 0% after more that 24 hours.
State
{
    "battery": 0,
    "last_seen": 1636908221702,
    "linkquality": 75,
    "position": 0,
    "running": false,
    "voltage": 1298
}
I don't now if the voltage is ok. I read some ware that the device has a 7,4V 1000ma battery.
So, it working, with the 1.22 docker compose version and the firmware update of the CC2652 Slaesh stick

Regards,
Hans Schouten

daenny, 2021-11-14:
Yes, I think something is different in the voltage reporting. It's a 7.4V two cell lipo (according to https://notenoughtech.com/home-automation/aqara-roller-shade-driver-e1-zigbee2mqtt/). I also get voltage reading in the range of `1260-1330` at the moment with:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.curtain.acn002'],
    model: 'ZNJLBL01LM',
    description: 'Aqara roller shade companion E1',
    vendor: 'Xiaomi',
    fromZigbee: [fz.xiaomi_curtain_position, fz.battery, fz.cover_position_tilt, fz.ignore_basic_report, fz.xiaomi_curtain_options, fz.aqara_opple_report],
    toZigbee: [tz.xiaomi_curtain_position_state, tz.xiaomi_curtain_options],
    exposes: [e.cover_position().setAccess('state', ea.ALL), e.battery(), e.battery_voltage()],
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

The value does update and change. Battery stays at `0`. which makes sense, as all other battery templates expects at least a 3V battery if I read (https://github.com/Koenkk/zigbee-herdsman-converters/blob/cc45f434e7b867b404a7c2c4c005925b19163427/lib/utils.js#L132) correctly.

HenriMatthijssen, 2021-11-14:
Using above from @daenny I now get Voltage reporting. Further with setting invert_cover option the position seems reported fine now. 

I get for state now:
```
{
    "battery": 0,
    "linkquality": 93,
    "position": 0,
    "running": false,
    "voltage": 1301
}
```

However battery percentage is still not correctly shown:

Koenkk, 2021-11-15:
Then the only thing left here is to define a calculation to convert the `voltage` to `battery` (%). This is done by zigbee2mqtt here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/8f2aed87bdfd3215de72cd10e761dd2e8504ae77/converters/fromZigbee.js#L5534

Which boils down to:
- Figure out the highest and lowest possible `voltage`
- Define a new option/calculation in https://github.com/Koenkk/zigbee-herdsman-converters/blob/8f2aed87bdfd3215de72cd10e761dd2e8504ae77/lib/utils.js#L132
- Change https://github.com/Koenkk/zigbee-herdsman-converters/blob/8f2aed87bdfd3215de72cd10e761dd2e8504ae77/converters/fromZigbee.js#L5534 to `return {battery: batteryVoltageToPercentage(voltage, model.model === 'ZNJLBL01LM' ? 'NEW_OPTION_NAME' : '3V_2100'), voltage: voltage};`

wisekki, 2021-11-19:
Thanks for the great work! I've been sick and wasn't able to participate. After a lot of repairing I was finally able to actually move the roller shades up and down :) but that's about it. So what do I need to do in order to get it work better? Flash the slaesh's stick? And modify scripts?

redphx, 2021-11-22:
I have the Aqara Hub. Aqara app shows the battery at 55%. I tried to use the code above to get the voltage value but it didn't work. Z2M loaded the custom script but the exposed values are still the same (no voltage property, battery is `null`). I'm new to Z2M so not sure if I did anything wrong.

daenny, 2021-11-28:
I charged my two motor drivers to 100% toady. But the battery voltage is still reported as 1298-1300. So I am afraid that there is some more work left to figure out how to get a proper reading (voltage or percentage).

Gortosch, 2021-12-11:
Hello all,
I also have the problem that no battery status is displayed. Everything else works fine. Thanks for this. 
Any news on the battery problem yet? 

Is there anything I can do to help?

Greetings
Sven

fewdji, 2021-12-25:
Hi! Battery state not working. Got error
`ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: access to attribute 'update' of 'dict' object is unsafe. (value: {"battery":null,"linkquality":69,"position":0,"running":false,"update_available":null}, template: {{ value_json['update']['state'] == "available" }})`

totalitarian, 2022-01-05:
Same issue here, all works except battery

jeroenvdd, 2022-01-05:
> Same issue here, all works except battery

what is your firmware date is it 
05-18-2021 ? for the roller

Gortosch, 2022-01-05:
Hello,

i have the same issue. My firmware version is 05-18-2021.
[image]

Is there anything I can do to help?

Greetings
Sven

totalitarian, 2022-01-05:
> > Same issue here, all works except battery
> 
> what is your firmware date is it 05-18-2021 ? for the roller

05-18-2021

EloB, 2022-01-05:
I also have problem with battery. Apart from that I can Open/Close the blinder but I've lost the ability to set position. It was working and I had it connected to HB and was showing of that position feature and suddenly it stopped working. Even if I remove the device zigbee2mqtt and repair it doesn't reset that functionality...

jb123567, 2022-01-21:
I have two of these, seem to working well, however like others battery is just displaying a question mark. Anything I can do to help? Realistically these devices are the only zigbee devices I need to regularly charge, so it would be great to know when they need it.

invis-z, 2022-01-25:
With a lot of trial and error, I was able to find another possible place where the battery percentage is provided. It is in the aqaraOpple cluster. Attribute 1033(0x0409) is the charging state, 1034 is (presumably) battery percentage, I am 90% convinced that 1034 is battery but it would be great if someone can check/prove it with official app and hub. 

I was able to expose the data, but on the dashboard, the power column is still showing a question mark. I don't have much experience tweaking underhood stuff of z2m so I don't really know how to fix that. 

Also, some other stuff found during this process. 1026 is `curtain-range-flag` 1032 is `curtian-motor-speed`.
[Referencing this page](https://home.miot-spec.com/spec?type=urn%3Amiot-spec-v2%3Adevice%3Acurtain%3A0000A00C%3Alumi-acn002%3A1)

P.S. The page above shows the charging state 2 is not chargable, I am understanding it as battery full, and is the only state I had every seen while not charging(I got the device just for about 3-4 days). So, maybe check if anyone got a 1 (not charging)

Also, the attribute 247 includes a lot of the same information, position 101 seems to be battery. (This might help with what I am talking about: https://github.com/Koenkk/zigbee-herdsman-converters/blob/8a29cd0f2a9bf5243fdfe1623a647b612b66dd52/converters/fromZigbee.js#L5132 )

I don't really know how to add another option to the device. But apart from #10971 , most of the functionalities should be accessible with this new information. 

Below is a very rough version of the converter that I am using, it still needs a lot of work, try to just use it for reference.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const herdsman = require('zigbee-herdsman');
const manufacturerOptions = {
    xiaomi: { manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true },
};

const getKey = (object, value) => {
    for (const key in object) {
        if (object[key] == value) return key;
    }
};

const fzLocal = {
    lumi_curtain_acn002_power: {
        cluster: 'aqaraOpple',
        type: ['readResponse', 'attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            const data = msg.data;
            const lookup = {
                0: 'not_charging',
                1: 'charging',
                2: 'fully_charged',
                3: 'what',
            };
            if (data && data.hasOwnProperty('1033')) {
                const value = data['1033'];
                return {
                    charging_state: lookup[value],
                };
            }
            if (data && data.hasOwnProperty('1034')) {
                const value = data['1034'];
                return {
                    battery: value,
                };
            }

        },
    },
    lumi_curtain_acn002_options: {
        cluster: 'aqaraOpple',
        type: ['readResponse', 'attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            const data = msg.data;
            const lookup = {
                0: 'low',
                1: 'mid',
[... 155 lines cut ...]
    //     },
    // },
}

const definition = {
    zigbeeModel: ['lumi.curtain.acn002'],
    model: 'ZNJLBL01LM',
    description: 'Aqara roller shade companion E1',
    vendor: 'Xiaomi',
    fromZigbee: [fzLocal.lumi_curtain_acn002_power, fz.xiaomi_curtain_position, fz.cover_position_tilt, fz.ignore_basic_report, fz.xiaomi_curtain_options, fzLocal.aqara_opple],
    toZigbee: [tzLocal.ZNJLBL01LM_charging_state, tzLocal.ZNJLBL01LM_battery, tz.xiaomi_curtain_position_state, tz.xiaomi_curtain_options],
    exposes: [e.cover_position().setAccess('state', ea.ALL), e.battery(),
    exposes.numeric('battery', ea.STATE_GET).withValueMin(0).withValueMax(100).withUnit('%').withDescription('Battery percentage'),
    exposes.enum('charging_state', ea.STATE_GET, ['not_charging', 'charging', 'fully_charged', 'what']),
    ],
    ota: ota.zigbeeOTA,
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.endpoints[0];
        // await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
    },

};

module.exports = definition;
```

totalitarian, 2022-01-25:
Excellent work. Being able to set motor speed is a new option in the aqara app, so it would be great if we could also configure that in z2m.

Pete1979, 2022-01-25:
Hi,
Hoping this is the place to get help with my small issue.
I have one of these rollers and i have inverted it with the reset button. And when i toggle the invert setting in z2m for the unit, the slider is inverted, but not the buttons. So for it to work i need to toggle off the invert setting on the device in z2m, but then the slider is at 100 when the blind is open.

Gortosch, 2022-01-26:
Hello,

thanks invis-z for your great work. 👍
I hope someone from the developers can implement your information in the official version.

Grüße
Sven

Mustang1966, 2022-01-30:
Great work @invis-z . Batterystats are working in the edge version!

jb123567, 2022-01-31:
Battery state working perfectly (it appears). Has anyone been able to update the firmware? Mine is showing 05-18-2021, I believe reading online there is newer out there?

Gortosch, 2022-01-31:
Hi,

when is the "official" version supposed to come out? Does anyone know a date?

Grüße
Sven

Koenkk, 2022-01-31:
Next release is tomorrow

Koenkk, 2022-02-01:
Update has just been released (1.23.0)

> will we have to do anything to this shader to get updated stats?

No, just wait till the device reports this

jeroenvdd, 2022-02-01:
> Update has just been released (1.23.0)
> 
> > will we have to do anything to this shader to get updated stats?
> 
> No, just wait till the device reports this

super thx (btw is there a way to force check the version of z2m ? :D i see you github has new version but my HA doesn't see it :D

FlatulentCow, 2022-02-04:
Just a bit of thanks/feedback.  Had to rebind but that took a few seconds, now battery shows up and as a bonus it even reports the positioning without having to refresh.  Brilliant, big thanks to everyone involved.

----------------
zzh (CC2652R Stick with latest firmware)

bimal12, 2022-02-07:
Decided to try and set my device up last night, and wanted to thank everyone for getting this set-up. Another device working smoothly.

Adzter, 2022-02-08:
I'm running 1.23.0 with ConBee2/RaspBee2 (Coordinator revision 0x26580700).

It seems to work, but the state doesn't update. I have to call up/down/stop before it updates. I'm also not getting the battery info? Is there anything I need to change in the device settings, I'm very new to z2m.

hmoreau94, 2022-02-10:
Hey Guys, 

I'm running 1.23.0 with Coinbee 2. I personally have tried everything with the roller driver and can't seem to have a state update (until I press the stop button to trigger a refresh)

As suggested by @Adzter I've tried to re-pair the device but it didn't change anything (also the battery still doesn't show) 
[image]

Adzter, 2022-02-10:
Do you have the latest ConBee firmware? The documentation indicates only the latest supports this device.

hmoreau94, 2022-02-10:
@Adzter Here is a picture of the current firmware, not sure about how to check the last firmware version but mine seems to be 2.14.01. If it isn't the latest one, do you know any way I could update the firmware (either directly from HA or from my Mac) ?

kamildoleglo, 2022-02-10:
@hmoreau94 you have to check the Conbee2 firmware version (in zigbee2mqtt frontend it's in settings -> about -> coordinator firmware). If it's older than 0x26720700, you'll have to upgrade the firmware, see https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

hmoreau94, 2022-02-10:
@kamildoleglo @Adzter Nope I checked and indeed it is the latest firmware for the coinbee 2: 
[image]

kamildoleglo, 2022-02-10:
Strange, works fine for me and I seem to have the same FW versions of everything (I'm also using Conbee2)

hmoreau94, 2022-02-10:
@kamildoleglo After some fiddling around I managed to have it working (some mix of restarting and re-pairing). Thanks for your help guys and sorry for others experiencing the same issue as I don't really have a way to proceed to manage to get it working

Penait1, 2022-03-18:
I seem to have the same issue, the battery is showing up as a question mark. Restarting HA and re-pairing didn't work

I'm running 1.24 with a Conbee2 stick on firmware 0x26660700. The biggest difference with @hmoreau94 I can find is that my firmware states build date 11-24-2021 instead of 05-18-2021.

buhito81, 2022-03-23:
> I seem to have the same issue, the battery is showing up as a question mark. Restarting HA and re-pairing didn't work
> 
> I'm running 1.24 with a Conbee2 stick on firmware 0x26660700. The biggest difference with @hmoreau94 I can find is that my firmware states build date 11-24-2021 instead of 05-18-2021.
> 
> **Edit: Nvm, now running on the development branch of zigbee2mqtt and battery is showing!**

How did you manage to do that? I'm running 1.24 main on Conbee2 stick on firmware 0x2666070, on HA 2022.3.6 (HAOS 7.5), with firmware 11-24-2021. I don't have Battery, Cover Motor State,  or Cover running in available in HA...

Penait1, 2022-03-23:
> > I seem to have the same issue, the battery is showing up as a question mark. Restarting HA and re-pairing didn't work
> > I'm running 1.24 with a Conbee2 stick on firmware 0x26660700. The biggest difference with @hmoreau94 I can find is that my firmware states build date 11-24-2021 instead of 05-18-2021.
> > **Edit: Nvm, now running on the development branch of zigbee2mqtt and battery is showing!**
> 
> How did you manage to do that? I'm running 1.24 main on Conbee2 stick on firmware 0x2666070, on HA 2022.3.6 (HAOS 7.5), with firmware 11-24-2021. I don't have Battery, Cover Motor State, or Cover running in available in HA...
> 
> ...nor can it tell me if the cover is opened or closed - and the proper action to use can only be acquired by pressing the "stop" button.
> 
> [image]

I uninstalled the 'Zigbee2mqtt' addon and switched to the  'Zigbee2mqtt Edge' addon, which is also located in the store. The battery info started working for me.

States like if its closed/openand battery won't update automatically though for me, I have to refresh the values in Zigbee2mqtt in the 'Exposes' section, and then you have the correct values

den67rus, 2022-04-06:
After firmware update 11-24-2021. The ability to set the position of the curtains after five clicks on the device stopped working for me.

buhito81, 2022-04-13:
> > > I seem to have the same issue, the battery is showing up as a question mark. Restarting HA and re-pairing didn't work
> > > I'm running 1.24 with a Conbee2 stick on firmware 0x26660700. The biggest difference with @hmoreau94 I can find is that my firmware states build date 11-24-2021 instead of 05-18-2021.
> > > **Edit: Nvm, now running on the development branch of zigbee2mqtt and battery is showing!**
> > 
> > 
> > How did you manage to do that? I'm running 1.24 main on Conbee2 stick on firmware 0x2666070, on HA 2022.3.6 (HAOS 7.5), with firmware 11-24-2021. I don't have Battery, Cover Motor State, or Cover running in available in HA...
> > ...nor can it tell me if the cover is opened or closed - and the proper action to use can only be acquired by pressing the "stop" button.
> > [image]
> 
> I uninstalled the 'Zigbee2mqtt' addon and switched to the 'Zigbee2mqtt Edge' addon, which is also located in the store. The battery info started working for me.
> 
> States like if its closed/openand battery won't update automatically though for me, I have to refresh the values in Zigbee2mqtt in the 'Exposes' section, and then I have the correct values

If states don't work for you with Conbee2 on firmware 0x2666070, try upgrading to Conbee2 firmware 0x26720700. I did it today, and after re-pairing the device(s), all information is there:

[image]

Just figured I'd let you know.

0ip, 2022-04-24:
Mine does not always report the new position after setting it via `/set`: `{position: x}`. The reported value after the operation corresponds to the old one. Has anyone observed this yet?

And something else: As soon as the battery is below 30 %, the device does not manage to open/pull up the blind anymore.

Am not entirely happy with that thing 🙁

Penait1, 2022-04-25:
> Mine does not always report the new position after setting it via `/set`: `{position: x}`. The reported value after the operation corresponds to the old one. Has anyone observed this yet?
>
> And [...]

It’s strange; I have two of these opening and closing the same size shades, and one of them I had already to charge twice and is used less frequently, the other one is doing fine with 41% battery left.

also similair behaviour with dieing below 30% battery, is this a manifactoring issue, or some sort of battery drain from z2mqtt? Any way to observe the messages (maybe the one draining is being spammed with messages while in use?)

invis-z, 2022-04-26:
> Mine does not always report the new position after setting it via `/set`: `{position: x}`. The reported value after the operation corresponds to the old one. Has anyone observed this yet?
>
>
>
> And [...]

The device does not report the position until it has reached the set position or a new position is set but only if the new position will change the motor direction. 

About the battery, I suspect it is due to the high power draw caused by the motor. While idling, since the power useage is low, the battery will have a higher voltage than it being hit by the running motor. And if you keep sending set to the device to keep the motor running until the device refresh the battery stat internally, it will jump directly to something like 3%.

In reality, the power draw of the motor will alao depend on your setup, i.e. your cover and its accessories, mainly the resistance it creates on the motor.

Most of the issues mentioned here is due to the design of the device. There isn't much that can be done if the device did not send the data.

0ip, 2022-04-26:
> The device does not report the position until it has reached the set position or a new position is set but only if the new position will change the motor direction.

I have expressed myself incorrectly. After it has physically reached the targeted position, it still reports a faulty position. Only the subsequent manual retrieval of the position with `/get`: `{state:""}` provides the correct result.

invis-z, 2022-04-26:
> I have expressed myself incorrectly. After it has physically reached the targeted position, it still reports a faulty position. Only the subsequent manual retrieval of the position with `/get`: [...]

Did you mean `zigbee2mqtt/{your device name}` is not reporting the correct position? Not with `/get` or `/set`. It should be in json and looking for `"position"` 

I can't recreat this problem on my device so please include more information. What is your device firmware build date? And where exactly are you seeing the wrong data?

0ip, 2022-04-26:
Scenario: The current position is 4 % and I want to decrease it to 0 % (blind moves up in my case). For that, I'm publishing `/set`: `{position: 0}`. Phyiscally, it actually reached 0 % now, but the last messages do not reflect that.
```
Info MQTT publish: topic 'zigbee2mqtt/fl1-lir-rollerblind', payload '{"battery":94,"charging_status":false,"illuminance":0,"illuminance_lux":0,"last_seen":"2022-04-26T12:00:08.237Z","linkquality":185,"motor_state":"pause","position":4,"running":false,"state":"OFF","temperature":26,"update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/fl1-lir-rollerblind', payload '{"battery":94,"charging_status":false,"illuminance":0,"illuminance_lux":0,"last_seen":"2022-04-26T12:00:08.572Z","linkquality":182,"motor_state":"rising","position":4,"running":true,"state":"OFF","temperature":26,"update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/fl1-lir-rollerblind', payload '{"battery":94,"charging_status":false,"illuminance":0,"illuminance_lux":0,"last_seen":"2022-04-26T12:00:10.917Z","linkquality":193,"motor_state":"pause","position":4,"running":false,"state":"OFF","temperature":26,"update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/fl1-lir-rollerblind', payload '{"battery":94,"charging_status":false,"illuminance":0,"illuminance_lux":0,"last_seen":"2022-04-26T12:00:10.922Z","linkquality":185,"motor_state":"pause","position":4,"running":false,"state":"OFF","temperature":26,"update":{"state":"idle"},"update_available":false}'
```

It does not always happen, but in such cases, as mentioned above, I can manually send a `/get` and all of a sudden it reveals its true position:
```
Info MQTT publish: topic 'zigbee2mqtt/fl1-lir-rollerblind', payload '{"battery":94,"charging_status":false,"illuminance":0,"illuminance_lux":0,"last_seen":"2022-04-26T12:07:17.379Z","linkquality":185,"motor_state":"pause","position":0,"running":false,"state":"OFF","temperature":26,"update":{"state":"idle"},"update_available":false}'
```

Firmware is 11-24-2021, Z2M 1.25, zStack3x0, coordinator 20220219, slaesh CC2652R

invis-z, 2022-04-26:
It happens only occationaly, but the device report wrong/old position info while stopping. https://github.com/Koenkk/zigbee-herdsman-converters/pull/4163 should fix it.

0ip, 2022-04-26:
@invis-z Thanks for the quick fix. A similar problem perhaps: have you ever noticed that the actual position differs from the desired position by one? So for example if I set 5%, ultimately 4% or 6% is reported?

[26 more replies]

closed 2022-10-11 by github-actions[bot]