Koenkk/zigbee2mqtt #9453: Zigbee2mqtt initial build fail (npm run build contains 2 errors)

Faustino-ai, 2021-11-02:
## What happened
Zigbee2mqtt refuses to start. Using `npm start` returns error as shown in [https://pastebin.com/xQcNbEu4](url)
Tried `npm run build` as mentioned in other threads, however it returns with 2 errors as shown in [https://pastebin.com/J8CNsUV7](url)

## What did you expect to happen
Zigbee2mqtt start as usual. Version 1.19 started without any issue.

## How to reproduce it (minimal and precise)
`npm start` and `npm run build`

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2652P (zzh)
Adapter firmware version: CC2652R_coordinator_20210708

danieledwardgeorgehitchcock, 2021-11-02:
Have you done what it suggests, and use the `--scripts-prepend-node-path` option?

Faustino-ai, 2021-11-03:
> Have you done what it suggests, and use the `--scripts-prepend-node-path` option?

I don't think I did that. May I know where to find relevant information on this? I didn't see this suggestion at all.

danieledwardgeorgehitchcock, 2021-11-03:
It’s in your error logs

closed 2021-12-12 by github-actions[bot]