Koenkk/zigbee2mqtt #9454: Aqara TVOC sensor (VOCKQJK11LM) stops updating

JDrewes, 2021-11-02:
## What happened
I have 3 Aqara TVOC devices (VOCKQJK11LM). At first, all three worked fine for about a week. Then, no more updates were reported for one device through Z2M, a few days later also for another.
Re-pairing devices worked - but only at the cost of one of the other three randomly "falling of the network" (stopping to update and LQI being marked as unavailable). While working, LQI is usually from 80 to 110.

I have done many repairings (sometimes close to a zigbee router, sometimes at the coordinator), and I cannot get all three devices to keep updating at the same time for longer than a day. I don't know if they would all stop if I just waited long enough, or if they are somehow in conflict with each other?

Sometimes, when pressing the single button on the device for a long time to reconnect, it actually won't do so unless I do one of the following:
unplug/replug the zigbee coordinator and retart zigbee2mqtt
or
repair next to a different device (if it was connected through the router before, re-pair at the coordinator and vice versa).

I have a total of 8-9 devices connected, all Aqara (except for the coordinator, which is a CC2531 Sonoff-type USB stick). None of them have ever "gone off the network" or stopped reporting, only the TVOCs do that.

I found that changing the channel of my 2.4GHz wifi has changed things a little (it now takes a bit longer for the first device to stop updating), but the problem still persists. I don't believe it is simply related to interference because the other Aqara-zigbees have no problems.

Thanks for reading and thanks for Z2M in the first place!

## What did you expect to happen
devices should keep updating 

## How to reproduce it (minimal and precise)
just pair and wait a few days

## Debug info
Zigbee2MQTT version: 1.21.0 (commit #70891eec)
Adapter hardware: CC2531
Adapter firmware version: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201127,"transportrev":2},"type":"zStack12"}'

JDrewes, 2021-11-02:
See also discussion #9171 Aqara TVOC does not publish changes

ocso, 2021-11-07:
I can confirm the same issue with my device.

Fafitude, 2021-11-10:
J'ai le même problème avec une conbee 2 et en version 1.22.0.
Une idée du problème ?

hb020, 2021-11-15:
Voting up here, same issue. 
On CC2652RB from slaesh, with 88 average link quality.
Zigbee2MQTT version 1.21.2 commit: 4492212
Coordinator type zStack3x0
Coordinator revision 20210120
Frontend version 0.6.12.

What seems to matter:
* setting Zigbee2MQTT to "enable join" keeps the connection a bit longer. 
* the problem started when I moved the sensor from a position when I had an LQI average at 60, to a position where LQI average is better (88). Will experiment some more.

kovacsbalu, 2021-12-17:
I can confirm the same issue with my device also

Bobbyhax, 2021-12-18:
There is a firmware update for the VOCKQJK11LM. I updated from 0024 to 0026, that addresses this issue.
To get the update, I connected the sensor to a Aqara G2H camera, and sign in with a Aqara account in the app, then I could update (didn't get the update in Homekit mode)

After the update was done, I reconnected the sensor to my Z2M, and it works fine for now

[image]

hb020, 2021-12-18:
now I'm not used to aqara/mijia/xiaomi devices in general, but that seems to indicate that the various other aqara hub devices (Camera G3, E1, M2, M1S, Camera G2H and Aqara Hub) could also work. But I don't have any. Anyone care to confirm? I feel more like getting a E1 or M2 if needed.

gyver, 2021-12-18:
I have the same model and unfortunately the same problems with updates stopping very quickly. It is usually a matter of minutes or around 1 hour after adding it to the network before all updates stop.

I tried letting the network decide the router used to connect or forcing 3 different ones and get the same result : usually I get updates for some of the 3 main attributes (humidity, temperature, voc). The updated attributes are not always the same ones but until now there was never a try were all of them were reported and on all tries eventually all of them stop being updated altogether.

I don't have any Aqara hub/account to update the firmware like @Bobbyhax. Is there a workaround to get this device working or a way to update it without an Aqara hub/account ?

gyver, 2021-12-18:
Note for reference : I use the git master version of zigbee2mqtt (recently updated, just checked and it is in sync with this repository) on pi4 with a pizigate+ managing 49 devices among them 15 routers from various manufacturers. The only non default configuration are for the pizigate, compatibility with homeassistant and choosing channel 25 to avoid nearby Wifi networks interference.

MansM, 2021-12-20:
I wonder if it should be removed from the supported list or at least have a note... as the current situation has made it useless (I wouldn't have bought it if I knew about this issue before)

Bobbyhax, 2021-12-20:
Here is an update: They day after the firmware updated, I lost connection to the sensor again, and I needed to press on the button to get it connected. I do not recommend this sensor, and would buy it again! 
I read on Reddit that it is many others that have the same problems as we have. Someone suggests that the sensor have bad connection to the battery, and have placed cardboard between the battery and lid, and has solved the issue for them. 

https://www.reddit.com/r/Aqara/comments/p03684/no_response_on_my_brand_new_tvoc_sensor/

patrykwojtynski, 2022-01-04:
Unfortunately I also experience missing values after pairing and I can't find a way of updating it without Aqara hub. I tried with TuYa ethernet hub, no luck.

leoluk, 2022-01-05:
It should be possible for someone with an Aqara hub to sniff the upgrade and dump the OTA file. Would be greatly appreciated!

Fafitude, 2022-01-05:
Bonjour, ça serait top en effet de pouvoir le mettre à jour par OTA via Z2M.
Est-ce possible ?

gyver, 2022-01-06:
I've had what seemed initially like a totaly separate problem with a SonOff SNZB-02 temperature/humidity sensor. It suddenly stopped updating and I couldn't make it work by either :

- simple repair,
- let it sit 30 minutes with its battery out before pairing,
- smashing the button just after pairing while configuring,
- force-remove it before repairing
(various workarounds found around the initial configuring failure)...

When I looked at the logs I saw problems with the reporting configuration of the temperature measure (I can dig up logs on demand if needed).

Until I found a procedure that worked, which is :
- force-remove it,
- restart zigbee2mqtt,
- repair the device.

Restarting zigbee2mqtt is mandatory so there's something done on restart that solves the configuration step of the interview to fail. I verified that adding the restart is the key by testing the procedure again with another SonOff SNZB-02 that was outside (and believed to have been damaged months ago by the weather as they are not rated for outside use) and it worked, this other device came back properly on the first try.

Finally I remembered my problems with the Aqara TVOC sensor and the force-remove, z2m restart, repair seems to work with it too ! At least it is working correctly for more than 9 hours now (at least 9x more uptime than anything I got before and with all values reported instead of at best 2 out of 3).

Hope it can help others and maybe point to at least one source of the problem.

patrykwojtynski, 2022-01-07:
Thanks @gyver. I was sceptical about it but I gave it a try and followed your instructions right after I saw them.

And... It's working. What a surprise, so far 12h.

Previously I also added cardboard on battery because it felt loose, but it didn't change anything.

Fafitude, 2022-01-07:
Bonjour,

quand ça se bloque comme dit ci dessous, j'appuie sur le bouton au dessus et c'est reparti pour plusieurs jours (ou heures ...).
Suite à la mise à la dernière mise à jour il ne captait plus rien. J'ai fait une manipulation similaire à Gyver et c'est reparti. Je suis vraiment déçu de cet appareil

maximilian1001, 2022-01-07:
> Bonjour,
> 
> quand ça se bloque comme dit ci dessus, j'appuie sur le bouton qui se trouve au dessus de l'appareil et c'est reparti pour plusieurs jours (ou heures ...). Suite à la mise à la dernière mise à jour de Z2M, il ne captait plus rien. J'ai fait une manipulation similaire à Gyver et c'est reparti. Je suis vraiment déçu de cet appareil
@Fafitude
Hello,

when it hangs as said above, I press the button on the top of the device and it's gone for several days (or hours ...).
Following the update of the last Z2M update, it no longer received anything. I did a similar manipulation to Gyver and it's off again. I am really disappointed with this device

gyver, 2022-01-08:
I can confirm that pressing the button on top of the device unfreezes it in my case (after force unpair/z2m restart/pair which led to a freeze ~16 hours later).

When I pressed the button, it forced an e-ink display refresh (blinking full black/white for people not familiar with e-ink displays), and immediately refreshed the values both on display and to zigbee2mqtt.

Note that the TVOC values where largely overestimated just after the refresh (6000+ instead of the usual 15-100 range) and went down relatively quickly after : seems like a bug in the TVOC estimation on the device. One probable explanation for that bug is that it uses a low energy accumulator that it periodically fetch and reset (waking a microcontroller to do it that needs more energy and is sleeping most of the time), dividing the value by time spent since last evaluation. When the device freezes I suspect the clock freezes too but not the accumulator...

gyver, 2022-01-13:
Follow-up: my sensor is working fine for 5 days straight now. So since the re-pairing procedure (see my earlier messages) I had to press the button on top one time over ~6 days to keep it working.

It is a bother to not know if/when it will fail to send updates again and need manual intervention but at least it has been useful for nearly a week.

leoluk, 2022-01-13:
I bought two of these recently to give it a try, and they've been working perfectly for more than a week now! (with ZHA, but that shouldn't make much of a difference).

- The battery issue is real - both devices would randomly stop working and show "low battery" until I put a piece of cardboard on top of both batteries to apply more pressure.
- When pairing or reconfiguring, pressing the button repeatedly to make sure that reporting for all attributes was configured.

The temperature readings are accurate down to 0.1 °C, it's a very nice device.

nicjansma, 2022-01-16:
I've had a similar issue with 3x VOCKQJK11LM devices not reliably reporting.  After upgrading them to the 0026 firmware with a Aqara hub, I switched them back to my Conbee II.

At that point I could not reliably get them to send updates beyond the initial join.  They might send a few updates, then nothing more, no matter how often I pushed the button or had them rejoin.

I've finally found one issue that _may_ be affecting others, so I thought I'd share.  In the _Reporting_ tab in Z2M, I found the _Min rep change_ interval for Temp and Humidity had been set to `100`, which my understanding means 10*C.

[image]

After I changed these down to `0` (with a `60` second min interval), I'm now reliably getting data from them each minute. (I may bump them up to 5 mins or something soon).

[image]

I'm not sure if the _Min rep change_ of `100` came from the factory, or firmware update to 0026, or something with Z2M, but all 3 devices had this issue and I had to fix it.

Might not be an issue for anyone else here!  But since this pops up in Google for `VOCKQJK11LM` maybe others will see this and this could be a fix.

gyver, 2022-01-16:
My device has now being working for 8 days with the same settings @nicjansma had before adjusting. So they are probably the defaults for at least our firmware versions.

I don't think the min rep change is in 1/10th of a °C or humidity %. The unit used is more probably 1/1000th to match the precision that can be selected up to 1/1000th. I'd guess this is something documented in the Zigbee protocol and common across all temp/humidity sensors.

@nicjansma : note that simply changing the values triggers a reconfiguration that might have helped unrelated to the values used (my problems were clearly linked to failures in the device reporting configuration), you can check this by reverting your change or trying other values to find out if you need these values or simply to change them once. If the min rep change is low, this probably means that the device will need to wake up more often and it will probably negatively affect the battery life so you might want to be less aggressive.

For reference, my device has :
- firmware build date : 20210204.
- firmware version : 2020?www. (seems mangled somehow with an unprintable character between 2020 and www.)

Anonymenet77, 2022-01-16:
Hello,

I tried this method it worked for 5 days then nothing

Until I found a procedure that worked, which is :

force-remove it,
restart zigbee2mqtt,
repair the device.

@nicjansma 

How do you save the settings?

When I change the settings it comes back to default. :(

leroyloren, 2022-01-20:
v0.0.0_0027

http://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.airmonitor.acn01/20211223104612_OTA_lumi.airmonitor.acn01_0.0.0_0027_20211222_3BA1AA.ota

[20211223104612_OTA_lumi.airmonitor.acn01_0.0.0_0027_20211222_3BA1AA.zip](https://github.com/Koenkk/zigbee2mqtt/files/7905484/20211223104612_OTA_lumi.airmonitor.acn01_0.0.0_0027_20211222_3BA1AA.zip) (rename **".ZIP"** to **".OTA"**)

[image]

Anonymenet77, 2022-01-20:
@leroyloren how to update without hub aqara?

leroyloren, 2022-01-20:
Hi,

you need to do an OTA for the device via https://github.com/Koenkk/zigbee-OTA

@Koenkk

nsuresh81, 2022-01-21:
@Anonymenet77 I am have the same problem too. the TVOS is freezing. the display not being updated.

I just ordered an E1 hub for the sole purpose of updating the firmware. I'm not sure if I should cancel the order or try using the OTA method. I'm using HA with zigbee2MQTT on a CC2652R stick from Electrolama.

hb020, 2022-01-22:
Just created the PR. Also created a PR to add OTA capability to the sensor in herdsman. However, still trying to figure out how to go from here.

Otnow, 2022-01-23:
> Other option: build from the dev branch, but haven't yet found how to do that.

Try this: https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

elratt0r, 2022-01-24:
Has anyone the OTA file for 0.0.0_26? Would like to try that one. 0027 is messing with the simple connector response, 0022 works mostly reliable with bind of 0x000c/0x0055 to coordinator but has battery issues.

patrykwojtynski, 2022-01-25:
I was able to do OTA from dev branch. Before doing it I followed one more time the repairing steps with Force Remove and reboot. Took 55min to do it.

Hopefully now will be more stable. Thanks for the PRs and all advices.

TheLordVader, 2022-01-26:
> I was able to do OTA from dev branch. Before doing it I followed one more time the repairing steps with Force Remove and reboot. Took 55min to do it.
> 
> Hopefully now will be more stable. Thanks for the PRs and all advices.

Great! Please let us know the outcome, and thanks for all the efforts put into this annoying issue.

hb020, 2022-02-07:
Installed the new firmware (via latest zigbee2mqtt release, and full reinstall of the sensor), but the sensor still drops out. So that was not the solution.

leroyloren, 2022-02-07:
Hi, I don't think it's a drop, but the device is completely unusable like a ZigBee. 
It's more like a VOC thermometer/hygrometer with a display. 
Zigbee communication of this device makes no sense at all.

 If you really need to measure VOC, buy one **Frient (Develco) AQSZB-110**

TheLordVader, 2022-02-07:
Well, I have updated to the latest firmware as well, both via Zigbee2MQTT and via a newly bought Aqara E1 hub just for this purpose. 
I have two of these units, one of them is connected directly my CC2652P and works well after the update. The other one however, is connected via a IKEA Trådfri bulb, and drops/loses connection after a while.

leroyloren, 2022-02-07:
Even so, the device is unusable. To compare the picture.

The value with the gauge is Develco (updates every few minutes)
Aqara, for example, last time before approx. hour (the value on its display is the same)

[image]

amaciuc, 2022-02-07:
> Yes, I agree with that. The Aqara has a very, very sluggish interval. I have one of my units close to my monitor with HA, and 9 times out of 10 there are different values on the Aqara display vs in HA.

Idem here

elratt0r, 2022-02-07:
Keep in mind that the EInk is not updated that often. Same applies for temperature and humidity values. I have two TVOC running now. One with deconz on FW0026 and one with Z2M/CC2531 with FW0027 (Z2M does not care about the missing simple descriptors and reads all the values from the respective clusters/sets the matching report intervals).

The values seem to be accurate so far (compared to each other and a temp/hum sensor from Aqara right next to them).

The TVOC does not seem to like being connected to a router (I tried IKEA Repeater, OSRAM Smart+ and Aqara Power Plug). In that constellation I loose connectivity dependig on the report interval after 1-2 days (the shorter the report interval the sooner it looses the connection).

elratt0r, 2022-02-07:
ls12 is a Frient Device
ls13 is TVOC@FW0026 via deconz
ls19 is TVOC@FW0027 via Z2M

The dots show the update intervals, so far all looks good. I modified the report intervals and that helped with the continuity of the reports.

[image]

elratt0r, 2022-02-10:
Hi @JornDL

all min/max/change:
genAnalogInput: 10/300/30
Temperature: 10/300/20
Humidity: 10/300/100

When writing the values I pressed the button on top once before sending, so the device would be awake. (Found out by sniffing the traffic that this seemed to be necessary to make a successful write).

JornDL, 2022-02-17:
> Hi @JornDL
>
> all min/max/change: genAnalogInput: 10/300/30 Temperature: 10/300/20 Humidity: 10/300/100
>
> When writing the values I pressed the button on top once before sending, so the device would be [...]

Hi @elratt0r 

How did you write the rep change to 30 on the genAnalogInput ?

When I try this and read it back, it is just empty.

amaciuc, 2022-02-17:
> Hi @elratt0r
>
> How did you write the rep change to 30 on the genAnalogInput ?
>
> When I try this and read it back, it is just empty.

[image]

I changed the `min rep interval` to 3000, that means the sensor will forcibly update at every 30 minutes.

@elratt0r thank you for your feedback. You just saved my day. Now, I have no longer issues with the update of values.

leoluk, 2022-02-17:
> When I try this and read it back, it is just empty.

How would you read it back?

JornDL, 2022-02-17:
> > When I try this and read it back, it is just empty.
> 
> How would you read it back?

I'm using deCONZ GUI

[image]

elratt0r, 2022-02-17:
> > How would you read it back?
> 
> I'm using deCONZ GUI, so I can read config and write config.

deCONZ has a problem with the genAnalogInput conversion for the GUI and the TVOC
is not complying to the ZigBee Analog Cluster Spec (it reads as u16 instead of single precision).
See also my post at the deCONZ Forum post [Aqara TVOC ZHAAirQuality data#24](https://forum.phoscon.de/t/aqara-tvoc-zhaairquality-data/1160/24)

It is possible to write the reportable change correctly from deCONZ gui but not to read it back.

elratt0r, 2022-02-17:
> I changed the `min rep interval` to 3000, that means the sensor will forcibly update at every 30 minutes.
> 
> @elratt0r thank you for your feedback. You just saved my day. Now, I have no longer issues with the update of values.

@amaciuc 
Thank you, glad my fiddling with those devices was worth it :-)

The interval fields are seconds, so your max report is now 50 minutes (according to your picture).

Do you keep track of the reports (influx, log, ...)?
Does it match the set intervals?

amaciuc, 2022-02-17:
> @amaciuc Thank you, glad my fiddling with those devices was worth it :-)
>
> The interval fields are seconds, so your max report is now 50 minutes (according to your picture).
>
> Do you keep track of the [...]

Oh, you are right. 
[image]
The value of this parameter forces the sensor to update at every 30 minutes
> Do you keep track of the reports (influx, log, ...)? Does it match the set intervals?

No, I don't. I observed the sensor is updating periodically and that was enough for me

leoluk, 2022-02-19:
Hm, interesting. I set mine to a 300s interval on my z2m network and it promptly stopped reporting after a short while, requiring a button click to be awakened:

[image]

Meanwhile, the one on my ZHA network (with default reporting settings... 3600s or so) goes for weeks at a time before requiring a click. Perhaps some firmware counter overflow and the different experiences reported here are simply due to different reporting intervals?

elratt0r, 2022-02-23:
@leoluk My ls13 / TVOC@FW0026 via deconz stopped reporting after 23 days. Complete radio silence (nothing to be seen on the sniffer) - ls19 / TVOC@FW0027 is still running. As soon as I am back from overseas I will investigate further

nicjansma, 2022-02-27:
I continue to have issues with my 3x VOCKQJK11LM devices stopping reporting after 1-7 days.  They're all on FW0027.  A doube-click of the top button a few times usually wakes them, refreshes the screen, and they continue reporting.

I opened a support case with Aqara.  Unfortunately they state they do not support the product unless used with an Aqara hub:

> But sorry to say the Aqara devices are not officially supported by Combee II Zigbee hub. We haven't tested the Aqara devices with the Combee II Zigbee hub, so we can't guarantee the compatibility and stability

This is disappointing but besides the point.

My 3x VOCKQJK11LM were on my Zigbee network with a Conbee II hub and other wired extenders.  I moved one of my VOCKQJK11LM devices to an Aqara E1 hub, and waited.  Sure enough, 9 days later it had the same problem -- stopped reporting, and double-clicking the top button a few times started it up again.

Evidence: I put it on the Aqara E1 hub on 2/16:

[image]

By 2/25 it stopped reporting.  As of 6:05pm you can see it had stopped reporting around 8:00am that morning:

[image]

This is what this is what the device looked like:

[image]

To get the device to produce more updates, I did the same procedure that I mentioned in my previous email: I double-clicked the top button a few times.  Eventually, it does a "black then white" refresh of the e-ink display, and this is what it looked like:

[image]

Note the temp and humidity jumped at that point.

The updates then started again at 18:05:

[image]

[image]

I know this isn't a general support ticket for the VOCKQJK11LM but to me this is pretty good evidence of a firmware problem with those devices.  I have ~12 other Aqara temp and door sensors that all continue to report reliably on my network.  All 3x of my VOCKQJK11LM show a constant issue.  As of right now, one of the sensors stopped reporting 17 hours ago, one stopped reporting 20 hours and, and the 3rd is still reporting reliably.

I haven't heard back yet from Aqara support.  But I'm not planning on wasting any more time debugging this with Z2M until they release another firmware.

maximilian1001, 2022-02-27:
@nicjansma Did you contact Aqara support again after you found out that the problem also occurs with the original Aqara E1 Hub? What do they say about it? As long as the error occurs there, we don't need to look for an error in our custom zigbee networks...

nicjansma, 2022-03-04:
Aqara support is fairly responsive to my emails, but they keep requesting more information (e.g. that I put the device back on the E1 hub, get its Accessory ID, share screenshots again, etc).  I keep emphasising that it's not just my 3x TVOCs, but dozens of people in this Github issue, [reddit threads](https://www.reddit.com/r/Aqara/comments/p03684/no_response_on_my_brand_new_tvoc_sensor/) and the [Amazon page](https://www.amazon.com/gp/product/B094R8RBWT/) all reporting something seemingly similar.

I've asked if their development team is aware of any issue like this.  Will let you know what they respond with.

siccovansas, 2022-03-04:
I'm also struggling with intermittent unresponsiveness of this device.

I'm still on the 0024 firmware, and can't figure out how to update. I'm running the dev branch of Zigbee2MQTT and in the logs it states `"update_available":true`, but it doesn't seem to apply update by itself (I've waited for a week).

I've tried to apply the update manually with `mosquitto_pub -t 'zigbee2mqtt/bridge/request/device/ota_update/update' -m '{"id": "<MY_DEVICE_ID>"}'`, but then I get this error:

```
Zigbee2MQTT:debug 2022-03-04 11:42:44: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id": "<MY_DEVICE_ID>"}'
Zigbee2MQTT:info  2022-03-04 11:42:44: Updating '<MY_DEVICE_ID>' to latest firmware
Zigbee2MQTT:info  2022-03-04 11:42:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating '<MY_DEVICE_ID>' to latest firmware","meta":{"device":"<MY_DEVICE_ID>","status":"update_in_progress"},"type":"ota_update"}'
Zigbee2MQTT:debug 2022-03-04 11:43:07: Updating to latest '<MY_DEVICE_ID>' (lumi.airmonitor.acn01)
Zigbee2MQTT:debug 2022-03-04 11:43:07: Using endpoint '1'
Zigbee2MQTT:debug 2022-03-04 11:43:24: Received Zigbee message from '<MY_DEVICE_ID>', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":198}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-04 11:43:24: MQTT publish: topic 'zigbee2mqtt/<MY_DEVICE_ID>', payload '{"battery":100,"humidity":44.8,"linkquality":76,"temperature":20.73,"update":{"state":"available"},"update_available":true,"voc":198,"voltage":3000}'
Zigbee2MQTT:debug 2022-03-04 11:43:37: Update of '<MY_DEVICE_ID>' failed (Error: Device didn't respond to OTA request)
Zigbee2MQTT:info  2022-03-04 11:43:37: MQTT publish: topic 'zigbee2mqtt/<MY_DEVICE_ID>', payload '{"battery":100,"humidity":44.8,"linkquality":76,"temperature":20.73,"update":{"state":"available"},"update_available":true,"voc":198,"voltage":3000}'
Zigbee2MQTT:info  2022-03-04 11:43:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of '<MY_DEVICE_ID>' failed (Device didn't respond to OTA request)","meta":{"device":"<MY_DEVICE_ID>","status":"update_failed"},"type":"ota_update"}'
Zigbee2MQTT:info  2022-03-04 11:43:37: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"<MY_DEVICE_ID>"},"error":"Update of '<MY_DEVICE_ID>' failed (Device didn't respond to OTA request)","status":"error"}'
Zigbee2MQTT:error 2022-03-04 11:43:37: Update of '<MY_DEVICE_ID>' failed (Device didn't respond to OTA request)
Zigbee2MQTT:debug 2022-03-04 11:43:37: Error: Device didn't respond to OTA request
    at requestOTA (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:196:15)
    at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:283:21)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:256:21)
Zigbee2MQTT:debug 2022-03-04 11:43:54: Received Zigbee message from '<MY_DEVICE_ID>', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":191}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-03-04 11:43:54: MQTT publish: topic 'zigbee2mqtt/<MY_DEVICE_ID>', payload '{"battery":100,"humidity":44.8,"linkquality":76,"temperature":20.73,"update":{"state":"available"},"update_available":true,"voc":191,"voltage":3000}'
```

Am I doing something wrong?

nicjansma, 2022-03-04:
Aqara support's response:

> We found your TVOC went offline that time, so it doesn't update the data in the Aqara Home app if you didn't press the button of the TVOC.
> Please let us know if you found something new.

Yeah!  Going "offline" is the problem!  Devices shouldn't have to be manually woken up randomly by pressing a button once a week!

I'm trying to get this issue escalated.  I repeatedly mention how others are seeing similar issues, but so far that is not being acknowledged in my emails at all.

@siccovansas I had purchased the E1 hub to update these devices.  After updating I switch them back to my Conbee II.  *but* I have the issue on the 0027 firmware so I wouldn't recommend that as a fix.

elratt0r, 2022-03-04:
@nicjansma I can confirm the problem with FW 0022, 0026 and 0027. I put them all on an Aqara Hub M2 on 26.02.2022. They all stopped working between 3.3.2022 and 4.3.2022 (all in a timeframe of ~140 to 150 hours). Nothing else connected to the Hub M2, all sensors in close proximity (~3m). A single click on the button issued a network rejoin request and the values are reported again.

I have restarted the setup now and sniffing the Zigbee traffic 24/7 until it will break again, maybe sth. can be found there.

leroyloren, 2022-03-04:
[image]

http://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.airmonitor.acn01/20220222202427_OTA_lumi.airmonitor.acn01_0.0.0_0029_20220222_17EC2C.ota

[20220222202427_OTA_lumi.airmonitor.acn01_0.0.0_0029_20220222_17EC2C.ota.zip](https://github.com/Koenkk/zigbee2mqtt/files/8188078/20220222202427_OTA_lumi.airmonitor.acn01_0.0.0_0029_20220222_17EC2C.ota.zip) (remove .zip)

hb020, 2022-03-05:
Added to Zigbee-OTA. This will find its way in the next update for zigbee2mqtt (for those that do not want to buy a Aqara hub) .

bluechris, 2022-03-12:
I updated yesterday with the custom way and so far it works fine. I really hope that this firmware will fix the connection problems

nicjansma, 2022-03-24:
I've updated all 3 of my TVOC sensors to the latest firmware, `0.0.0_0029`.

Unfortunately, 2 of those froze (same symptoms as before) at around 24 and 30 hours, and had to be interacted with (double-button-push) to rejoin.  The third is still going, but that's probably just luck.

Sadly, Aqara support hasn't been responsive to my emails for the last 3 weeks.

[37 more replies]

closed 2022-08-26 by github-actions[bot]