Koenkk/zigbee2mqtt #9457: Ikea OTA Update E1743/E1812

alexwala, 2021-11-02:
Checking [Ikea Release Notes](https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html) i found that for E1743/E1812 new firmware versions are available:

- TRÅDFRI remote, SYMFONISK sound remote, TRÅDFRI shortcut button (V-2.3.080)
- TRÅDFRI on/off switch, open/close remote (V-2.3.079)

z2m UI informed me about new updates, currently firmware build displayed on UI:

`E1743	20190723	2.2.010`
`E1812	20190715	2.3.015`

[fw.ota.homesmart.ikea](http://fw.ota.homesmart.ikea.net/feed/version_info.json) lists new versions:

`10005778-10.1-TRADFRI-onoff-shortcut-control-2.3.079.ota.ota.signed`
`10054470-TRADFRI-shortcut-button-2.3.080.ota.ota.signed`

Performed update, took about 30min for each device.

Got update SUCCESSFUL, but versions on UI stays on old firmware build version:

`E1743	20190723	2.2.010`
`E1812	20190715	2.3.015`

Tried to do a update check again, but response is: no update available.

Was the update performed correctly ?
How to check firmware version on device ?
Do you still have to use the test server for ota (due corrupted firmwares) ?

Unfortunately [fw.test,ota.homesmart.ikea](http://fw.ota.homesmart.ikea.net/feed/version_info.json) lists older versions:

`10005778-10.1-TRADFRI-onoff-shortcut-control-2.2.010.ota.ota.signed`

EricV-hue, 2021-11-02:
I also struggled with the updated sensors. Same issue: after updating the 'old' frimware version remained visiblem in the UI.
Stopped zigbee2mqgt in hassio, removed the usb-stick (CC2652RB) and started Z2M again. Firmware version was listed OK then.

Other issues/observations:
- after the update the bindig to a group did not work anymore
- removing the remote from the zigbee net was only possible by forced removal.
- after re-pairing  binding of the remote to a group was still not possible
- direct binding to e.g. a bulb is possible now
- this direct binding can be done without removal of the default_binding_group
- added 'LevelCtrl' and 'OnOff' bindings to the coordinator, to make the remote accessible for hassio

robertalexa, 2021-11-02:
Duplicate of pinned issue:
https://github.com/Koenkk/zigbee2mqtt/issues/7716

robertalexa, 2021-11-02:
The solution is as follows, after you have upgraded.

Stop z2m, unplug adapter, plug adapter, start z2m
Unbind remote from default bind group
Unbind remote from custom group (this seems to be broken now, it was working for me before too)
Bind remote to 1st device in the old group (light)
Bind remote to 2nd device in the old group (light)
Bind remote to coordinator (so you can use the scenes buttons if the bulbs dont support it)

alexwala, 2021-11-03:
finally it is working: multiple pairings and reboots using https://github.com/Koenkk/zigbee2mqtt/issues/9457#issuecomment-957760527
tip: try pairing the device closer to the coordinator [Zigbee2MQTT FAQ](https://www.zigbee2mqtt.io/guide/faq/#interview-fails)
thanks all !

closed 2021-11-03 by alexwala