Koenkk/zigbee2mqtt #9459: TS011F (_TZ3000_mraovvmm) socket with cc2531 version 1,2

Shaman2, 2021-11-02:
I try to use TS011F (_TZ3000_mraovvmm) socket with cc2531 version 1,2 coordinator. Zigbee2mqtt last version.
But I have a problem:

{
    "last_seen": "2021-11-02T12:03:06.265Z",
    "linkquality": 94,
    "current": null,
    "energy": null,
    "power": null,
    "power_outage_memory": null,
    "state": "OFF",
    "voltage": null
}

I can change state and see linkquality, but current, voltage and energy I dont receive. What I need to do for support it?

andiukas, 2021-11-03:
Same problem with blitzwolf BW-SHP15 socket, with BW-SHP13 no problem.

Shaman2, 2021-11-03:
> Same problem with blitzwolf BW-SHP15 socket

yes, this is blitzwolf BW-SHP15 socket. It TS011F (_TZ3000_mraovvmm)

closed 2021-12-11 by github-actions[bot]