Koenkk/zigbee2mqtt #9464: UNSUPPORTED_ATTRIBUTE setting Philips LTW010 color_temp_startup

regystro, 2021-11-02:
## What happened
Setting color_temp_startup from the Frontend results in error

## Debug info
Zigbee2MQTT version: 1.22.0
Adapter hardware: CC2531
Adapter firmware version: 20201127

## Device generating error
Hue white ambiance E26/E27: https://www.zigbee2mqtt.io/devices/8718696548738.html#philips-8718696548738
Firmware build date: 20161206
Firmware version: 1.19.0_r19755 (no OTA available)

## Relevant log
```
debug 2021-11-02 17:17:54: Publishing 'set' 'color_temp_startup' to 'Hue white ambiance E27'
error 2021-11-02 17:17:54: Publish 'set' 'color_temp_startup' to 'Hue white ambiance E27' failed: 'Error: Write 0x0017880103dbe7a4/11 lightingColorCtrl({"startUpColorTemperature":370}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
debug 2021-11-02 17:17:54: Error: Write 0x0017880103dbe7a4/11 lightingColorCtrl({"startUpColorTemperature":370}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:293:23)
    at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:375:22)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:944:13)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:247:36)
```

Thanks in advance.

danieledwardgeorgehitchcock, 2021-11-03:
That is a very old device firmware for your device - The power on behaviour was added in Philips Hue firmware's > 2018 (yours is 2016) so, that would explain the error.

There should definitely be a new firmware for your device. Do you have a hue hub? If so, it might be worth connecting the device to that, and searching for new updates.

regystro, 2021-11-04:
You are right. Once connected the bulb to the Hue bridge it downloaded a couple of updates: 1.19.0 -> 1.50.2 -> 1.88.1
Now setting color_temp_startup from the Frontend in Z2M works without a flaw. Thanks!!

Problem is I was offered the bulb's firmware update from the OTA tab in Z2M frontend, and that caused the downgrade to 1.19.0. So what I think needs fixing is the current firmware detection and/or offering for that Philips bulb, because even knowing for sure mine has 1.88.1 firmware, the OTA tab still reports:

- Firmware build date: 20161206
- Firmware version: 1.19.0_r19755

I even restarted both MQTT server and Zigbee2MQTT, but I get same result regarding firmware versions reporting.
When I press "Check for new updates" nothing changes.

I don't know how often is the device firmware info updated and where does it come from. Maybe it's a matter of waiting.

danieledwardgeorgehitchcock, 2021-11-04:
What version of Zigbee2MQTT are you using? I think a lot of the OTA stuff got fixed a couple of releases ago...

regystro, 2021-11-04:
Zigbee2MQTT version: 1.22.0
Just in case also tried switching on and offf "IKEA TRADFRI OTA use test url", but got same result.

closed 2021-12-13 by github-actions[bot]

heresiarch, 2022-02-13:
I have the same issue can't upgrade via z2m:
Philips [](https://www.zigbee2mqtt.io/supported-devices/#v=Philips)[8718696548738](https://www.zigbee2mqtt.io/devices/8718696548738.html#philips-8718696548738)| 20161206 | 1.19.0_r19755
z2m latest docker image koenkk/zigbee2mqtt:1.23.0
configuration.yaml contains url for old firmwares:
```
ota:
    zigbee_ota_override_index_location:  'https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json'
```
I could upgrade with the settings above two other Hues to 1.88.1 version over 1.50.2_r30933:
[9290012573A](https://www.zigbee2mqtt.io/devices/9290012573A.html#philips-9290012573a)
[9290011370B](https://www.zigbee2mqtt.io/devices/9290011370B.html#philips-9290011370b)

heresiarch, 2022-02-13:
After many tries finally I got it working with out setting the ota option in configuration.yaml.
The Hue is now on 
[](https://www.zigbee2mqtt.io/supported-devices/#v=Philips)[8718696548738](https://www.zigbee2mqtt.io/devices/8718696548738.html#philips-8718696548738)	20191218	[1.50.2_r30933](https://www.philips-hue.com/en-us/support/release-notes/lamps)