Koenkk/zigbee2mqtt #9465: Incorrectly recognized tuya switch module model

drbios, 2021-11-02:
## What happened
Incorrectly recognized tuya switch module model, i bought this module https://es.aliexpress.com/item/1005003108226482.html?spm=a2g0s.9042311.0.0.165d63c0j7xmby 
Supposed to have power status memory, timer functions, switch type, etc.
It should be recognized as TS0002_switch_module but it is recognized as TuYa TS0012

[image]

## What did you expect to happen
be recognized as TS0002_switch_module fully featured

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.22.0-dev commit: 5709898c
Adapter hardware: CC2652P (egony RFstar)
Adapter firmware version: zStack3x0 20210708

Koenkk, 2021-11-08:
This is clearly a TS0012 and not a TS0002 (see the Zigbee model in your screenshot)

drbios, 2021-11-08:
> This is clearly a TS0012 and not a TS0002 (see the Zigbee model in your screenshot)

As indicated in the vendor information that is in the link, the unit should have access to choose the type of switch, behavior when turning on, etc. but I only have the options of on and off.  Does this mean that I was tricked and didn't really have any of those options?  or as I consulted you, could it not be a case of erroneous rebirth of the model by z2m?

Koenkk, 2021-11-10:
> ould it not be a case of erroneous detectiob of the model by z2m?

I don't think this is the case, this is just what the device send.

Maybe the `TS0012` also supported the switch type, you can try with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0012'],
    model: 'TS0012',
    vendor: 'TuYa',
    description: 'Smart light switch - 2 gang',
    whiteLabel: [{vendor: 'Vrey', model: 'VR-X712U-0013'}, {vendor: 'TUYATEC', model: 'GDKES-02TZXD'},
        {vendor: 'Earda', model: 'ESW-2ZAA-EU'}],
    toZigbee: extend.switch().toZigbee.concat([tz.tuya_switch_type]),
    fromZigbee: extend.switch().fromZigbee.concat([fz.tuya_switch_type]),
    extend: extend.switch(),
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right'), exposes.presets.switch_type_2()],
    endpoint: (device) => {
        return {'left': 1, 'right': 2};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        device.powerSource = 'Mains (single phase)';
        device.save();
    },
};

module.exports = definition;
```

see https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-adding-your-device how to use the external converter

drbios, 2021-11-10:
> I don't think this is the case, this is just what the device send.
>
> Maybe the `TS0012` also supported the switch type, you can try with the following external converter:
>
> ```js
> const fz = [...]

it works :) ... now the ts0012 module reports Switch_type attribute and no error in the log
[image]

It would be possible to try to add the "initial state" of the switch (this module is supposed to remember the last state it was in)

And do the same test with the TS0013?

Koenkk, 2021-11-11:
Great! added it for TuYa TS0011, TS0012, TS0013 and TS0014

drbios, 2021-11-11:
> Great! added it for TuYa TS0011, TS0012, TS0013 and TS0014

Thanks 👌👌
any chance to add "initial state" or access the "timer" or binding options?

Koenkk, 2021-11-11:
> any chance to add "initial state" or access the "timer" or binding options?

I don't see this is supported yet by other TuYa devices, you can check https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#instructions to add support for this yourself.

DJTerentjev, 2021-11-12:
I think we need separate light switch and relay module even it's has same TS0013 model.
Not sure, but I think light switch doesn't have this option.
@drbios what is Zigbee Manufacturer of your switch module?

drbios, 2021-11-14:
> @Koenkk I think we should separate light switch and relay module even it's has same TS001* model. I am sure, light switch doesn't have this option "switch type". And at least picture should be different. One more TS0014 switch module doesn't exist.
> 
> [Please review!](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3335/files)
> 
> [Also images added. ](https://github.com/Koenkk/zigbee2mqtt.io/pull/1011)
> 
> @drbios i see you have same module I got, once @Koenkk approve changes you should get "initial state" as well. "timer" is supported not by devise, but by gateway - you can do it with your automation system. And binding should be supported by device, but this module doesn't support it, so z2m can't add it.

awesome, thanks a lot for take my request.
I updated z2m to the latest dev branch and ts0012_switch_module works perfect (power_on_behavior works )
Only missing ts0013_module_switch and ts0011_switch_module. Those two modules are the same as ts0012_switch_module with 1 and 3 gangs respectibilty

DJTerentjev, 2021-11-14:
@drbios all tree were added same time. Check the [link.](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3335/files) What do you mean missing?

drbios, 2021-11-14:
> @drbios all tree were added same time. Check the [link.](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3335/files) What do you mean missing? If all work as you expect, please close this issue.

Sorri about my English, by missing  I mean that ts0011 and ts0013 did not change.
Maybe becose the manufacturer is different 
[image]
[image]

DJTerentjev, 2021-11-14:
@drbios I got it. Zigbee manufacture is also matter.  I will add it now. Once @Koenkk approve it will be fixed.

drbios, 2021-11-14:
> @drbios I got it. Zigbee manufacture is also matter. I will add it now. Once @Koenkk approve it will be fixed.

I really appreciate the attention and I am very sorry for the delay in delivering all the information.  great job 👏 👍 💪

drbios, 2021-11-14:
> > @drbios I got it. Zigbee manufacture is also matter. I will add it now. Once @Koenkk approve it will be fixed.
> 
> I really appreciate the attention and I am very sorry for the delay in delivering all the information. great job 👏 👍 💪

i updated and all modules works perfect

drbios, 2021-11-15:
Just a little detail...change the picture ... z2m shows a switch panel not a switch module
Everything else is perfect.
Many thanks

closed 2021-12-23 by github-actions[bot]

phretor, 2023-03-01:
Sorry if I resume an old issue: I hope this can help others and digging deeper.

@drbios how do you know that binding is not supported, if I may ask? :-) I have many of these devices and I'm willing to go as far as it takes talking to the relevant parties (e.g., Tuya, Zemismart) to convince them that they need to expose `genOnOff` within the `Output Clusters` of each endpoint. It seems that they only expose `genOta` in the output cluster, which simply makes it impossibile to bind to something useful (i.e., `genOnOff` on the `Output Clusters` means that we're looking at the button pressed event), but this doesn't mean that they don't support binding as a function at all.

Also, and this question is for @Koenkk: why if I unbind an endpoint I cannot re-bind it on the same cluster? In this case I took a ZM-L03E-Z, unbound endpoint 2 (which was bound to the coordinator on the `onOff` cluster), and now I cannot see that cluster again:

[image]

dinde, 2023-07-16:
Hello,

I'm sorry to reopen this issue but something is definitly not working as expected as @phretor reported.
**Context**
Zigbee2mqtt as Home Assistant Module version 1.32.1
Home Assistant 2023.7.2
Coordinator: ZBDongle-P, type: zStack3x0, firmware: 20230507

**Issue 1**
I installed several MOES Switchs reported on z2m as:
- TS0011 -> _TZ3000_hhiodade -> Moes ZS-EUB_1gang which is: MOES ZS-EU1-WH-MS
- TS0012 -> _TZ3000_18ejxno0 -> TuYa TS0012 which is: MOES ZS-EU2-WH-MS
- TS0013 -> _TZ3000_qewo8dlz -> TuYa TS0013 which is: MOES ZS-EU3-WH-MS

**Issue 2**
On TS0011 & TS0012, there is a "Mode of the backlight: Off/Normal/Inverted" which is correct
On TS0013, there is "Intensity of the backlight: Low/Medium/High" which is wrong.
It should be as TS001[1-2] "Mode of the backlight: Off/Normal/Inverted."

**Issue 3**
Zigbee Direct Binding which is supported by MOES app as advertised.
Z2M does not expose correctly Clusters and there is a bug as @phretor described.
If you Unbind an Endpoint (let's say 2) and want to bind it again as it was with the coordinator: Clusters are empty.
You need to reconfigure the device to get it back.
If you want to direct bind an Endpoint (let's say 2 to another identical MOES switch) to achieve a wireless 2 ways with Zigbee Direct Binding: Clusters are empty (OnOff missing) so you can't bind it.

I also tried to directly publish a payload without success on MQTT.

Not sure if i'd rather open a new issue but it seems to be related to the OP problem.

Thanks a lot for your wonderful work @Koenkk !