Koenkk/zigbee2mqtt #9474: Cant Open Zigbee to MQTT becuase 21-11-02 21:33:29 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.4:8099 ssl:default [Connect call failed ('172.30.33.4', 8099)] and Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'

DJAMIRSAM, 2021-11-03:
## What happened
when i press the open in WebUI button i get a 502 Bad Gateway error
and a error on logs
Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'

i also get this error in HA logs
21-11-02 21:33:29 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.4:8099 ssl:default [Connect call failed ('172.30.33.4', 8099)]

its trying to connect to 172.30.33.4 except my IP range is 192.168.1.XXX

## What did you expect to happen
what open a ui

## How to reproduce it (minimal and precise)
install the plugin

## Debug info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: Conbee II
Adapter firmware version: Dont Know

Please Help

francisp2, 2021-11-03:
> Zigbee2MQTT version: 1.18.1-1

Don't use the deprecated daniel welch add-on. Use the official add-on.

https://github.com/zigbee2mqtt/hassio-zigbee2mqtt

HelpITru, 2021-12-04:
I have the same problem 

21-12-05 00:44:04 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.3:8099 ssl:default [Connect call failed ('172.30.33.3', 8099)]
21-12-05 00:45:53 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.3:8099 ssl:default [Connect call failed ('172.30.33.3', 8099)]
21-12-05 00:46:10 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.3:8099 ssl:default [Connect call failed ('172.30.33.3', 8099)]

Zigbee2mqtt
Current version: 1.22.1-1

[image]

after reboot host (Intel Nuc with HAOS)

lucian0112, 2021-12-07:
Any fix for that? I run into same issue ...i also reinstalled addon but without success

simonlock, 2021-12-10:
I'm facing exactly the same error.

Tosia88, 2021-12-13:
I have the same situation.

simonlock, 2021-12-13:
Reflashing my Zigbee stick and/or incrementing the pan_id from the default 6754 to 6755 did the trick for me. Now it's working.

DJAMIRSAM, 2021-12-14:
fun fact a put the usb name incorrect
im using the nortek stick (zigbee zwave combo) it has a silicon labs chip

DavidFFerreira, 2021-12-18:
I have the same probleme with edge version, but it comes live after an about 10 minuts

OJBender, 2021-12-29:
It is ingress failing  to conect to the internal docker container at 172.30.33.1:8099, most likely wanting a TLS connection.

The error message in the logs tells
`[supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.1:8099 ssl:default [Connect call failed ('172.30.33.1', 8099)]`

The service however IS running on :8099 but uses http not https

### 
[core-ssh ~]$ nc 172.30.33.1 8099
GET /45df7312_zigbee2mqtt/dashboard

HTTP/1.1 404 Not Found
Content-Security-Policy: default-src 'none'
X-Content-Type-Options: nosniff
Content-Type: text/html; charset=utf-8

ErnestStaug, 2022-01-25:
Same issue

kokrp, 2022-01-26:
Same issue.

For what its worth:

**Supervisor log:**
ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.4:8099 ssl:default [Connect call failed ('172.30.33.4', 8099)]
ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.4:8099 ssl:default [Connect call failed ('172.30.33.4', 8099)]
INFO (SyncWorker_5) [supervisor.docker.interface] Stopping addon_45df7312_zigbee2mqtt application

**Zigbee2mqt log:**
INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.2 start
> node index.js
Zigbee2MQTT:error : Error while starting zigbee-herdsman
Zigbee2MQTT:error : Failed to start zigbee
Zigbee2MQTT:error : Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error : Exiting...
Zigbee2MQTT:error : Error: Error while opening serialport 'Error: Error: Resource busy, cannot open /dev/ttyACM0'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12

JanHACS, 2022-01-28:
I have exactly the same problem here on my brand new HA installation on a Pi 3b.

Koenkk, 2022-01-29:
@kokrp 

> Error while opening serialport 'Error: Error: Resource busy, cannot open /dev/ttyACM0'

https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-resource-temporarily-unavailable-cannot-lock-port

ErnestStaug, 2022-01-29:
I seems like this issue when the zigbee2mqtt doesn't load... nothing to do with SSL, bad error.

wzmigrod, 2022-01-29:
Have the same issue. When removing zigbee2mqtt  add-on and installing it everything seems OK until first reboot of the system
[image]
The zigbee stick seems to work and the port is defined corretcly
[image]

wzmigrod, 2022-01-31:
Seems that herdsman is somehow getting the zigbeee stick corrupt. Unistalled zigbee2mqtt, flashed CC2531, installed everything again and it worked perfectly for less than 24 hrs, then everything repeated - stcik got stale and there's been no communication between stick & zigbee devices. After restarting the HA bad gateway (error 502) and same errors as in the screenshots I pasted above. Re-flashing the zigbee stick solved the issue immediately; I am afraid temporarily

kokrp, 2022-02-02:
> @kokrp
> 
> > Error while opening serialport 'Error: Error: Resource busy, cannot open /dev/ttyACM0'
> 
> https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-resource-temporarily-unavailable-cannot-lock-port

@Koenkk 
Thanks - I checked this but this is unfortunately not applicable. Stick is working fine, no power issues, etc.
I tried it a few months ago and it worked back then but since I didn't use it I disabled it. when I decided it was time to start using it, this appeared. No idea what has changed.

stanislavhannes, 2022-02-21:
Same here. And I also have the CC2531 Usb Stick, but I didn't flash it. Is it necessary? Because I know that it worked like a year before, where I already installed zigbee2mqtt on a supvervised HA (raspberry pi 4 x64), but obviously an older version.

peerkn, 2022-02-25:
Hello,
for the solution look here:

https://community.home-assistant.io/t/zigbee2mqtt-bad-gateway-with-conbee-2/325568/25
Post 13
You have to add the adapter in the config.yaml of zigbee2mqtt. It is in a seperate folder.

serial:
  port: /dev/ttyACM0 [insert your port here]
  adapter: deconz       [this is the important line here for Conbee ii]

kokrp, 2022-03-28:
Hi,

This topic will be closed in 7 days without a solution. Unfortunately I miss the skills to investigate and debug this error.
Is there anyone out there with a suggestion on how to proceed, next steps? Any pointers, suggestions, ideas, are very much appreciated. Thanks.

For what it's worth, please find below my configuration details:

=======================
**Hardware:**

**ttyACM0**
/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2193771-if00
Subsystem: tty
Device path: /dev/ttyACM0
ID: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2193771-if00
**Attributes:**
DEVLINKS: >-
  /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2193771-if00
  /dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.4:1.0
DEVNAME: /dev/ttyACM0
DEVPATH: >-
  /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.0/tty/ttyACM0

=======================
**zigbee2mqtt configuration.yaml**

data_path: /config/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  user: _<user>_
  password: _<password>_
serial:
  port: /dev/ttyACM0
  adapter: deconz
advanced:
  log_level: warn
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental: {}
availability: false
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: '-d -d'
  log: false

=======================
Zigbee2mqtt log

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing... 
[15:54:17] INFO: Socat not enabled, marking service as down
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing... 
[15:54:18] INFO: MQTT available, fetching server detail ...
[15:54:18] INFO: MQTT server settings not configured, trying to auto-discovering ...
[15:54:19] INFO: Configuring 'mqtt://core-mosquitto:1883' mqtt server
[15:54:19] INFO: Previous config file found, checking backup
[15:54:19] INFO: Creating backup config in '/config/zigbee2mqtt/.configuration.yaml.bk'
[15:54:20] INFO: Adjusting Zigbee2mqtt core yaml config with add-on quirks ...
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[15:54:21] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.24.0 start
> node index.js
CREATED DECONZ ADAPTER
Zigbee2MQTT:error 2022-03-28 15:54:28: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-03-28 15:54:28: Failed to start zigbee
Zigbee2MQTT:error 2022-03-28 15:54:28: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-03-28 15:54:28: Exiting...
Zigbee2MQTT:error 2022-03-28 15:54:28: Error: Error while opening serialport 'Error: Error: Resource busy, cannot open /dev/ttyACM0'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:182:28)
    at SerialPort._error (/app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
[15:54:29] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.24.0 start
> node index.js

kokrp, 2022-04-01:
FYI. I noticed someone opened an issue: https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/304

Koenkk, 2022-04-02:
@kokrp this is not something that can be fixed from z2m, it means another process is locking the port and therefore z2m cannot use it. https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-resource-temporarily-unavailable-cannot-lock-port

kokrp, 2022-04-11:
@Koenkk , I understand. I was hoping for someone smarter/with more experience than me to point me in the right direction. Unfortunately the information on the page you refer to did not bring my any further. Regardless, thanks for the support!

timesout404, 2022-04-27:
I have same problem. zigbee2mqtt not able to start.
whereas ZHA is working fine with the bridge

I am using Sonoff zigbee 3 dongle

alex-bristol, 2022-05-02:
I did have same problem, on my test Raspberry Pi with Sonoff Zigbee 3.0 USB Dongle Plus.  So I did brand new fresh install of HA and latest 2022.04.0, then installed Mosquitto broker then install Zigbee2MQTT.  Still did not work, same error, "Error while starting zigbee-herdsman".
 
I spotted the serial port was missing from Zigbee2MQTT config file, even though dongle was attached when installed, so added it and now it works perfectly.  My full working config reads, the last two lines I added:
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: '-d -d'
  log: false
mqtt: {}
serial:
  port: /dev/ttyUSB0

Antx0n, 2022-05-02:
Hi Everyone,
Similar error here. Fresh install on barmetal NUC like, Sonoff Zigbee 3.0 USB Dongle Plus flashed and recognized by HA, installed mosqito, and the addon from the official repo, here is my basic conf:
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: '-d -d'
  log: false
mqtt:
  server: mqtt://192.168.1.24:1883
serial:
  port: /dev/ttyUSB0
  
  And in the configuration.yaml:
  
  homeassistant: true
mqtt:
  base_topic: zigbee2mqtt
  user: mqtt_user
  password: XXXXXX
  server: mqtt://192.168.1.24:1883
serial:
  port: /dev/ttyUSB0
  
  I have never been able to access in GUI, and in the temrinal I have:
  
  22-05-02 21:38:56 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.1:8099 ssl:default [Connect call failed ('172.30.33.1', 8099)]

atv2016, 2022-05-22:
Great, i get this now. Installed latest HA and now this, zigbee not starting because of ingress error.

Philou06, 2022-06-01:
Same thing for me, after latest HA upgrade, zigbee not strating of ingress error and all my zigbee devices are down, I tried many things but for now no one works

Philou06, 2022-06-14:
Yeah, I disabled deCONZ too but for me, that didn't solve the problem. What I had to change is to set the serial port to :
`serial:
port: /dev/ttyAMA0`
instead of:
`serial:
  port: /dev/ttyS0` as it was since 2 years.
Don't understand yet why I had to change it because I didn't change anything on my Zigbee Raspbee II hat, but I read somewhere that sometimes on RPI those port locations could be changing;
Finally, I retrieved all my Zigbee devices through a new pairing...

closed 2022-07-23 by github-actions[bot]

kokrp, 2022-08-15:
I finally found a moment to test.

Removing or disabling deCONZ from my existing set up didn't solve the issue. Error was the same.
I did a clean install on a spare card and this worked without issues. Just had to specify the proper port (check Settings-> System-> Hardware-> All hardware' and 'adapter: deconz' 

I reverted back to the original set up due to lack of time, but I plan to rebuild the system from scratch, assuming the issue will not reappear when adding other integrations or add-ons. If I learn anything from that, I'll drop that here.

t-my, 2023-02-09:
Experiencing this after updating to 1.30.1 via HA add-ons.

D3SL, 2023-03-22:
Getting this now out of the blue on version 1.30.2. Some door sensors started being a bit spotty in the past day or two. I don't think it's zigbee2mqtt because _everything_ MQTT is broken.

gabimal, 2024-01-03:
Any updates on this matter? It is still happening in Z2M version 1.35.0-1