Koenkk/zigbee2mqtt #9477: Moes HY368 thermostat (TS0601_thermostat) gives errors about unrecognized DP #13 and #115

groenmarsmannetje, 2021-11-03:
## What happened

The thermostats frequently raise these errors in the log.

```
zigbee-herdsman-converters:tuyaThermostat: Unrecognized DP #13 with data {"status":182,"transid":184,"dp":13,"datatype":5,"fn":0,"data":{"type":"Buffer","data":[0]}}
zigbee-herdsman-converters:tuyaThermostat: Unrecognized DP #115 with data {"status":182,"transid":201,"dp":115,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}}
```

## What did you expect to happen

No errors

## How to reproduce it (minimal and precise)

I added to zigbee2mqtt and after that I see the errors in the log file.

## Debug info
Zigbee2MQTT version: 1.22.0
Adapter hardware: ZZH
Adapter firmware version: 20210708

Sdahl1234, 2021-11-10:
I have the same issue. Same thermostat,, version 1.22.0-2

`zigbee-herdsman-converters:tuyaThermostat: Unrecognized DP #13 with data {"status":6,"transid":49,"dp":13,"datatype":5,"fn":0,"data":{"type":"Buffer","data":[0]}}

zigbee-herdsman-converters:tuyaThermostat: Unrecognized DP #115 with data {"status":6,"transid":74,"dp":115,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}}
`

josvink66, 2021-11-15:
Same issue here! I think the issue started after the last version.

Koenkk, 2021-11-15:
A guide can be found here on how to reverse engineer what these data points mean: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

josvink66, 2021-11-15:
Thanks @Koenkk hopefully someone can perform the steps. Unfortunately I don't have the Tuya Zigbee Gateway. So I cannot look it up.

Enzokot, 2021-11-20:
Same here and  I don't have the Tuya Zigbee Gateway too

Koterak, 2021-11-24:
I checked for myself:
DP 13: 
`Fault alarm`
[image]

DP 115:
`Window status`
[image]

So probably both are boolean

Enzokot, 2021-12-25:
Still need fix

Enzokot, 2022-01-25:
Keep alive

Enzokot, 2022-02-25:
Keep alive

Sdahl1234, 2022-03-15:
Now it seems we only need DP13 to be added 👍 
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3992

Sdahl1234, 2022-04-05:
I have a valve that very often get stuck when it calibrates during battery change.
first it reports:
zigbee-herdsman-converters:tuyaThermostat: Unrecognized DP #13 with data {"dp":13,"datatype":5,"data":{"type":"Buffer","data":[0]}}
as soon as its stuck and displays E1 it reports:
zigbee-herdsman-converters:tuyaThermostat: Unrecognized DP #13 with data {"dp":13,"datatype":5,"data":{"type":"Buffer","data":[2]}}

E1 is motor fault
E2 is sensor fault

i dont know what data is sends with sensor fault but with only 2 error codes it would be diffrent from 0 and 2

Hope someone can use this information and implement it.

Sdahl1234, 2022-04-07:
Added the valve to tuya again to verify the fault.

[image]

Sdahl1234, 2022-04-07:
[image]

Enzokot, 2022-05-08:
Keep alive

Enzokot, 2022-06-11:
The log is clean for me

Enzokot, 2022-07-17:
Still for me
`zigbee-herdsman-converters:tuyaThermostat: Unrecognized DP #13 with data {"dp":13,"datatype":5,"data":{"type":"Buffer","data":[0]}}`

Koenkk, 2022-07-17:
@Enzokot see https://github.com/Koenkk/zigbee2mqtt/issues/9477#issuecomment-968699420

Enzokot, 2022-07-18:
> @Enzokot see [#9477 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/9477#issuecomment-968699420)

Sorry, but I don't have the Tuya Gateway

closed 2022-08-26 by github-actions[bot]

groenmarsmannetje, 2024-03-17:
Reopened. I still see this error in logging