Koenkk/zigbee2mqtt #9480: Can't set "power_outage_memory" if "optimistic"=False

zen2, 2021-11-03:
**Zigbee2mqtt: 1.22.0
Coordinator: 1352P-2
Firmware: 20210708**

Recently I've added several plugs _Tuya TS011F_plug_1_ and _TS011F_plug_3_.
I could have set _"power_outage_memory"_ on few of them but couldn't on all of them.
The fact that I set _"optimistic"=False_ for all this plugs give me the idea to try to set the _"power_outage_memory"_ with "optimistic"=True and that have worked. As I have a bunch of these plugs, I've really test it with different scenarios and the conclusion is: **It's not possible to modify _"power_outage_memory"_ if  _"optimistic"=False_.**

I think this problem have appears before 1.22.0. I got it with 1.21.2-dev too but I find the correlation between  _"power_outage_memory"_ and _"optimistic"_ while running on 1.22.0.

taimaiduc, 2021-12-21:
Up

Koenkk, 2021-12-21:
Since this setting is not reported by the device, it will indeed not be updated on the MQTT side (that is the whole point of optimistic false). The setting however should work correctly even when optimistic false is set.

zen2, 2021-12-21:
> Since this setting is not reported by the device, it will indeed not be updated on the MQTT side (that is the whole point of optimistic false). The setting however should work correctly even when optimistic false is set.

I understand the point of optimistic option but the problem is that if it set to false, you can't set power_outage_memory.
If you set it, the value doesn't change at all. Is it expected behaviour ?

Koenkk, 2021-12-21:
> If you set it, the value doesn't change at all. Is it expected behaviour ?

Yes, since the device not report this issue itself and you disabled the optimistic reporting.

zen2, 2021-12-21:
> > If you set it, the value doesn't change at all. Is it expected behaviour ?
> 
> Yes, since the device not report this issue itself and you disabled the optimistic reporting.

I didn't see from log the obvious connection between optimistic and set value.
But all right so I know it now. I close the ticket. Thanks.

closed 2021-12-21 by zen2

Inferi0r, 2021-12-21:
> > If you set it, the value doesn't change at all. Is it expected behaviour ?
> 
> Yes, since the device not report this issue itself and you disabled the optimistic reporting.

You sure about it actually incorporates the changed value? As stated here by several people (I bought and tested 10 of myself together with someone else) https://github.com/Koenkk/zigbee2mqtt/issues/9987 for the "manufacturerName":"_TZ3000_w0qqde0g","model":"TS011F_plug_3" variant the changed power_outage_memory states are reported back correctly but the real state is always 'off' - the most unfortunate option.

zen2, 2021-12-23:
So I've to check on my side the behaviour of all my plugs since I got 15 of them:
- 5 x  _TZ3000_dpo1ysak TS011F_plug_3 (square white)
- 5 x _TZ3000_w0qqde0g TS011F_plug_3 (round orange)
- 1 x _TZ3000_cphmq0q7 TS011F_plug_1 (square white)

I will test the power_outage_memory of all of them and will report back here.

sergey-1976, 2022-01-17:
I have 3 x TS011F_plug_3 _TZ3000_typdpbpg (round orange) - changed power_outage_memory states are reported back correctly but the real state is always 'off'. :(
And 1 x TS011F_plug_3 _TZ3000_typdpbpg (square white) - real state is correct.