Koenkk/zigbee2mqtt #9486: Moes BRT-100-TRV didnt update local_temperature after local_temperature_calibration is set

KartoffelToby, 2021-11-03:
If local_temperature_calibration  is set, local_temperature should be updated instandly with the new value.

craiq, 2021-11-09:
confirm, I've the same problem. local_temperature_calibration has no effect, after hours too.
With the older TS0601_thermostat this works just fine.

IvanUpa, 2021-11-09:
Same problem. Two BRT-100-TRV

craiq, 2021-11-09:
Any hints how to fix this?

IvanUpa, 2021-11-10:
A workaround to calibrate from an external temperature sensor is to ignore the local_temperature_calibration.
It works like this for me: {{states (temp_sensor) | float (0) | round (0) - (state_attr (climate_device, 'current_temperature') | float (0) | round (0))}}

craiq, 2021-11-10:
@IvanUpa how are you using this temperature difference then?
I want to set the target temperatur via the climate entity or manual on the trv

IvanUpa, 2021-11-10:
> @IvanUpa how are you using this temperature difference then? I want to set the target temperatur via the climate entity or manual on the trv

I can't say for sure why this works. As if, if you do not touch TRV, then "local_temperature" is returned without correction.
In any case, the final problem is solved, for a week in the z2m the data is correct, and the temperature in the rooms is maintained according to the settings.

KartoffelToby, 2021-11-10:
But "tvr" Not touched is the Problem, as soon as the tvr wake up from sleep state the Temperature ist correctly updated, so my idea is to fix this in z2m
with the following Logic:

if the calibration is set, update the current temp with it internal in z2m if a event direcly from the device is fired Override the current temp. if everythig is correct the current temp dosent changed.

KartoffelToby, 2021-11-10:
btw the calibration isn't reported to HA and i don't know why.

other tvr works great.

but i think the "instand" intern Update with the current Temperature and calibration should be done on any Thermostats

IvanUpa, 2021-12-12:
The problem is relevant.

yavasilek, 2021-12-22:
Проблема актуальна.

mrybas, 2021-12-29:
i have the same issue
>  {{states (temp_sensor) | float (0) | round (0) - (state_attr (climate_device, 'current_temperature') | float (0) | round (0))}}

@IvanUpa 
how do you use this value later?
write in `local_temperature_calibration`

IvanUpa, 2021-12-29:
@mrybas 
I do not know, stopped using this method. Try integration:
https://github.com/KartoffelToby/ai_thermostat

eraycetinay, 2022-01-22:
It is a big problem if you are using an external temperature sensor.  It is not updating the latest temperature after the calibration.  So home assistant is not able to know what is the latest temperature on trv. and that makes the external sensor usage meaningless. 

Probably it is going to work if you are going to make a calibration setting once every hour.  But if you want to make calibration settings once every 5 minutes you will broke the trv system. 

so, i want to make a solution to this by using automations but im not able to see when local_temperature is updated.  climate.trv object is updated after calibration but i cant be sure when the local_temperature is updated exactly. so i believe, we can create automation if we have a chance to catch the local_temperature update's exact moment.  any idea to make this done?

closed 2022-03-29 by github-actions[bot]

Kettenknecht, 2022-04-14:
Problem still exists

rotilho, 2022-07-25:
Are there any solution for this or returning is the only option?

Der-Nax, 2022-09-25:
> It is a big problem if you are using an external temperature sensor. It is not updating the latest temperature after the calibration. So home assistant is not able to know what is the latest temperature on trv. and that makes the external sensor usage meaningless.
> 
> Probably it is going to work if you are going to make a calibration setting once every hour. But if you want to make calibration settings once every 5 minutes you will broke the trv system.
> 
> Update: I fixed my problems. I stopped using trv's temperature control and i created new virtual climate devices by 'generic_thermostat' and i used boolean inputs and external temperature sensors on that virtual climate devices. Created a script which sets the real brt climate devices to 5 degree and 35 degree according the position of boolean switch's on/off state. So i used my real trv's like a switch based water heater in that way. At the end of the day, my whole house has a perfect heating system now. The only negative point is; im not able to use brt's smart positioning feature. Now its always %100 or %0. At least i have a system that is more sensitive to temperature changes and works healthier.

Could you please share your script and settings of 'generic_thermostat'?

willhaggan, 2022-09-26:
having the same problem with the local temperature calibration setting and also the update of the local temperature seems very slow back to z2m

Alex2zl, 2022-10-10:
> It is a big problem if you are using an external temperature sensor. It is not updating the latest temperature after the calibration. So home assistant is not able to know what is the latest [...]

Please show your configuration of the created new virtual climate device.
I think many users will find this information useful. Thanks in advance

eraycetinay, 2022-10-10:
sorry for the delay.

in configuration yaml i added a virtual trv. 
i created an input_boolean (on/off) and added it to heater param.  (input_boolean.trv1_durum)
also i added the external thermometer sensor to target_sensor param.  (sensor.termometre_1_temperature)

```
climate:
  - platform: generic_thermostat
    name: TRV1
    heater: input_boolean.trv1_durum
    target_sensor: sensor.termometre_1_temperature
    min_temp: 10
    max_temp: 30
    ac_mode: false
    cold_tolerance: 0.25
    hot_tolerance: 0.5
    initial_hvac_mode: "heat"
    precision: 0.5
```

for sync between virtual and real trv i used 2 automation.
first automation triggers when boolen status changes.

```
alias: Trv1_sync
description: ""
trigger:
  - platform: state
    entity_id: input_boolean.trv_status
condition: []
action:
  - service: climate.set_temperature
    data:
      temperature: |-
        {% if is_state('input_boolean.trv1_durum','off') %}
          5
        {% elif is_state('input_boolean.trv1_durum','on') %}
          35
        {% endif %}
    target:
      device_id: DEVICE_ID_OF_REAL_TRV
mode: single
```

second automation might not be necessary but its effective on problems created by first automation
so second automation runs the first automation every hour.  
basically, virtual trv will change the status between on and off of input_boolean
and when input boolean status changed, first automation will set the degree to 5 or 35 of real trv
if degree is 5, real trv is not going to work. if it is 35, it will work.   
also second automation is a life vest. it will fix problem if any problem occurs on first automation.
hassio restarts, electric cuts etc might create problems. so use both for a better results.

```
alias: Trv_sync
description: ""
trigger:
  - platform: time_pattern
    hours: /1
    minutes: "0"
    seconds: "0"
condition: []
action:
  - service: automation.trigger
    data: {}
    target:
      entity_id:
        - automation.trv1_sync
        - automation.trv2_sync (if you have)
        - automation.trv3_sync (if you have)
        - automation.trv4_sync (if you have)
mode: single
```

if you have multiple trv's like mine.  create multiple from the 'first automation' and generic_thermostat virtual climate for each. 1 secondary automation will be enough for all (every hour fixer)

i used this for 6 months and works perfectly. 
i hope it helps.
cheers.