Koenkk/zigbee2mqtt #9489: Schneider Electric CCTFR6700 don't expose energy

stephaneSBO, 2021-11-03:
All devices CCTFR6700 don't expose Energy 

[image]

`2021-11-01 12:52:03: MQTT publish: topic 'zigbee2mqtt/FilPilote-ChambreParents', payload '{"device":{"friendlyName":"FilPilote-ChambreParents","ieeeAddr":"0x000b3cfffeff8f35","manufacturerID":4190,"manufacturerName":"Schneider Electric","model":"CCTFR6700","networkAddress":1439,"powerSource":"Mains (single phase)","type":"Router","zclVersion":3},"energy":null,"linkquality":69,"local_temperature":-327.68,"occupied_heating_setpoint":30,"pi_heating_demand":13,"power":33,"schneider_pilot_mode":"pilot","system_mode":"heat"}'`

Version Zigbee2MQTT
1.22.0 commit: 0a1fccf
Type de Coordinateur
zStack3x0
Révision du Coordinateur
20210708
Version Interface
0.6.30

Are there any CCTFR6700 users with the energy return ? This equipment does report update firmware version, I dont know if there.

closed 2021-12-11 by github-actions[bot]

ShellCode33, 2022-12-20:
Can confirm, `energy` remains `null`.

Thezap, 2023-11-23:
This is still the case, any clue on who to fix it ?

kerin444, 2023-11-28:
Same here, working ok with HomeAssistant ZHA but null with Z2M

jeremylvln, 2024-03-19:
Same here :x

Thezap, 2024-03-20:
Hello after some digging I made up working in Z2M. 
The bug is very strange but resolved itself when you make a manual query (with the dev console pages)

By making a read like this the correct value for  `currentSummDelivered` is return and after 1 minute, this value is correctly reported in the expose pages

Point de terminaison = 2
cluster = seMetering
attribute: 
- summaFormatting
- status
- currentSummDelivered
- unitOfMeasure
- multiplier
- divisor

<details>
  <summary>Some screenshot (UI in French)</summary>
  
[image]

[image]

</details>