Koenkk/zigbee2mqtt #9490: availability pings are sent to removed devices

okastl, 2021-11-03:
## What happened
After removing devices, the new availability check continues to ping the now non-existing devices by their ieee address. The friendly name doesn't exist anymore.

## What did you expect to happen
Stop pinging non existent devices.

## How to reproduce it (minimal and precise)
Enable availability check. The simple boolean one is enough. Restart z2m for the change to effect.
Wait a while until all devices have received their ping.
Remove one of the devices. Wait for z2m to ping the non existent device by the ieee address.

## Debug info
Zigbee2MQTT version: 1.22.0 commit: cf68507e
Adapter hardware: CC26X2R1 (slaesh)
Adapter firmware version: 20210708

Koenkk, 2021-11-04:
Should be fixed now, please confirm.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

okastl, 2021-11-04:
> Should be fixed now, please confirm.
> 

I tried 1.22.0-dev commit: a09aa424, and the problem is fixed. Thank you.

closed 2021-11-05 by okastl