Koenkk/zigbee2mqtt #9491: Sonoff Bridge - adapter:ezsp error

2wheelsnz, 2021-11-04:
Hi there.
I installed Zigbee2mqtt Edge  v1.22.0-1 within in Home Assistant yesterday with hope of using with Sonoff Brdge that has been flashed with Tasmota firmware.
I followed the instructions to change the config file to the following:
    port: tcp://192.168.1.93:8888
    adapter: ezsp

However, I get an error and can't save the configuration file.

Failed to save add-on configuration, does not match regular expression ^zstack|deconz|zigate$. 
  Got {'data_path': '/config/zigbee2mqtt', 'external_converters': [], 'devices': 'devices.yaml', 
  'groups': 'groups.yaml', 'homeassistant': True, 'permit_join': False, 'mqtt': {'base_topic': 
  'zigbee2mqtt', 'server': 'mqtt://mqtt.gardenshed.nz:1883'}, 'serial': {'port': 'tcp://192.168.1.93:8888', 
  'adapter': 'ezsp'}, 'advanced': {'log_level': 'debug', 'pan_id': 6754, 'channel': 11, 
  'network_key': [1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13], 'availability_blocklist': [], 'availability_passli...

If I have the port as just the tcp://192.168.1.93:8888 I can save the file but zigbee2mqtt won't start.
As soon as I try to add the adapter: ezsp is when I get the error.

What I have found is that if I leave the configuration file as default, I can start zigbee2mqtt and then with Visual Studio Code  I can manually modify the /config/zigbee2mqtt/configuration.yaml file and change the serial information to this:
serial:
  port: tcp://192.168.1.93:8888
  adapter: ezsp

Zigbee2mqtt will then communicate with the Sonoff bridge and I can bind my zigbee devices.  Great.  However, if Home Assistant or zigbee2mqtt is restarted, then my manual change is lost.

## Debug info
Zigbee2MQTT version: 1.22.0-1
Adapter hardware: Sonoff Bridge
Adapter firmware version: Tasmota 9.5.0.2

Kind regards
Barry

jjiik, 2021-11-08:
Same here with Silvercrest (Lidl) ZigBee GW...

jjiik, 2021-11-08:
After switching to GitHub repo and reinstalling Edge all is working as expected.

Hedda, 2021-11-17:
EZSP support is still in a very experimental phase so you should really post your feedback here -> Koenkk/zigbee-herdsman#319

I do not think that any developer is currently actively working on improving or maintaining that experimental support(?).

The only tip I got is to try upgrade EmberZNet NCP UART EZSP firmware Zigbee firmware version 6.7.9 

https://github.com/arendst/Tasmota/tree/development/tools/fw_SonoffZigbeeBridge_ezsp

Also, please take note that WiFi based Zigbee bridges are now not recommended with the EZSP protocol. See the warnings about it:

https://www.zigbee2mqtt.io/guide/adapters/#notes

WiFi-based Serial-to-IP bridges are not recommended for Silicon Labs EZSP adapters as the serial protocol does not have enough fault-tolerance to handle packet loss or latency delays that can normally occur over WiFi connections. If cannot use a locally connected USB or UART/GPIO adapter then the recommendation is to use remote adapter that connected via Ethernet (wired) to avoid issues with EZSP caused by WiFi connections.

https://www.zigbee2mqtt.io/advanced/remote-adapter/connect_to_a_remote_adapter.html

WiFi-based Serial-to-IP bridges are not recommended for Silicon Labs EZSP adapters as the serial protocol does not have enough fault-tolerance to handle packet loss or latency delays that can normally occur over WiFi connections.

More warning about Zigbee to WiFi bridges for EZSP is not recommended for production as it does not do well with packed loss:

https://github.com/zigpy/bellows#warning-about-zigbee-to-wifi-bridges

https://www.home-assistant.io/integrations/zha/#warning-about-wi-fi-based-zigbee-to-serial-bridgesgateways

FYI, I also understand that the EFR32 board antenna design of the "SM-011 V1.0" Zigbee module used inside ITead's Sonoff ZBBridge have very high receive sensitivity to electromagnetic interference (EMI) / radio-frequency interference. This means that it will be extra important to keep it away from any other electronic devices or appliances that may generate an electromagnetic field (EM Field) or send our interfering radio signals. That includes the computer that you run your home automation application software on as well as any devices such as for example external harddrives or storage devices, power-supplies, etc., and of course any Wi-Fi access points or routers nearby.

Note! The new ITead Zigbee 3.0 USB Dongle Model: 9888010100045 (Hardware Revision Version 1.3) circuit board share the same type of antenna design so its radio is as well extremely sensitive to signal interference on its receiving end. The current recommendation for using it is to connect it via a long USB extension cable and try to move it as far away from all electronics and other possible sources of signal interference.

PS: Another general tip that always applies when building a Zigbee network is to begin by adding a few good mains-powered Zigbee router devices so that they will make use a stable baseline. Many start by buying three "IKEA TRADFRI Signal Repeater" (E1746).

https://zigbee.blakadder.com/Ikea_E1746.html

https://www.ikea.com/us/en/p/tradfri-signal-repeater-30400407/

https://www.ikea.com/gb/en/p/tradfri-signal-repeater-80424255/

https://www.ikea.com/se/sv/p/tradfri-signalfoerstaerkare-10400408/

closed 2022-01-16 by github-actions[bot]