Koenkk/zigbee2mqtt #9492: is there any way to unban device banned while deletion

lotohov, 2021-11-04:
i accidentaly push ban device while removing it from network.
and cant find any way to get them back . putting it on whitelist and reinstalling from scratch doesn help

## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

danieledwardgeorgehitchcock, 2021-11-04:
If you edit your configuration.yaml, the blacklist is at the bottom and you should find your device in there

lotohov, 2021-11-04:
configuration.yaml refer to homeassistant. i just reinstall all from scrath. and dropped mqtt topics related to zigbee2mqtt

closed 2021-12-13 by github-actions[bot]