Koenkk/zigbee2mqtt #9493: power plug ZigBee EU sparkle

snowmassi, 2021-11-04:
please support

0xa4c138c05e64c38f
TS011F
_TZ3000_nfnmi125
0x81E8

thanks

GimpArm, 2021-11-28:
I think its just a simple switch, here's a device definition that works for me.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');

const definition = {
  fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_nfnmi125'}],
  model: 'TS011F',
  vendor: 'Sparkleiot',
  description: '10A smart switch',
  fromZigbee: [fz.on_off],
  toZigbee: [tz.on_off],
  exposes: [exposes.presets.switch()],
};

module.exports = definition;
```

simie7, 2021-11-28:
Hello, sharing my config:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS011F'],
    model: '_TZ3000_nfnmi125',
    vendor: 'Shenzhen Sparkle IoT',
    whiteLabel: [{vendor: 'Sparkle IoT', model: 'XH-002P'}],
    description: 'Zigbee Smart Plug',
    supports: 'on/off, power measurement',
    fromZigbee: [
        fz.ignore_basic_report,
        fz.on_off,
        fz.electrical_measurement,
        fz.metering,
        fz.tuya_switch_power_outage_memory,
    ],
    toZigbee: [
        tz.on_off,
        tz.tuya_switch_power_outage_memory,
    ],
    exposes: [
        e.switch(),
        e.power(),
        e.current(),
        e.voltage().withAccess(ea.STATE),
        e.energy(),
        exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore']).withDescription('Recover state after power outage'),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
            acVoltageMultiplier: 1, acVoltageDivisor: 1, acCurrentMultiplier: 1, acCurrentDivisor: 1000, acPowerMultiplier: 1,
            acPowerDivisor: 1,
        });
    },
};

module.exports = definition;
```
The problem is, that I can't get energy consumption values. I can't query anything energy-related from `haElectricalMesurement` cluster. 

And I get an error in log:
```
debug 2021-11-28 20:49:10: Received Zigbee message from 'zSocket02', type 'raw', cluster '57344', data '{"data":[8,132,10,2,208,72,2,0,10],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2021-11-28 20:49:10: No converter available for '_TZ3000_nfnmi125' with cluster '57344' and type 'raw' and data '{"data":[8,132,10,2,208,72,2,0,10],"type":"Buffer"}'
```
Cluster `57344` in DEC is `E000` in HEX. I wasn't able to find it in 
`node_modules/zigbee-herdsman/dist/zcl/definition/cluster.js`, 
so my guess is that its not supported yet.

Info from `database.db`:
```
{"id":2,"type":"Router","ieeeAddr":"0xa4c1384a8c57f201","nwkAddr":11932,"manufId":4417,"manufName":"_TZ3000_nfnmi125","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53264":2,"switchType":0}},"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_nfnmi125","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}},"haElectricalMeasurement":{"attributes":{"acVoltageMultiplier":1,"acVoltageDivisor":1,"acCurrentMultiplier":1,"acCurrentDivisor":1000,"acPowerMultiplier":1,"acPowerDivisor":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ccca8b0","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b001ccca8b0","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b001ccca8b0","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-917712964},"lastSeen":1638132747889}
```

GimpArm, 2021-11-29:
The only one that I can get working from your config is the power_outage_memory. I don't know what the z2m policy is about submitting devices with broken functionality. Do you remove what doesn't work currently or leave it until the core has the functionality implemented? That's why I only did on/off, so on/off + power_outage_memory are the only parts that currently work for me.

simie7, 2021-11-29:
I did full configuration from examples, expecting that it will work. But yes, you are right - only those things are currently working, so up to you which part of configuration to keep.

bertkuh, 2021-12-01:
Hi thanks for what you did so far. I can only get the code working if i remove the line
`const extend = require('zigbee-herdsman-converters/lib/extend');`
but i guess it makes no difference if the power management is not available anyway

Koenkk, 2021-12-01:
@simie7 

> The problem is, that I can't get energy consumption values. I can't query anything energy-related from haElectricalMesurement cluster.

What is logged in the debug log when refreshing the power value from the frontend?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

simie7, 2021-12-01:
Hi @Koenkk , 

pressing yellow `Reconfigure`, I am getting only those messages:
```
debug 2021-12-01 19:34:06: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"zSocket01","transaction":"eys19-11"}'
info  2021-12-01 19:34:06: Configuring 'zSocket01'
info  2021-12-01 19:34:06: Successfully configured 'zSocket01'
info  2021-12-01 19:34:06: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"zSocket01"},"status":"ok","transaction":"eys19-11"}'
```
Refreshing values from frontend (Devices/device_id/Exposes):
```
debug 2021-12-01 19:35:29: Received MQTT message on 'zigbee2mqtt/zSocket01/get' with data '{"state":""}'
debug 2021-12-01 19:35:29: Publishing get 'get' 'state' to 'zSocket01'
info  2021-12-01 19:35:29: MQTT publish: topic 'zigbee2mqtt/zSocket01', payload '{"current":null,"energy":null,"last_seen":1638380129512,"linkquality":47,"power":null,"power_outage_memory":"restore","state":"OFF","voltage":null}'
debug 2021-12-01 19:35:29: Received Zigbee message from 'zSocket01', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2021-12-01 19:35:29: MQTT publish: topic 'zigbee2mqtt/zSocket01', payload '{"current":null,"energy":null,"last_seen":1638380129512,"linkquality":44,"power":null,"power_outage_memory":"restore","state":"OFF","voltage":null}'
```
But sometimes I am getting those messages in log:
```
debug 2021-11-28 20:49:10: Received Zigbee message from 'zSocket02', type 'raw', cluster '57344', data '{"data":[8,132,10,2,208,72,2,0,10],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2021-11-28 20:49:10: No converter available for '_TZ3000_nfnmi125' with cluster '57344' and type 'raw' and data '{"data":[8,132,10,2,208,72,2,0,10],"type":"Buffer"}'
```
Some details [above](https://github.com/Koenkk/zigbee2mqtt/issues/9493#issuecomment-981153732).

Koenkk, 2021-12-02:
> Refreshing values from frontend (Devices/device_id/Exposes):

I see you do a get on the state:

```
debug 2021-12-01 19:35:29: Received MQTT message on 'zigbee2mqtt/zSocket01/get' with data '{"state":""}'
debug 2021-12-01 19:35:29: Publishing get 'get' 'state' to 'zSocket01'
```

Can you do a get on the power?

Reading https://github.com/zigpy/zha-device-handlers/issues/910 I get the impression that this device does not support power measurements, are you sure it actually does?

simie7, 2021-12-02:
Sorry for dumb question, but how to do it from GUI? Inside device menu I can't  find a way of quering this data. I've got a state as there is refresh button under `Exposes` menu. I can't query from dev console, as values from `haElectricalMeasurement` fail:
```
debug 2021-12-02 12:32:34: Received MQTT message on 'zigbee2mqtt/zSocket01/1/set' with data '{"read":{"attributes":["rmsVoltage"],"cluster":"haElectricalMeasurement","options":{}}}'
debug 2021-12-02 12:32:34: Publishing 'set' 'read' to 'zSocket01'
error 2021-12-02 12:32:57: Publish 'set' 'read' to 'zSocket01' failed: 'Error: Read 0xa4c1388a487cc633/1 haElectricalMeasurement(["rmsVoltage"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 3135 - 1 - 254 - 2820 - 1 after 10000ms)'
debug 2021-12-02 12:32:57: Error: Read 0xa4c1388a487cc633/1 haElectricalMeasurement(["rmsVoltage"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 3135 - 1 - 254 - 2820 - 1 after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
```
IDK how to query otherwise?
Regarding support of energy monitoring - seller says [yes](https://www.aliexpress.com/item/1005002761054613.html?spm=a2g0s.9042311.0.0.5fdc4c4dIgBO9c), best proof would be to pair with original Tuya gateway, but I don't have it at the moment.

simie7, 2021-12-02:
Removed as external converter and added `manufacturerName` in `tuya.js`:
```
    {
        fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_cphmq0q7', applicationVersion: 67},
            {modelID: 'TS011F', manufacturerName: '_TZ3000_ew3ldmgx'},
            {modelID: 'TS011F', manufacturerName: '_TZ3000_ps3dmato'},
            {modelID: 'TS011F', manufacturerName: '_TZ3000_mraovvmm'},
            {modelID: 'TS011F', manufacturerName: '_TZ3000_jvzvulen'},
            {modelID: 'TS011F', manufacturerName: '_TZ3000_nfnmi125'}],

        model: 'TS011F_plug_1',
        description: 'Smart plug (with power monitoring)',
        vendor: 'TuYa',
        whiteLabel: [{vendor: 'LELLKI', model: 'TS011F_plug'}],
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
        toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.rmsVoltage(endpoint, {change: 5});
            await reporting.rmsCurrent(endpoint, {change: 50});
            await reporting.activePower(endpoint, {change: 10});
            await reporting.currentSummDelivered(endpoint);
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            device.save();
        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
            e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
                .withDescription('Recover state after power outage')],
    },

```
Got some logs trying to reconfigure:
```
debug 2021-12-03 00:21:25: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"zSocket01","transaction":"9e9i1-7"}'
info  2021-12-03 00:21:25: Configuring 'zSocket01'
error 2021-12-03 00:21:48: Failed to configure 'zSocket01', attempt 2 (Error: ConfigureReporting 0xa4c1388a487cc633/1 haElectricalMeasurement([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 3135 - 1 - 12 - 2820 - 7 after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7))
info  2021-12-03 00:21:48: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"zSocket01"},"error":"Failed to configure (ConfigureReporting 0xa4c1388a487cc633/1 haElectricalMeasurement([{\"attribute\":\"rmsVoltage\",\"minimumReportInterval\":5,\"maximumReportInterval\":3600,\"reportableChange\":5}], {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 3135 - 1 - 12 - 2820 - 7 after 10000ms))","status":"error","transaction":"9e9i1-7"}'
```

Koenkk, 2021-12-03:
https://github.com/Koenkk/zigbee2mqtt/issues/9493#issuecomment-984503465 proofs it doesn't support reporting energy measurements in the same way as other TuYa plugs do. Getting more convinced this device does not support power measurements.

stefanschaedeli, 2021-12-08:
I seem to have exactly the same case, but with even another "manufacture name)

` Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_w0qqde0g'

Any straight forward way to get this addional name into support?

simie7, 2021-12-08:
@stefanschaedeli you can try to add your device id [here](https://github.com/Koenkk/zigbee2mqtt/issues/9493#issuecomment-985051624) or add custom converter like [this](https://github.com/Koenkk/zigbee2mqtt/issues/9493#issuecomment-981153732), changing id to yours. You will now if it works only after trying.

stefanschaedeli, 2021-12-08:
> @stefanschaedeli you can try to add your device id [here](https://github.com/Koenkk/zigbee2mqtt/issues/9493#issuecomment-985051624) or add custom converter like [...]

Maybe a stupid question but I just don't fined the the way at the moment. How can I modify the tuya.js in the container hosted environment. This file seems to be part of the non persitant volume and I apply the change persistant...

filippopetri, 2021-12-10:
Hi, I have the same plug TS011F with manufacturer name: _TZ3000_ew3ldmgx.
The pairing works flawlessly and the device shows up as supported. The switch works and the power_outage_memory works as well, but I cannot get the energy values with Z2M. I tried pairing with ZHA and I can get all the correct energy, power and voltage values so the plug can measure power as specified in the aliexpress seller description.

[image]

[image]

bertkuh, 2021-12-10:
Hi 
I got them working just yesterday
I went with the :latest-dev version and after i restartet everything AND reconnecting my zigbee stick it works. Remember to put power_outage_memory to off if you want to see values and it talkes a while sometimes.

linuxrrze, 2021-12-14:
> Maybe a stupid question but I just don't fined the the way at the moment. How can I modify the tuya.js in the container hosted environment. This file seems to be part of the non persitant volume and [...]

Same problem here - running zigbee2mqtt in docker container. Restarting the container reverts the modifications to tuya.js.

My device is (labeled "Blitzwolf SHP15"):

0x84ba20fffe77b83e
TS011F / _TZ3000_u5u4cakc

ShadNex, 2021-12-15:
> Same problem here - running zigbee2mqtt in docker container. Restarting the container reverts the modifications to tuya.js.
> 
> My device is (labeled "Blitzwolf SHP15"):
> 
> 0x84ba20fffe77b83e TS011F / _TZ3000_u5u4cakc

Same situation for me. Exact same manuafacturerName _TZ3000_u5u4cakc as you have. Power on-off works with the custom converter, but no measurements being reported.

ShadNex, 2021-12-16:
I found out something:
The measurements are being reported my zigbee2mqtt when manually querying the attributes via the dev console:
[image]

They then also update in the device detail page:
[image]

So my guess is that zigbee2mqtt simply doesn't periodically query the device about the measurements and so they're stuck at null values?

This is the converter i used:

```
module.exports = [
{
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_u5u4cakc'}],
    model: 'BW-SHP15',
    description: 'Smart plug (with power monitoring)',
    vendor: 'BlitzWolf',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
    toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.rmsVoltage(endpoint, {change: 5});
        await reporting.rmsCurrent(endpoint, {change: 50});
        await reporting.activePower(endpoint, {change: 10});
        await reporting.currentSummDelivered(endpoint);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
    exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
        e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
            .withDescription('Recover state after power outage')],
}
];
```

Koenkk, 2021-12-16:
Can you check if power measurements work now in the latest-dev branch? If not please post your data/database.db entry of this device?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

ShadNex, 2021-12-17:
> Can you check if power measurements work now in the latest-dev branch? If not please post your data/database.db entry of this device?
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Installed the Edge Version on Home Assistant and re-paired the plug. Still the same issue: Power on and off work, but measurements are not reported. Manually refreshing via the dev console still works.

Database entry:
```
{
   "id":4,
   "type":"Router",
   "ieeeAddr":"0x84ba20fffe77b1f1",
   "nwkAddr":9034,
   "manufId":4098,
   "manufName":"_TZ3000_u5u4cakc",
   "powerSource":"Mains (single phase)",
   "modelId":"TS011F",
   "epList":[
      1,
      242
   ],
   "endpoints":{
      "1":{
         "profId":260,
         "epId":1,
         "devId":266,
         "inClusterList":[
            0,
            3,
            4,
            5,
            6,
            1794,
            2820,
            57344,
            57345
         ],
         "outClusterList":[
            25,
            10
         ],
         "clusters":{
            "genBasic":{
               "attributes":{
                  "modelId":"TS011F",
                  "manufacturerName":"_TZ3000_u5u4cakc",
                  "stackVersion":0,
                  "dateCode":""
               }
            },
            "haElectricalMeasurement":{
               "attributes":{
                  "acCurrentDivisor":1000,
                  "acCurrentMultiplier":1
               }
            },
            "seMetering":{
               "attributes":{
                  "divisor":100,
                  "multiplier":1
               }
            }
         },
         "binds":[
            {
               "cluster":6,
               "type":"endpoint",
               "deviceIeeeAddress":"0x00124b0024c2b2ae",
               "endpointID":1
            },
            {
               "cluster":2820,
               "type":"endpoint",
               "deviceIeeeAddress":"0x00124b0024c2b2ae",
               "endpointID":1
            },
            {
               "cluster":1794,
               "type":"endpoint",
               "deviceIeeeAddress":"0x00124b0024c2b2ae",
               "endpointID":1
            }
         ],
         "configuredReportings":[
            {
               "cluster":2820,
               "attrId":1285,
               "minRepIntval":5,
               "maxRepIntval":3600,
               "repChange":5
            },
            {
               "cluster":2820,
               "attrId":1288,
               "minRepIntval":5,
               "maxRepIntval":3600,
               "repChange":50
            },
            {
               "cluster":2820,
               "attrId":1291,
               "minRepIntval":5,
               "maxRepIntval":3600,
               "repChange":10
            },
            {
               "cluster":1794,
               "attrId":0,
               "minRepIntval":5,
               "maxRepIntval":3600,
               "repChange":[
                  1,
                  1
               ]
            }
         ],
         "meta":{
            
         }
      },
      "242":{
         "profId":41440,
         "epId":242,
         "devId":97,
         "inClusterList":[
            
         ],
         "outClusterList":[
            33
         ],
         "clusters":{
            
         },
         "binds":[
            
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      }
   },
   "appVersion":69,
   "stackVersion":0,
   "hwVersion":1,
   "dateCode":"",
   "zclVersion":3,
   "interviewCompleted":true,
   "meta":{
      "configured":680236614
   },
   "lastSeen":1639723047242
}
```

petrus0x13, 2021-12-17:
@Koenkk I think your change in dev branch is regarding the device with mn: '_TZ3000_nfnmi125' (as in original request), but @ShadNex seems to have '_TZ3000_u5u4cakc'. Generally I think many different devices has been mentioned in this topic:

_TZ3000_nfnmi125
_TZ3000_u5u4cakc
_TZ3000_ew3ldmgx
_TZ3000_w0qqde0g

In my case _TZ3000_nfnmi125 doesn't respond to manual request performed by dev console. But maybe it will be worth to try adding _TZ3000_u5u4cakc to the pooled type.

ShadNex, 2021-12-17:
> @Koenkk I think your change in dev branch is regarding the device with mn: '_TZ3000_nfnmi125' (as in original request), but @ShadNex seems to have '_TZ3000_u5u4cakc'. Generally I think many [...]

Right - I'm still using the custom converter from my previous comment. But would it make any difference? I've simply copied the part from the tuya.js file into a separate one with the ID of my device.
@petrus0x13 did you try the manual request while the device is turned on? For me it didn't work either right after re-adding the device, but only after turning it once. After that, the query worked also while the device was turned off.

filippopetri, 2021-12-17:
I can confirm that manually requesting the data works flawlessly for my _TZ3000_ew3ldmgx plugs as suggested by @ShadNex

Koenkk, 2021-12-17:
Did some more fixed, please check if it work in the latest-dev in 4 hours from now. If it does **not** work:
- Is it detected as a `TS011F_plug_3`? (and **not** `TS011FF_plug_1`)
- What is the data/database.db entry of this device?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

capelito, 2021-12-17:
Thank you so much for looking at this. I have multiple _TZ3000_nfnmi125 power plugs. Unfortunately, they still don't work with this update (I'm using the current dev version (zigbee-herdsman-converters 14.0.348). Reading via the dev console does not work either.

This is the output of the dev console:
`Publish 'set' 'read' to '0xa4c13870026832c8' failed: 'Error: Read 0xa4c13870026832c8/1 haElectricalMeasurement(["activePower"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 39299 - 1 - 49 - 2820 - 1 after 10000ms)'`

And this is the entry in the database:
`{
  "id": 8,
  "type": "Router",
  "ieeeAddr": "0xa4c13870026832c8",
  "nwkAddr": 64727,
  "manufId": 4417,
  "manufName": "_TZ3000_nfnmi125",
  "powerSource": "Mains (single phase)",
  "modelId": "TS011F",
  "epList": [
    1,
    242
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 266,
      "inClusterList": [
        3,
        4,
        5,
        6,
        57344,
        57345,
        0
      ],
      "outClusterList": [
        25,
        10
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "stackVersion": 0,
            "dateCode": "",
            "zclVersion": 3
          }
        },
        "genOnOff": {
          "attributes": {
            "20480": 1,
            "32768": 0,
            "onOff": 0,
            "onTime": 0,
            "offWaitTime": 0,
            "tuyaBacklightMode": 1,
            "moesStartUpOnOff": 0
          }
        },
        "manuSpecificTuya_3": {
          "attributes": {
            "53264": 0,
            "switchType": 0
          }
        }
      },
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [],
      "outClusterList": [
        33
      ],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 65,
  "stackVersion": 0,
  "hwVersion": 1,
  "dateCode": "",
  "zclVersion": 3,
  "interviewCompleted": true,
  "meta": {},
  "lastSeen": 1639679488623,
  "useImplicitCheckin": true
}`

ShadNex, 2021-12-18:
@Koenkk thanks so much for looking into it!

I can report that with the latest dev version 1dd1163, my device (_TZ3000_u5u4cakc) now automatically reports the measurements correctly without a custom converter and without manually querying. It is detected as TS011F_plug_3.
[image]

I noticed though, that it refreshes the values quite slow, about 1-2x per minute only. Is this normal?
```
debug 2021-12-18 06:44:16: Received MQTT message on 'zigbee2mqtt/Basement_Washer_Dryer/set' with data 'ON'
debug 2021-12-18 06:44:16: Publishing 'set' 'state' to 'Basement_Washer_Dryer'
info  2021-12-18 06:44:16: MQTT publish: topic 'zigbee2mqtt/Basement_Washer_Dryer', payload '{"current":0,"energy":0.02,"linkquality":84,"power":0,"power_outage_memory":"restore","state":"ON","voltage":236}'
debug 2021-12-18 06:45:06: Received Zigbee message from 'Basement_Washer_Dryer', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":40,"rmsCurrent":175,"rmsVoltage":236}' from endpoint 1 with groupID 0
info  2021-12-18 06:45:06: MQTT publish: topic 'zigbee2mqtt/Basement_Washer_Dryer', payload '{"current":0.18,"energy":0.02,"linkquality":84,"power":40,"power_outage_memory":"restore","state":"ON","voltage":236}'
debug 2021-12-18 06:45:14: Received Zigbee message from 'Basement_Washer_Dryer', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,2]}' from endpoint 1 with groupID 0
info  2021-12-18 06:45:14: MQTT publish: topic 'zigbee2mqtt/Basement_Washer_Dryer', payload '{"current":0.18,"energy":0.02,"linkquality":87,"power":40,"power_outage_memory":"restore","state":"ON","voltage":236}'
debug 2021-12-18 06:45:55: Saving state to file /config/zigbee2mqtt/state.json
debug 2021-12-18 06:46:06: Received Zigbee message from 'Basement_Washer_Dryer', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":40,"rmsCurrent":171,"rmsVoltage":236}' from endpoint 1 with groupID 0
info  2021-12-18 06:46:06: MQTT publish: topic 'zigbee2mqtt/Basement_Washer_Dryer', payload '{"current":0.17,"energy":0.02,"linkquality":72,"power":40,"power_outage_memory":"restore","state":"ON","voltage":236}'
debug 2021-12-18 06:46:14: Received Zigbee message from 'Basement_Washer_Dryer', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,2]}' from endpoint 1 with groupID 0
info  2021-12-18 06:46:14: MQTT publish: topic 'zigbee2mqtt/Basement_Washer_Dryer', payload '{"current":0.17,"energy":0.02,"linkquality":84,"power":40,"power_outage_memory":"restore","state":"ON","voltage":236}'
```

Koenkk, 2021-12-18:
@ShadNex yes, you can control the measurement poll interval https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#options

@capelito then I expect your plug doesn't support power metering at all

ShadNex, 2021-12-18:
Awesome, thanks for your help @Koenkk ! 🥳

simie7, 2021-12-18:
@capelito maybe you have original Tuya gateway to check, if this device has energy measurement capabilities? As I've stuck with the same situation: I have 4 plugs `_TZ3000_nfnmi125` without energy measurement support. And I don't have a way to test, if some features are originally supported.

capelito, 2021-12-18:
> @capelito maybe you have original Tuya gateway to check, if this device has energy measurement capabilities?

No, unfortunately I don't have a Tuya Gateway either. But maybe I'll buy one for testing because I bought the sockets especially for their power monitoring support, which according to the product description, they should have.

Unfortunately, you can't even buy a test model on AliExpress to check if it works because it may be that with the next order the seller sends completely different models.

elzershark, 2021-12-22:
now only integrate one data point for setting the interfall (60sec) and that shows whether the device is on or off. then everything would work.

tatec22, 2021-12-28:
I also received the `_TZ3000_u5u4cakc` plugs.

Manually reading the data via the dev console works, but the measurement_poll_intervall does not seem to work for me.

The values remain the same unless I update them manually...

Currently I'm on 1.22.1-dev commit: 1e75dd5.

Can anyone help me?

orbital253, 2022-01-03:
I bought some _TZ3000_u5u4cakc plugs and when I use the switch on the plug it doesn't show the new state in z2m. Power reporting works ok when I set it to 10 seconds but switch state doesn't get reported. when I manually refresh the state it gets reported. 
@Koenkk Is is possible to query the switch state with a poll interval?

nel-stefan, 2022-01-03:
I have the same issue, the plug is queried correctly and setting the state with zigbee2mqtt works, but using the button on the device doesn't update the state. If I press the refresh button in zigbee2mqtt it does update though. 

Looking at the logs for the power updates it does expose the state, but the value is what is being set before.

tatec22, 2022-01-04:
I have to correct myself: Now I also get the power reporting working with the poll interval.

But like @orbital253 and @nel-stefan I don't get any updates on the plug state when pressed manually.

orbital253, 2022-01-04:
I found a workaround to this problem in home assistant that seems to work

```
alias: TS011F_plug_3 state get
description: 'Poll state every 15 seconds'
trigger:
  - platform: time_pattern
    seconds: /15
condition: []
action:
  - service: mqtt.publish
    data:
      topic: zigbee2mqtt/friendly name/get
      payload: '{"state": ""}'
mode: single

```

Koenkk, 2022-01-04:
@orbital253 

> @Koenkk Is is possible to query the switch state with a poll interval?

Good idea, added this now for the `_TZ3000_u5u4cakc`, state will be polled together with the power measurements.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

elzershark, 2022-01-05:
can you use that? zigbee sniffer wireshark. switched on / off on the device itself
[wireshark.zip](https://github.com/Koenkk/zigbee2mqtt/files/7814855/wireshark.zip)

git-developer, 2022-01-09:
Issue #9057 is probably related: reporting must be enabled actively by sending a request for certain values to the device ([implemented](https://github.com/Koenkk/zigbee2mqtt/issues/9057#issuecomment-1007771295) by now).

Koenkk, 2022-01-09:
#9057 is indeed related. since we now match the behaviour of the TuYa gateway this can be closed (https://github.com/Koenkk/zigbee2mqtt/issues/9057#issuecomment-1008004900)

closed 2022-01-09 by Koenkk