Koenkk/zigbee2mqtt #9495: Issue after switching to CC2652

adi-dev, 2021-11-04:
I just got ZZH (CC2652) to upgrade my existing setup with CC2531. Everything works with old, CC2531 coordinator. I followed correct procedure: stop zigbee2mqtt service, update config file - port to tttyUSB0 and rtscts set to false. The service zigbee2mqtt starts fine, reports connection to coordinator `Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'` and receives updates from devices.
Problem starts, when I try to operate switch (unfortunately, only 3 devices I can operate, all the same) I have timeout error:
```
Publish 'set' 'state' to 'Light Stairs 1' failed: 'Error: Command [masked] genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 56155 - 1 - 4 - 6 - 11 after 10000ms)'
```
After swapping back to CC2531, everything works fine again.
Running test script: https://gist.github.com/omerk/0ee0e447a9e36786b4ff71d8f8126a23 suggest that communication with ZZH is fine, result from this script:
```
Got 7 bytes in response to PING command: b'\xfe\x02a\x01Y\x06='
PASS|OK
```

adi-dev, 2021-11-07:
I don't know if this was the cause of the issue or my trial and error, but the final steps before I got this working was:

1. I read somewhere that people changed the `pan_id` to get around a similar issue, so I got to change it, but I found there was no `pan_id` in `configuration.yaml` file. Furthermore, I added the missing line with different `pan_id` (had to update the backup file as it was throwing an error)
2. It worked with the new `pan_id`, I parried one device and can control it.
3. I changed back the `pan_id` and all other devices started working

I seam to me like the missing pan_id in `advanced` -> `network` config was cause

rodrigoscoelho, 2021-12-08:
Just to inform, all my problems related to this were gone after upgrading to the latest version...

adi-dev, 2021-12-17:
I think, it is safe to consider this one as closed

closed 2021-12-17 by adi-dev