Koenkk/zigbee2mqtt #9497: Develco SPLZB-131 Power plug update rate too frequent

andjohansen, 2021-11-04:
## What happened
After adding 6 Develco adapters to my zigbee network the logs are completely filled up with logs from them.
Every device is reporting status every 5 sec. Ever since installing these adapters my network has had quite a lot of crashes and I suspect that this might be the cause. The network might be overloaded because of this. 
Is there a way to limit the update frequency? 

I checked the devices.js for the develco. 
I see that for the SPLZB-132 there is some changes to limit the update frequency. 
Tried to copy this to the SPLZB-131 config and added this as 
external_converters:
  - develco.js

This did not seem to do any difference. 
## What did you expect to happen
Normal update frequency
## How to reproduce it (minimal and precise)

## Debug info
Version of Zigbee2MqttAssistant (this software): 0.3.157 (0.3.157+Branch.master.Sha.ea388c3bbc2d43c4c9c62f27bb6764911ba4acc9) Release
Version of Zigbee2Mqtt: 1.22.0-dev
Coordinator version: 20210708

[image]

sjorge, 2021-11-04:
You should be able to reconfigure the reporting using the frontend without doing any modifications to the files.

On the reporting tab of the devices, raise the minimal reporting time on instantaneousDemand or currentSummDelivered. If that does not help, enable debug logging so you can see what exactly is being reported and then increase the minimal reporting time for that one.

ccorderod, 2021-11-04:
andjohansen, which coordinator are you using? I have 2 ZB network supported by a CC1352P-2 TI launchpad with over 100 devices on each, and a bunch of meters (20+) reporting every 5 seconds and everything works quite well.

andjohansen, 2021-11-04:
Using same as you. TI launchpad CC1352P-2. I have 77 devices. But as I figured out now it's only 2 of 4 Develco are having this issue. And then the 2 others are some Datek adapters that are doing the same. 
I wasn't aware of the Frontend. Set it up now and adjusted the reporting times. This seems to have calmed down the network now. Found another post somewhere about some stale entries in the db file. I had a lot og groups with 'null' as groupid. So removed all of these and it seems to be much more reponsive now. 
Thanks for your suggestions. :)

closed 2021-12-12 by github-actions[bot]