Koenkk/zigbee2mqtt #9499: Problem with new firmware Xiaomi Aqara E1 door & window contact sensor

Whastie, 2021-11-04:
Hi

I am not sure if it has something to do or not, but since I updated the firmware to work with these sensors, I am having problems with device loss.

Is it normal to see devices without any connection on the map?
There are devices that temporarily could not change their state such as the socket outlets.
Or with a door micro that I have mounted to count the m3 of gas that I consume, it has stopped counting and gas consumption if there is.

[image]

What do you think??
Is it possible that the router firmware will also have to be updated?

Koenkk, 2021-11-04:
This is not a problem when it occurs with Xiaomi end devices, see https://www.zigbee2mqtt.io/guide/faq/#why-are-some-links-missing-from-my-networkmap

closed 2021-11-04 by Koenkk

Whastie, 2021-11-04:
But I had this device before performing the firmware update and it had given me no problem, I noticed that it seems that it wants to detect it as a 14LM model but it is an 11LM

it is the same sensor that wants to detect it as both types.

I am going to try to mount the other coordinator with the previous firmware and I tell you.

[image]

Whastie, 2021-11-04:
Hi

With the old firmware it works fine

With this firmware I have the device linked to the coordinator and it does not give me a problem

I have done the following tests:
Test 1. With CC2652P coordinator and CC2652P ZigBee Stick mit Z-Stack firmware. The MCCGQ11LM has returned to work without problem
All the MCCGQ14LMs that I had already linked and configured have continued to work.
I have tried to add a new MCCGQ14LM and it was not reporting status of course because I am with the old firmware before you sent the modification.

Test 2. with CC2652P cordinator and new firmware (CC2652_CC1352_Xiaomi_fix) the MCCGQ14LM work well and link well. But I have several MCCGQ11LMs that have started to give problems of not receiving the signal from the contact.
I have deleted the device and tried to link it again, but it seems that it tries to link it with two models the 11LM and the 14LM and it loops giving an error.

Have I explained myself well?

[image]

[image]

Whastie, 2021-11-05:
Hi Koenn

can you look at the commented please?

Koenkk, 2021-11-05:
The difference between these fws is very small, and I can't imagine that change influences the connectivity behavior of the MCCGQ11LM

Whastie, 2021-11-06:
Well, I had to put the old firmware.
I'll leave it like that until more people try it and see how it works.
The funny thing is that the 14LM devices that were synchronized with the new firmware when mounting the old one, are not giving problems.