Koenkk/zigbee2mqtt #9502: Zigbee network stops working after a minute.

wimb0, 2021-11-04:
## What happened
Zigbee Network stops working after about one minute with (SRSP - AF - dataRequest after 6000ms) messages.
Devices work fine when Z2M has just started. Can switch lights, see door switches do their thing etc.
But after a while nothing happens anymore.

Has been working fine for a long time, but am having these issues since a few weeks. Not really sure when it started.

## What did you expect to happen
That the zigbee network keeps running.

## How to reproduce it (minimal and precise)
Restart Z2M

## Debug info
Zigbee2MQTT version: 1.22.0 / 1.21.2 / 1.19.1 in docker
Adapter hardware: SlaeshCC2652RB (on USB extension lead)
Adapter firmware version: 20210708

Running Ubuntu 20.04 with Z2M in Docker.

ProtoThis, 2021-11-07:
Not a solution but wanted to add I am having the same problem with the same USB stick

Koenkk, 2021-11-07:
Does the issue disappear when disabling the availability feature?

ProtoThis, 2021-11-07:
> Does the issue disappear when disabling the availability feature?

I Will check (not sure what setting that is but Will probally be in the docs 😉). Tried downgrading to the previous version but issue still exists. Will change the setting and let you know.

ProtoThis, 2021-11-07:
@Koenkk It appears to still be broken (after about an hour it breaks). Disabled the availability settings .

The log shows the following:
`[31mZigbee2MQTT:error[39m 2021-11-07 11:32:21: Publish 'set' 'state' to 'ecodim_light_1' failed: 'Error: Command 0x086bd7fffe5deb1b/1 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'

[32mZigbee2MQTT:info [39m 2021-11-07 11:32:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'ecodim_light_1' failed: 'Error: Command 0x086bd7fffe5deb1b/1 genOnOff.off({}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"ecodim_light_1"},"type":"zigbee_publish_error"}'

[31mZigbee2MQTT:error[39m 2021-11-07 11:32:27: Publish 'set' 'state' to 'ecodim_light_1' failed: 'Error: Command 0x086bd7fffe5deb1b/1 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'

[32mZigbee2MQTT:info [39m 2021-11-07 11:32:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'ecodim_light_1' failed: 'Error: Command 0x086bd7fffe5deb1b/1 genOnOff.off({}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"ecodim_light_1"},"type":"zigbee_publish_error"}'
`

wimb0, 2021-11-08:
I will try this also, will get back when I've tested this.
For some reason, 1 out of 20 restarts it suddenly work fine.

ProtoThis, 2021-11-08:
I downgraded to version 1.20.0 which had been running since last night and seems to be working fine till now. Will try to upgrade to a higher version to see where the problem starts occuring.

wimb0, 2021-11-09:
In my case, Availability was already off, accorinding to the GUI.
It was not in my configuration.yaml. What is the default setting?

Earlier this week I updated my Ubuntu 20.04 install, and the kernel is now `Linux optiplex 5.4.0-90-generic #101-Ubuntu SMP Fri Oct 15 20:00:55 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
`.
Since then it seems to be working fine. Even after reboot Z2M starts fine, which normally it did not do.

melad, 2021-11-11:
I just want to add I have the same Stick and the same errors. Will also try to downgrade to 1.20.0

Sonsi79, 2021-11-11:
I have the same error with CC2530 but after few hours (~18). Before updating to 1.22 never had errors like this. I will also try downgrade to 1.20 now

closed 2021-12-19 by github-actions[bot]