Koenkk/zigbee2mqtt #9506: Zigbee2mqtt (Supervised Add-on) fails to start: UnhandledPromiseRejectionWarning ConBeeII after upgrade to 1.22.0-2

nathan-curtis, 2021-11-04:
## What happened
Upgraded to zigbee2mqtt 1.22.0-2

## What did you expect to happen
Normal operation

## How to reproduce it (minimal and precise)
Restart Zigbee2MQTT after upgrade

## Debug info
Zigbee2MQTT version: 1.22.0-2
Adapter hardware: Conbee II
Adapter firmware version: deCONZ_ConBeeII_0x26720700.bin.GCF     

LOG:
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing... 
[16:52:27] INFO: [32mSocat not enabled, marking service as down[0m
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing... 
[16:52:28] INFO: [32mMQTT available, fetching server detail ...[0m
[16:52:28] INFO: [32mMQTT server settings not configured, trying to auto-discovering ...[0m
[16:52:28] INFO: [32mConfiguring 'mqtt://core-mosquitto:1883' mqtt server[0m
[16:52:29] INFO: [32mMQTT credentials not configured, trying to auto-discovering ...[0m
[16:52:29] INFO: [32mConfiguring'addons' mqtt user[0m
[16:52:29] INFO: [32mPrevious config file found, checking backup[0m
[16:52:29] INFO: [32mCreating backup config in '/config/zigbee2mqtt/.configuration.yaml.bk'[0m
[16:52:29] INFO: [32mAdjusting Zigbee2mqtt core yaml config with add-on quirks ...[0m
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[16:52:30] INFO: [32mZigbee Herdsman debug logging enabled[0m
[16:52:30] INFO: [32mHanding over control to Zigbee2mqtt Core ...[0m
> zigbee2mqtt@1.22.0 start
> node index.js
(node:366) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'options' of undefined
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:92:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:83:58)
    at start (/app/index.js:96:18)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:366) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:366) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
[16:52:36] INFO: [32mZigbee Herdsman debug logging enabled[0m
[16:52:36] INFO: [32mHanding over control to Zigbee2mqtt Core ...[0m
> zigbee2mqtt@1.22.0 start
> node index.js
(node:403) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'options' of undefined
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:92:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:83:58)
    at start (/app/index.js:96:18)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:403) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:403) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
[16:52:42] INFO: [32mZigbee Herdsman debug logging enabled[0m
[16:52:42] INFO: [32mHanding over control to Zigbee2mqtt Core ...[0m
> zigbee2mqtt@1.22.0 start
> node index.js
(node:440) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'options' of undefined
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:92:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:83:58)
    at start (/app/index.js:96:18)
(Use `node --trace-warnings ...` to show where the warning was created)
[... 485 lines cut ...]
> node index.js
(node:1957) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'options' of undefined
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:92:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:83:58)
    at start (/app/index.js:96:18)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:1957) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:1957) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
[16:56:50] INFO: [32mZigbee Herdsman debug logging enabled[0m
[16:56:50] INFO: [32mHanding over control to Zigbee2mqtt Core ...[0m
> zigbee2mqtt@1.22.0 start
> node index.js
(node:1994) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'options' of undefined
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:92:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:83:58)
    at start (/app/index.js:96:18)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:1994) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:1994) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
[16:56:55] INFO: [32mZigbee Herdsman debug logging enabled[0m
[16:56:56] INFO: [32mHanding over control to Zigbee2mqtt Core ...[0m
> zigbee2mqtt@1.22.0 start
> node index.js

nathan-curtis, 2021-11-05:
Tried a number of things to get it to pass the issue - including reloading the stick's current firmware.  Nothing worked - same errors as above.  Rolled back (restored Snapshot/Backup) to 1.21.2-1.  Problem no longer exists.

nathan-curtis, 2021-11-09:
@Koenkk checking if you need additional info

Koenkk, 2021-11-10:
@nathan-curtis do you use any external converters, if yes can you try removing them. I see a lot of external converter related errors in your log

nathan-curtis, 2021-11-10:
I use three:
external_converters:
  - NYCE3010.js
  - PP-WHT-US.js
  - SV01-612-MP-1-1.js
  
 The NYCE3010 basically is a duplicate of what you have for the NYCE3011 (was going to ask to add support) The SV01-612-MP is a variant of the 610/611 Keen vent (again a dupe of existing to add support for a variant)    The PP-WHT-US is the Peanut Plug.  This device refuses to show up correctly due to other notes you've made about the device type the device DOES NOT return.  I can remove that one but I cannot pull the devices out of the mesh - they're my main repeaters.

Here's the issue.  Unfortunately these three items make the BULK of my zigbee network (I have a few sengled bulbs and a couple GE outlets.. but  if I pull these I basically dont have anything left.  It takes out all my door sensors and vents, basically the entire reason I have Zigbee at all.  I'm using Zigbee2MQTT SPECIFICALLY instead of ZHA because I could get these working where as ZHA couldn't.
  
Also due to the severe impact to my prod system, I simply had to rollback  - so I cannot reproduce the issue at the moment without upgrading and therefore taking the system down. - not acceptable due to WAF.

closed 2021-12-19 by github-actions[bot]