Koenkk/zigbee2mqtt #9509: Silvercrest/Lidl TS0601 - _TZE200_chyvmhay did not work but works with https://github.com/Lavisx/lidl-valve-z2m/blob/main/lidl_radiator_valve.js

gmarchart, 2021-11-05:
Hello,
not sure if i should report here or there - i opened also -> #3266  [](https://github.com/Koenkk/zigbee-herdsman-converters/issues/3266)
--------------------------------------

Hello,
using conbee2 on Raspberrypi, zigbee2mqtt latest version - Current version: 1.22.0-2. Ha, latest version.Home Assistant 2021.10.7

System Health
version	core-2021.10.7
installation_type	Home Assistant OS
dev	false
hassio	true
docker	true
user	root
virtualenv	false
python_version	3.9.7
os_name	Linux
os_version	5.10.17-v7
arch	armv7l
timezone	Europe/Vienna
Host-Betriebssystem	Home Assistant OS 6.6
Update-Channel	stable
Supervisor-Version	supervisor-2021.10.8
Docker-Version	20.10.8
Speicherplatz gesamt	58.0 GB
Speicherplatz genutzt	12.7 GB
Gesund	true
Unterstützt	true
Board	rpi3
Supervisor-API	ok
Versions-API	ok
Installierte Add-ons	Mosquitto broker (6.0.1), Samba share (9.5.1), Check Home Assistant configuration (3.9.0), TasmoAdmin (0.15.0), File editor (5.3.3), AirCast (3.1.1), SSH & Web Terminal (9.0.1), Portainer (2.0.0), deCONZ (6.10.0), Node-RED (10.0.1), Duck DNS (1.14.0), Let's Encrypt (4.11.0), Remote Backup (4.2.7), InfluxDB (4.2.1), Grafana (7.2.0), Zigbee2mqtt (1.22.0-2), Log Viewer (0.12.0)
Zigbee Modell
TS0601
Zigbee Hersteller
_TZE200_chyvmhay
Beschreibung
Radiator valve with thermostat
Unterstützungsstatus
unterstützt

IEEE Addresse
0x0c4314fffe56bf22
Netzwerk Adresse
0x01BD
Hersteller
TuYa
Modell
TS0601_thermostat

status is supported, but batterie is not recognized, device say full, mqtt say low and blinks...
other settings to change is not possible in mqtt. no reaction. either schedule, or temperatur or something else, nothing works. i have 4 Valves, all does the same, supported, but no reaction on any changes when done in mqtt.

Bevor i used Lavisx converter - that works mostly - but had some issues, but mainly it works, see below -
There is my custom converter until it will be officially implemented.
https://github.com/Lavisx/lidl-valve-z2m/blob/main/lidl_radiator_valve.js

Below is the first request i opened about this Valve, Lavisx wrote then a temp. converter - Request was closed in the meantime.
TS0601 - TZE200_chyvmhay' Thermostat not supported - please add if possible #8934
Closed
gmarchart opened this issue on 1 Oct · 29 comments

But with this official converter there is no functionality.
i allready removed the thermostat device from mqtt and re-pair it, also a reset on the valve done, but the situation was the same. Parameter to change in mqtt did not reflect on Thermostat. It seems that the datapoints from the tuya device is not the same as used in Lavisx Converter, because his one works mainly, the official one not.

Do you need additional debug or testing output from the device ?
mqtt valve status:

{
"auto_lock": "AUTO",
[... 151 lines cut ...]
Info MQTT publish: topic 'zigbee2mqtt/Lidl-Thermostat-WZ', payload '{"auto_lock":"AUTO","away_mode":null,"away_preset_days":0,"away_preset_temperature":{"0":6,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"battery_low":{"0":2,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"boost_time":34,"child_lock":null,"comfort_temperature":false,"current_heating_setpoint":0,"eco_temperature":null,"holidays":[{"hour":5,"minute":34,"temperature":24},{"hour":42,"minute":36,"temperature":34},{"hour":4,"minute":42,"temperature":92},{"hour":34,"minute":96,"temperature":42},{"hour":32,"minute":34,"temperature":96},{"hour":42,"minute":96,"temperature":34}],"linkquality":255,"max_temperature":{"0":19,"1":1,"2":1,"3":0,"4":0,"5":34,"6":0,"7":0},"min_temperature":34,"position":{"0":1,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"update_available":null,"valve_detection":null,"window_detection":"OFF","window_detection_params":{},"workdays":[{"hour":4,"minute":34,"temperature":24},{"hour":42,"minute":36,"temperature":34},{"hour":4,"minute":42,"temperature":92},{"hour":34,"minute":96,"temperature":42},{"hour":32,"minute":34,"temperature":96},{"hour":42,"minute":96,"temperature":34}]}'
Info MQTT publish: topic 'zigbee2mqtt/Lidl-Thermostat-WZ', payload '{"auto_lock":"AUTO","away_mode":null,"away_preset_days":0,"away_preset_temperature":{"0":6,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"battery_low":{"0":2,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"boost_time":34,"child_lock":null,"comfort_temperature":false,"current_heating_setpoint":0,"eco_temperature":null,"holidays":[{"hour":5,"minute":34,"temperature":24},{"hour":42,"minute":36,"temperature":34},{"hour":4,"minute":42,"temperature":92},{"hour":34,"minute":96,"temperature":42},{"hour":32,"minute":34,"temperature":96},{"hour":42,"minute":96,"temperature":34}],"linkquality":255,"max_temperature":{"0":19,"1":1,"2":1,"3":0,"4":0,"5":34,"6":0,"7":0},"min_temperature":34,"position":{"0":1,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"update_available":null,"valve_detection":null,"window_detection":"OFF","window_detection_params":{},"workdays":[{"hour":4,"minute":34,"temperature":24},{"hour":42,"minute":36,"temperature":34},{"hour":4,"minute":42,"temperature":92},{"hour":34,"minute":96,"temperature":42},{"hour":32,"minute":34,"temperature":96},{"hour":42,"minute":96,"temperature":34}]}'
Info MQTT publish: topic 'zigbee2mqtt/Lidl-Thermostat-WZ', payload '{"auto_lock":"AUTO","away_mode":null,"away_preset_days":0,"away_preset_temperature":{"0":6,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"battery_low":{"0":2,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"boost_time":34,"child_lock":null,"comfort_temperature":false,"current_heating_setpoint":0,"eco_temperature":null,"holidays":[{"hour":5,"minute":34,"temperature":24},{"hour":42,"minute":36,"temperature":34},{"hour":4,"minute":42,"temperature":92},{"hour":34,"minute":96,"temperature":42},{"hour":32,"minute":34,"temperature":96},{"hour":42,"minute":96,"temperature":34}],"linkquality":255,"max_temperature":{"0":19,"1":1,"2":1,"3":0,"4":0,"5":34,"6":0,"7":0},"min_temperature":34,"position":{"0":1,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"update_available":null,"valve_detection":null,"window_detection":"OFF","window_detection_params":{},"workdays":[{"hour":4,"minute":34,"temperature":24},{"hour":42,"minute":36,"temperature":34},{"hour":4,"minute":42,"temperature":92},{"hour":34,"minute":96,"temperature":42},{"hour":32,"minute":34,"temperature":96},{"hour":42,"minute":96,"temperature":34}]}'
Info MQTT publish: topic 'zigbee2mqtt/Lidl-Thermostat-WZ', payload '{"auto_lock":"AUTO","away_mode":null,"away_preset_days":0,"away_preset_temperature":{"0":6,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"battery_low":{"0":2,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"boost_time":34,"child_lock":null,"comfort_temperature":false,"current_heating_setpoint":0,"eco_temperature":null,"holidays":[{"hour":5,"minute":34,"temperature":24},{"hour":42,"minute":36,"temperature":34},{"hour":4,"minute":42,"temperature":92},{"hour":34,"minute":96,"temperature":42},{"hour":32,"minute":34,"temperature":96},{"hour":42,"minute":96,"temperature":34}],"linkquality":255,"max_temperature":{"0":19,"1":1,"2":1,"3":0,"4":0,"5":34,"6":0,"7":0},"min_temperature":34,"position":{"0":1,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"update_available":null,"valve_detection":null,"window_detection":"OFF","window_detection_params":{},"workdays":[{"hour":4,"minute":34,"temperature":24},{"hour":42,"minute":36,"temperature":34},{"hour":4,"minute":42,"temperature":92},{"hour":34,"minute":96,"temperature":42},{"hour":32,"minute":34,"temperature":96},{"hour":42,"minute":96,"temperature":34}]}' [... cut ...]
Debug Publishing 'set' 'local_temperature_calibration' to 'Lidl-Thermostat-Schlafzimmer'
Debug Publishing 'set' 'local_temperature_calibration' to 'Lidl-Thermostat-WZ'
Debug Received MQTT message on 'zigbee2mqtt/Lidl-Thermostat-Kueche/set/local_temperature_calibration' with data '2.5'
Debug Publishing 'set' 'local_temperature_calibration' to 'Lidl-Thermostat-Kueche'
Debug Publishing 'set' 'local_temperature_calibration' to 'Lidl-Thermostat-VZ'
Debug Publishing 'set' 'local_temperature_calibration' to 'Lidl-Thermostat-Schlafzimmer'

Thanks for a check,

Br
Guenther

miragecze, 2021-11-13:
Hello, I have same issue...

Koenkk, 2021-11-19:
@Lavisx could you create a pull request with https://github.com/Lavisx/lidl-valve-z2m/blob/main/lidl_radiator_valve.js ?

Lavisx, 2021-11-20:
@Koenkk I haven't anty experience with pull request and I don't know how I may  integrate the converter to your project. Is possible that you will use my code and integrate it to your project?
Or is someone who can do it?

BerlinSnoop, 2021-11-20:
Another newbie (github+MQTT) here with the same problem...

Koenkk, 2021-11-22:
Added the definition from @Lavisx , assuming this can be closed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

ciB89, 2021-11-23:
As @Lavisx pointed out earlier, his converter is not yet finished, so this converter works a little better but still has some problems.

ciB89, 2021-11-23:
When trying to set the temp from Home Assistant I get the following error and z2m even crashes:

`Payload '{"auto_lock":"AUTO","away_mode":"OFF","away_preset_days":0,"away_preset_temperature":{"0":6,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"battery_low":{"0":2,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"binary_one":null,"binary_two":null,"boost_time":34,"child_lock":"UNLOCK","comfort_temperature":false,"current_heating_setpoint":"12.0","current_heating_setpoint_auto":"21.0","detectwindow_temperature":null,"detectwindow_timeminute":null,"eco_temperature":null,"holidays":[{"hour":5,"minute":34,"temperature":24},{"hour":42,"minute":36,"temperature":34},{"current":true,"hour":4,"minute":42,"temperature":92},{"hour":34,"minute":96,"temperature":42},{"current":true,"hour":32,"minute":34,"temperature":96},{"hour":42,"minute":96,"temperature":34}],"holidays_schedule":"05:34/24°C 42:36/34°C 04:42/92°C 34:96/42°C 32:34/96°C 42:96/34°C","linkquality":null,"local_temperature":"12.0","local_temperature_calibration":null,"max_temperature":{"0":19,"1":1,"2":1,"3":0,"4":0,"5":34,"6":0,"7":0},"min_temperature":34,"position":{"0":1,"1":34,"10":96,"11":42,"12":96,"13":34,"14":96,"15":42,"16":96,"17":34,"2":24,"3":42,"4":36,"5":34,"6":68,"7":42,"8":92,"9":34},"preset":"manual","system_mode":"heat","voltage":1510,"window_detection":"OFF","window_detection_params":{},"workdays":[{"hour":4,"minute":34,"temperature":24},{"hour":42,"minute":36,"temperature":34},{"current":true,"hour":4,"minute":42,"temperature":92},{"hour":34,"minute":96,"temperature":42},{"current":true,"hour":32,"minute":34,"temperature":96},{"hour":42,"minute":96,"temperature":34}],"workdays_schedule":"04:34/24°C 42:36/34°C 04:42/92°C 34:96/42°C 32:34/96°C 42:96/34°C"}' is not a Number`

Koenkk, 2021-11-23:
@ciB89 the `No network route' (205)` error is not caused by the converter but by the device not being reachabel.

ciB89, 2021-11-23:
Well, I sucessfully paired it twice now, and it's sitting close to my coordinator. By the way, I just realized that z2m crashes whenever I publish a change on the device...

ciB89, 2021-11-23:
It also crashes when I change the temperature directly on the physical device o0

ciB89, 2021-11-23:
This is the last log entry before crashing:

```
info  2021-11-23 21:46:29: Configuring '0x0c4314fffe618e95'
debug 2021-11-23 21:46:29: Received Zigbee message from '0x0c4314fffe618e95', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,1,84],"type":"Buffer"},"datatype":2,"dp":24,"fn":0,"status":0,"transid":24}' from endpoint 1 with groupID 0
info  2021-11-23 21:46:29: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe618e95', payload '{"away_mode":null,"binary_one":null,"binary_two":null,"child_lock":null,"comfort_temperature":null,"current_heating_setpoint_auto":null,"detectwindow_temperature":null,"detectwindow_timeminute":null,"eco_temperature":null,"linkquality":111,"local_temperature":"34.0","local_temperature_calibration":null,"voltage":null}'
```

ciB89, 2021-12-01:
@Koenkk why was this pushed to stable today? When using this version it crashes z2m when trying to use the valve. Using the original code as an external converter still works though.

gmarchart, 2021-12-02:
> 

> It also crashes when I change the temperature directly on the physical device

yes, have the same issue, regardless what to change in the widget, z2m crashes, also the local temperature in z2m seems not be updated, temperatur local calibration also did not work.

gmarchart, 2021-12-02:
it seems not to be a problem with Current version: 1.22.1-1 of z2m, because with the external converter from Lavisx i have not these issues as with the official converter. mode change, temperature calibration, local temperature works with the external converter, not all is perfect but it works (very slow update of local valve temperatur, that sucks a little bit, but mainly ok)

gmarchart, 2021-12-18:
hello, 
z2m seems to change the time of the valve or reset it to a default value of the valve - after setting the time on the valve it is ok, then one day after the time and date is totally wrong - year 2067 false month, false hour and minute - so the automatic schedule of the valve did not work correct, because the  day is false and the time ....

if i turn on debug, i see nothing on time change or any payload regarding time, but it seems did not work after i use the official converter, not the one from lavisx as external converter.

i will check that issue  with the external converter with lavisx if that is there also a problem or not - maybe a tuya.js problelm ?! somthing wrong or not fit for this valve ?!

i will check and give feedback soon.

gmarchart, 2021-12-22:
now i use the z2m official version with the external converter from lavisx ->  https://github.com/Lavisx/lidl-valve-z2m/blob/main/lidl_radiator_valve.js  - and that setup works mainly.

so i will use that setup until a new offical z2m version comes out - hopefully not with the false z2m edge setttings ...

gmarchart, 2022-01-05:
today i use now the edge version 1.22.2-dev commit: 4f887a1 without an external converter - and all is ok - there was an update for this time issue - now it works. i wait until this update is in the official branch, after that i will use it- for the moment it is ok to use edge  - all working as expected. so i close this issue.

closed 2022-01-05 by gmarchart