Koenkk/zigbee2mqtt #9517: How to pair KlikAanKlikUit ZCC-3500?

barthuijgen, 2021-11-05:
## What happened
I've purcused a KlikAanKlikUit ZCC-3500 because it was in the supported devices list for zigbee2mqtt but I can't find how to pair it. I even went and bought a KlikAanKlikUit remote (AYCT-102) to see if that would help.

The pairing process with the remote is as follows:
- Plug in the wall socket and it will enter pairing mode for 4 seconds (zigbee2mqtt picks up nothing during this, it seems to be not the regular zigbee interview pairing)
- During these 4 seconds press one of the ON buttons on the remote, this will trigger the pair process

## What did you expect to happen
zigbee2mqtt to pair automatically when the device is in "pair" mode.

So the question is... how do I pair this device? the device page on docs sometimes gives pairing instructions but they are missing for this device: https://www.zigbee2mqtt.io/devices/ZCC-3500.html

## How to reproduce it (minimal and precise)
Plug in the device and try to pair

## Debug info
Zigbee2MQTT version: Latest docker image koenkk/zigbee2mqtt
Adapter hardware: Conbee II
Adapter firmware version: I don't know

francisp2, 2021-11-05:
[image]

barthuijgen, 2021-11-05:
Thanks, but I think this option requires the bridge ([ICS-2000](https://klikaanklikuit.nl/product/internet-control-station/)) and the "smart" remote ([ZYCT-202](https://klikaanklikuit.nl/product/zigbee-afstandsbediening-met-muurbevestiging/)) which I both do not have. I only have the [AYCT-102](https://klikaanklikuit.nl/product/afstandsbediening-met-16-kanalen/) remote. Since the switch itself communicates via zigbee, just not official zigbee certified apparently. and zigbee2mqtt list it as supported device (and not the other devices like ICS-2000), I assumed it would be able to pair and work seperately.

francisp2, 2021-11-06:
Pairing is pairing. If that is the way to pair to their own bridge, it should be the way to pair to the Zigbee2MQTT coordinator.

barthuijgen, 2021-11-06:
Hmm... I'll try again. But when plugging it in ZigBee2mqtt shows no logs, nothing at all. It is in auto join mode and I've paired other devices that it all picked up automatically so I'm fairly sure the setup should be fine.

barthuijgen, 2021-11-06:
I've gone ahead and bought the [Ikea switch](https://www.zigbee2mqtt.io/devices/E1603_E1702_E1708.html) and this worked immediately. I'll be sending the ACC-3500 back in a couple days.

The issue can be closed for me, regardless I do think the ACC-3500 should be either removed from the compatible devices list or working pairing instructions added to the docs. The current situation may lead other users to run into the same problems as me.

closed 2021-12-14 by github-actions[bot]