Koenkk/zigbee2mqtt #9518: Error while starting zigbee-herdsman

belet-pl, 2021-11-05:
## What happened
**Error while starting zigbee-herdsman
Failed to start zigbee**

It happens on both ConBee II and CC2531
There is nothing specific in the log. Just entries I shared above. :(

## What did you expect to happen
start Zigbee2MQTT

## How to reproduce it (minimal and precise)
Don't know

## Debug info
Zigbee2MQTT version: 1.22.0-2
Adapter hardware: CC2531, Conbee II
Adapter firmware version:

sanyafifa, 2021-11-05:
a similar problem appeared two weeks ago, the SNZB-02 sensor stopped responding

guys, I need help :)

[image]

user514, 2021-11-07:
same here cc2531, coordinator 1.2. It happens in two different computers (different locations and networks). Both after updating HA . One with lasts versions og HA2021.10, the other when updating to 2021.11.1 yesterday.

  Error starting herdsman and something about json end of file.

evilmumi, 2021-11-07:
same here exactly after updating from 1.17.0 to 1.22.0 on my raspberry.
how can I go back to 1.21.1 if above guys writing it works

user514, 2021-11-07:
> how can I go back to 1.21.1 if above guys writing it works

  Not sure in your case. I can tell you that when you update to 1.21 home assistant asks you to save a backup of your current version of zigbee2mqtt (1.17) and config files. If you answered yes when asked, then you could go to:

  SUPERVISOR -> Backups (top menu) -> And search for that backup which will be called with the name of zigbee2mqtt 1.17. Open it, check marks on addon and config files and hit RESTORE. You will go back to your previous version. Comment us if that fixed your problem (temporaly of course). I'm the only one by name confirming it with the downgrade.

I think it is not easy to downgrade to certain version if you dont have the backup. The only thing that I found was about trying zigbee2mqtt edge, here: https://community.home-assistant.io/t/how-to-downgrade-zigbee2mqtt/231273/7 but I guess that would not fix the issue because edge will be even newer than 1.22.

user514, 2021-11-08:
@sanyafifa is using cc2531 dongle with coordinator firmware: CC2531_DEFAULT_20190608
@user514 is using cc2531 dongle with coordinator firmware: CC2531_DEFAULT_20190608
@belet-pl is using cc2531 dongle ????? unknown firmware
@evilmumi is using ????? dongle with ?????? firmware

...if we all are using same dongle, with same old firmware and no other people are complaining about latest updates of zigbee2mqtt, maybe problem could be solved updating our coordinator dongle to version CC2531_DEFAULT_20201127, which I think could be the last one available.

@belet-pl and @evilmumi could you please check which version of coordinator firmware you installed in your usb dongle on your zigbee network?.

mikepruett3, 2021-11-09:
Having similar/but different issues with my CC2531 adapter. A few days after the upgrade of HASS to core-2021.10.6 and Zigbee2mqtt v1.22.0-2 my adapter would be "Removed" after the addon started. Fixed by switching firmware from Z-Stack_Home_1.2 (default) to Z-Stack_Home_1.2 (source_routing).

evilmumi, 2021-11-09:
@user514 cc2531 dongle with unknown firmware - can you tell how i can read out version and how to update - i bought it fully flashed on private base

user514, 2021-11-09:
> Fixed by switching firmware from Z-Stack_Home_1.2 (default) to Z-Stack_Home_1.2 (source_routing).

  Thank you Mike. Good information. That goes in the direction of fixing it by updating coordinator firmware. This is just a guess, but I think it could be solved by flashing again 1.2 DEFAULT but with the last version 2020-11-27 I think. I think that because if not, I think much more people would be complaining with zigbee2mqtt last updates.

  I will let you know here if I have time to update coordinator dongle with 2020 version and test.

sanyafifa, 2021-11-09:
@user514 did you have Sonoff devices on your network?
I started with periodically disconnecting the sensor, to fix it I switched the CC2531 stick

> This is just a guess, but I think it could be solved by flashing again 1.2 DEFAULT but with the last version 2020-11-27.

We are waiting for information from the dear @Koenkk

user514, 2021-11-09:
> @user514 cc2531 dongle with unknown firmware - can you tell how i can read out version and how to update - i bought it fully flashed on private base

  In my case I bought the dongle clean and have to upload the firmware that was available at that time. CC2531_DEFAULT_20190608. I don't know if there is any way to know which firm is on the dongle easily. I know because that was the one I upload to it.

  To update, you need a cc degugger. Try to search at aliexpress 'cc debugger' and you will see the kit everyone has. You install the windows drivers for the debugger, install the texas instruments flash programmer, connect the USB dongle to the debugger with the included cable, download the proper firmware (for our dongles probably this one:
https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default  ) and upload to the dongle. Should me much better explained with pictures at zigbee2mqtt page. 'Flashing the coordinator' or something like that.

user514, 2021-11-09:
> @user514 did you have Sonoff devices on your network?

 ...if you mean, sonoff in general (wifi switch basic model), yes. But I think you mean sonoff zigbee devices: Not on any of the two networks I have. Some xiaomi and a cc2530 router (as repeater). Keep thinking is a problem with old coordinator firmware.

evilmumi, 2021-11-09:
@user514 i have this line on 1.20.0 : `info  2021-11-08 13:16:53: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":2,"minorrel":0,"product":0,"revision":"","transportrev":2},"type":"zStack12"}'
`
I tested 1.21.0 and it throws the same error as 1.22.0. I will ask guy where i bought stick if he know which firmware i have on it

user514, 2021-11-09:
> revision":"

...ooops. For some reason, your dongle is not giving that info. As you can see here, that is the proper way to check it:
https://www.zigbee2mqtt.io/guide/adapters/flashing/flashing_the_cc2531.html#how-to-check-the-installed-firmware-version

  Don't know why yours is not giving it.

> I tested 1.21.0 and it throws the same error as 1.22.0.

  Please, check the complete version numbers you are trying, because I have two versions of 1.21... once starts ok, and the other throws the error as:

1.21.1-1 zigbee2mqtt starts without errors.
1.21.2-1 won't start. Throws error to me.
(1.22.xx of course, the error continues.)

This is the error I get:
https://user-images.githubusercontent.com/19530873/140658240-e583a586-55bc-4352-b8a4-7853c3b957bc.jpg

  Please check if the version 1.21 that throws error to you is **1.21.2-1** (or I will be mistaken about the version that introduces the failure).

evilmumi, 2021-11-09:
Hi,

i just got information that my stick has SBL so I can update to new firmware from here:
https://github.com/Koenkk/zigbee2mqtt/issues/320

I will test it next 24h.

i just checkedout 1.20 it is 1.2.0  not 1.21....

sanyafifa, 2021-11-09:
> Я только что получил информацию, что у моей флешки есть SBL, поэтому я могу обновить прошивку отсюда:
> [# 320](https://github.com/Koenkk/zigbee2mqtt/issues/320)

Hey! @evilmumi 
I could not flash it in this way! When I received my CC2531, I flashed it through the raspberry.
If you succeed, let us know

user514, 2021-11-09:
> I will test it next 24h.
let us know how it goes... thanks.

evilmumi, 2021-11-10:
@user514 doing some stuff like this https://www.marrold.co.uk/2019/12/flashing-cc2530-cc2591-zigbee-module.html i think i crashed my 2531.
i have an sonoff 3.0 stick and will test it with 1.22.0 perhaps new pairing everything

user514, 2021-11-10:
> doing some stuff like this https://www.marrold.co.uk/2019/12/flashing-cc2530-cc2591-zigbee-module.html i think i crashed my 2531.

  You are linking to a page where they flash a cc2530-cc2591... and then you say you have a 2531. It's important to flash to a compatible version with your hardware. In this same page, same guy links to a page where he flash a 2531, here:

https://www.marrold.co.uk/2019/09/flashing-cc2531-zigbee-dongle-with.html
This one seems more correct than the one you are linking having in mind you have a 2531 dongle.

> i have an sonoff 3.0 stick and will test it with 1.22.0 perhaps new pairing everything

Be careful too with writing firmware that is not made for 3.0 like your sonoff. USB cc2531 uses version 1.2, at least mine. There are two different versions of coordinator firmware, one for 1.2 and other for 3.0. In the sonoff 3.0 I suppose should go the 3.0 version, don't know exactly which one 3.0 or 3.x, check that. I link you to the home directory where you select which firmware you like to download, 1.2 or 3.0 (some folders prior of the link I post earlier):

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator

  Check which is the right firmware for your sonoff 3.0 before proceeding.

evilmumi, 2021-11-10:
@user514: For me everything is now working. leave my 2530/2531 crashed away so long as i will get answer from seller .

Now I flashed with fw from https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin
which link me correct version for my sonoff stick and updating was very easy.

user514, 2021-11-10:
> I flashed with fw Z-Stack_3.x.0/bin

...ok, so you flashed with 3.x.0, version ---> coordinator_20210708.zip   right?. Good to know. Porbably I will be updating to coordinator 20201127 which is the last version for out 2531 and hope everything get solved to then.

evilmumi, 2021-11-10:
@user514 i flashed CC1352P2_CC2652P_launchpad_coordinator_20210708.zip

sanyafifa, 2021-11-14:
@user514 
flashed CC2531 and started Z2M! 
but SNZB-02 still does not transmit data :(

[image]

user514, 2021-11-14:
> but SNZB-02 still does not transmit data :(

...maybe because of new coordinator writen on dongle, SNZB-02 needs a new re-pair?.

user514, 2021-11-25:
SOLVED:
  Would like to update, that today, I upgraded my usb dongle cc2531 from CC2531_DEFAULT_20190608.zip and now I installed (.hex file) from CC2531_DEFAULT_20201127.zip on my coordinator dongle.

  With that. zigbee2mqtt starts without the zigbee-herdsman error. Everything went smooth, but I went the careful way and did it as clean as possible:

-remove old devices (force remove on some)
-supervisor -> reboot to stay on clean HA.

----
-One thing I learned here. Previously I have problems with devices connecting with coordinator directly, but they have a router close to them. To avoid that, This time I enable 'Permit Join' only over the device I want to function as entering gate. Por example:

   Permit pair (router-garage) -> click pairing door sensor of garage

This way, I got the sensor get connected and create route through the correct router. Not directly to coordinator.
---

 I re-pair every device. I renamed every device, forcing update HA entities (thanks for that k.)

  Everything goes normal. And the paths in zigbee map seems better thant before. Now testing everything is ok.

closed 2022-01-02 by github-actions[bot]