Koenkk/zigbee2mqtt #9520: User extensions are loaded multiple times, stop() is not called

okastl, 2021-11-05:
## What happened
When creating a user extension, changing and saving the extension in the web interface will cause the extension to be loaded multiple times. This is different to the behavior before the typescript refactor. The stop() function of the extension is not called.

## What did you expect to happen
The extension should be stopped and the changed version should be loaded.

## How to reproduce it (minimal and precise)

Install the attached test.js extension.
On every state change you will see "State changed Number 1" in the log.
Edit the extension with the web interface and save it:
`   async onStateChange(data) {
        console.log("State changed Number 2");
    }
`
On every state change you will see "State changed Number 1" and "State changed Number 2"in the log.

Edit the extension with the web interface and save it:
`   async onStateChange(data) {
        console.log("State changed Number 3");
    }
`
On every state change you will see "State changed Number 1", and  "State changed Number 3"in the log.
Note, that the stop() function is not called.

Related problem, if you exit z2m, stop() is called, but an exception is thrown:
´Failed to call 'TestExtension' 'stop' (TypeError: this.eventBus.removeListenersExtension is not a function
    at TestExtension.stop (D:\NB\z2m\zigbee2mqtt\dist\util\externally-loaded.js:15:23)´

[test.zip](https://github.com/Koenkk/zigbee2mqtt/files/7486547/test.zip)

See also: https://github.com/Koenkk/zigbee2mqtt/discussions/7642

## Debug info
Zigbee2MQTT version: 1.22.0-dev commit: a09aa424
Adapter hardware: CC26X2R1 (slaesh)
Adapter firmware version: 20210708

Koenkk, 2021-11-05:
- I've fixed the starting bug
- Regarding stop failing, the api has changed slightly, you should change `this.eventBus.removeListenersExtension(this.constructor.name);` to `this.eventBus.removeListeners(this);`
- 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

okastl, 2021-11-05:
> * I've fixed the starting bug

Just tried 1.22.0-dev commit: 2399ec04, which I believe should contain your fix, but the behavior is the same.

> * Regarding stop failing, the api has changed slightly, you should change `this.eventBus.removeListenersExtension(this.constructor.name);` to `this.eventBus.removeListeners(this);`
> 

Thanks, this fixes the exception on exit.

Koenkk, 2021-11-07:
Ah, I see the `eventbus.on` call is also wrong, so this should work:

```js
class TestExtension {
    constructor(zigbee, mqtt, state, publishEntityState, eventBus, settings, logger) {
        logger.info('Loaded Test');
        this.eventBus = eventBus;
        this.eventBus.on('stateChange', this.onStateChange.bind(this), this);
    }

    async onStateChange(data) {
        console.log("State changed Number 1");

    }

    async stop() {
        console.log("async stop called"); 
        this.eventBus.removeListeners(this);
    }
}

module.exports = TestExtension;

```

okastl, 2021-11-07:
> Ah, I see the `eventbus.on` call is also wrong, so this should work:
> ...

Indeed, it does! Thank you!

closed 2021-11-07 by okastl