Koenkk/zigbee2mqtt #9521: Sonoff's pairing issues

dsduarte, 2021-11-05:
Hello guys!
In the past few weeks I've been struggling to pair some sonoff sensors and swithces and I've reached a point where I can't get any further...

My main setup is:
- QNAP NAS running HA OS on a Virtual Machine;
- Zigbee2MQTT running as HA Add-on;
- Sonoff devices SNZB-01, SNZB-02, SNZB-03, SNZB-04, ZB-MINI and ZB-BASIC;
- CC2531 adapter.

With this setup, devices are not pairing... I enable join and start pairing device and nothing happens. The zigbee-herdsman log shows nothing... I thougth could be firmware issue on CC2531 and ordered CC-downloader an cables to update firmware and still the same.

I also have an alternative seteup with a borrowed Sonoff Bridge and replacing CC2531 with it, devices are working well...

That said, I starded an alternative setup to test the CC2531:
- Windows PC;
- Zigbee2MQTT running on the Windows;
- Sonoff devices SNZB-01, SNZB-02, SNZB-03, SNZB-04, ZB-MINI and ZB-BASIC;
- CC2531 adapter on USB COM port.
- MQTT broker on another HA VM.

On this setup, devices pair ok... very quick, just how it suposed to be.

I don't know what else to do to work with CC2531... I thought that it could be defective and ordered a SONOFF Zigbee 3.0 USB Dongle Plus, but its not arrived yet and while I'm waiting it to arrive I could be sure that the CC2531 is working.

But I have noticed some differences between data reported by the same device on each setup... here some logs...

Sonoff Bridge over TCP
info  2021-11-04 19:23:33: MQTT publish: topic 'zigbee2mqtt/TH01_zigbee', payload '{**"battery":9.5**,"humidity":71.04,"linkquality":136,"temperature":26.26,"voltage":2700}'
info  2021-11-04 19:23:49: MQTT publish: topic 'zigbee2mqtt/TH01_Lulu_zigbee', payload '{"**battery":68**,"humidity":72.89,"linkquality":136,"temperature":26.27,"voltage":2900}'

CC2531 on Windows PC
info  2021-11-04 17:36:33: MQTT publish: topic 'zigbee2mqtt_NEW/0x00124b00226cbddc', payload '{"**battery":21.5**,"humidity":66.29,"linkquality":134,"temperature":27.4,"voltage":2800}'
info  2021-11-04 17:36:43: MQTT publish: topic 'zigbee2mqtt_NEW/0x00124b002268c751', payload '{"**battery":92.5**,"humidity":65.27,"linkquality":139,"temperature":29.34,"voltage":2900}'

Is it possible to CC2531 parse the battery data info different than Sonoff Bridge? If so, is it possible that this could be the cause of not pairing?

What else can I do?

francisp2, 2021-11-07:
Don't quite get it, where do you use the Sonoff Bridge ? On the Qnap or on Windows ?

tomdudu38, 2021-11-08:
i'm on debian, nuc
snzb02, 04 and basic was working  here with cc2531.
i say 'was' because i switched adaptater for a 2652xx there is few weeks.

sjorge, 2021-11-10:
You're not alone: https://github.com/Koenkk/zigbee2mqtt/issues/8971

closed 2021-12-19 by github-actions[bot]