Koenkk/zigbee2mqtt #9523: HSB Issue

Wikibear, 2021-11-05:
### Discussed in https://github.com/Koenkk/zigbee2mqtt/discussions/9205

<div type='discussions-op-text'>

<sup>Originally posted by **Wikibear** October 18, 2021</sup>
Using 1.21.2 and have an issue with HSB brightness will be ignored. I founded an issue at Apr and seems to be fixed?!
[image]
This bulb is in use:
https://www.zigbee2mqtt.io/devices/404000_404005_404012.html
As you can see in picture brightness will not set half of illumination and stay on maximum.</div>

Wikibear, 2021-12-06:
Push

ceysa75, 2022-01-04:
Good to find someone having also this issue. I had already started a thread in the discussion section but no responses so far.

Have tests both: HSB and HSV notation. Brightness cannot be controlled via the third value B (or V).
E.G. {"color":{"hsv":"360,100,100"}} is identical to {"color":{"hsv":"360,100,50"}}. I am pretty sure, this was not an issue with older versions. I am using the following LED strip:

[https://www.zigbee2mqtt.io/devices/FL_120_C.html](url)

ceysa75, 2022-02-05:
This is still open. Any fix to be expected ??

joni1993, 2022-03-14:
Still a problem in 1.24.0 (7a2ddf24) 

MQTT excerpt:

> {"hsb":"349,73,74"}
{"brightness":0,"color":{"hue":349,"saturation":73,"x":0.4612,"y":0.2891},"color_mode":"hs","linkquality":3,"state":"OFF"}
{"hsb":"349,73,21"}
{"brightness":0,"color":{"hue":349,"saturation":73,"x":0.4612,"y":0.2891},"color_mode":"hs","linkquality":7,"state":"OFF"}
{"hsb":"349,73,3"}
{"brightness":0,"color":{"hue":349,"saturation":73,"x":0.4612,"y":0.2891},"color_mode":"hs","linkquality":7,"state":"OFF"}

Brightness is not set at all - Interestingly the brightness reports 0 despite the light being on. Brightness seems to be wrongly published after a setting it via "hsb".

Quautiputzli, 2022-03-22:
Hello, I have here the same problem. Will there be a solution?
Or have I go to older version? What is the last version it is working?

closed 2022-04-29 by github-actions[bot]

elvito, 2023-03-07:
Push... Brightness is not working via hsv