Koenkk/zigbee2mqtt #9527: Philips Hue white ambiance E27 800lm (8719514328242) not supported

Eagleman7, 2021-11-06:
I bought the Philips Hue White Ambiance E27 800 lumen Bluetooth yesterday, however when pairing it with zigbee2mqtt it shows as non supported. It seems the 1100lm version is supported:
[image]

I would like support for the regular 800lm version.

sjorge, 2021-11-06:
Can you post the database entry, should be relatively easy to add it.

Eagleman7, 2021-11-06:
Not sure exactly what you mean, maybe this helps:

[image]
Friendly name
0x001788010b637b8d

Eagleman7, 2021-11-07:
Thanks for the PR @sjorge

Eagleman7, 2021-11-13:
@sjorge now that is addded do I need to do anything in zigbee2mqtt to get it to a supported state?

sjorge, 2021-11-13:
Wait for the next release or switch to the dev release

closed 2021-11-13 by Eagleman7