Koenkk/zigbee2mqtt #9529: OTA: No updates available for Ubisys devices

theFork, 2021-11-06:
For each of multiple Ubisys device types (C4, S2, S1 and J1) I have a variety of different firmwares in my home.
Z2M claims that there is no update available.

## What happened

I describe the problem by taking the example of  two of my C4 devices:

| IEEE Address | Friedly Name|FW build date|
|-|-|-|
| 0x001fee0000002a44 | c4_eg_wz | 20180921-DE-FB0 |
| 0x001fee00000039c6 | c4_eg_flur | 20190211-DE-FB0 |

The build dates are taken from the GUI, in which the FW version column is empty.

When I press the "Check OTA" button of "c4_eg_wz" in the UI, the log (debug + herdsman-debug) looks as follows:
```
debug 2021-11-06 13:53:19: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"c4_eg_wz","transaction":"nmpu3-1"}'
info  2021-11-06 13:53:19: Checking if update available for 'c4_eg_wz'
info  2021-11-06 13:53:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'c4_eg_wz'","meta":{"device":"c4_eg_wz","status":"checking_if_available"},"type":"ota_update"}'
debug 2021-11-06 13:53:19: Check if update available for '0x001fee0000002a44' (C4 (5504))
debug 2021-11-06 13:53:19: Using endpoint '232'
debug 2021-11-06 13:53:25: Check if update available for '0x001fee000000578a' (S2 (5502))
debug 2021-11-06 13:53:26: OTA ubisys: got firmware page, status: 200, data.length: 57654
debug 2021-11-06 13:53:26: OTA ubisys: image found: /wp-content/uploads/10F2-7B29-0000-0004-01930221-spo-fmi4.ota.zigbee
debug 2021-11-06 13:53:26: OTA ubisys: image found: /wp-content/uploads/10F2-7B21-0000-0006-0193020E-spo-fmd.ota.zigbee
debug 2021-11-06 13:53:26: OTA ubisys: image found: /wp-content/uploads/10F2-7B28-0000-0004-0194020E-spo-rmd.ota.zigbee
debug 2021-11-06 13:53:26: OTA ubisys: image found: /wp-content/uploads/10F2-7B09-0000-0004-01090206-spo-fmi4.ota.zigbee
debug 2021-11-06 13:53:26: OTA ubisys: image found: /wp-content/uploads/10F2-7B01-0000-0006-01110206-spo-fmd.ota.zigbee
debug 2021-11-06 13:53:26: OTA ubisys: image found: /wp-content/uploads/10F2-7B08-0000-0004-01110206-spo-rmd.ota.zigbee
debug 2021-11-06 13:53:26: OTA ubisys: image found: /wp-content/uploads/10F2-7B04-0000-0007-01160206-spo-fmsh.ota.zigbee
debug 2021-11-06 13:53:26: OTA ubisys: image found: /wp-content/uploads/10F2-7B07-0000-0004-01160206-spo-rmsh.ota.zigbee
debug 2021-11-06 13:53:26: OTA ubisys: image found: /wp-content/uploads/10F2-7B0A-0000-0005-010A0206-m7b-r0.ota.zigbee
debug 2021-11-06 13:53:26: OTA ubisys: image found: /wp-content/uploads/10F2-7B02-0000-0001-010F0206-spo-fms.ota.zigbee
debug 2021-11-06 13:53:26: OTA ubisys: image found: /wp-content/uploads/10F2-7B02-0002-0007-010F0206-spo-fms.ota.zigbee
debug 2021-11-06 13:53:26: OTA ubisys: image found: /wp-content/uploads/10F2-7B05-0000-0004-010F0206-spo-rms.ota.zigbee
debug 2021-11-06 13:53:26: OTA ubisys: image found: /wp-content/uploads/10F2-7B03-0000-0006-010E0206-spo-fms2.ota.zigbee
debug 2021-11-06 13:53:26: Is new image available for '0x001fee000000578a', current '{"fieldControl":1,"manufacturerCode":4338,"imageType":31491,"fileVersion":17695238}', latest meta '{"hardwareVersionMin":0,"hardwareVersionMax":6,"fileVersion":17695238,"url":"https://www.ubisys.de/wp-content/uploads/10F2-7B03-0000-0006-010E0206-spo-fms2.ota.zigbee"}'
debug 2021-11-06 13:53:26: Update available for '0x001fee000000578a': NO
info  2021-11-06 13:53:28: No update available for 'c4_eg_wz'
info  2021-11-06 13:53:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"No update available for 'c4_eg_wz'","meta":{"device":"c4_eg_wz","status":"not_available"},"type":"ota_update"}'
info  2021-11-06 13:53:28: MQTT publish: topic 'zigbee2mqtt/c4_eg_wz', payload '{"linkquality":54,"update":{"state":"idle"},"update_available":false}'
info  2021-11-06 13:53:28: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"c4_eg_wz","updateAvailable":false},"status":"ok","transaction":"nmpu3-1"}'
```

## What did you expect to happen
I expected, that the C4 with the older FW gets the same or a later firmware as the other one.

## How to reproduce it
In my home this can be reproduced with all kinds of Ubisys devices. Most likely, all Ubisys devices are affected.

## Debug info
Zigbee2MQTT version: 1.18.1
Zigbee-herdsman version: 0.13.71
Adapter hardware: CC2652R
Adapter firmware version: 1.2.7

sjorge, 2021-11-06:
[image]

You need to check the firmware version,  1.9.x is the newest that brought the big update in base zstack version.

All the S1's were ordered directly from Ubisys during 2020-2021 usually in sets of 2. The all have different build date's, these don't seem to change on firmware update so I'm guessing they are stored in the section that does not get overwritten on update and is closer to a  manufacturing date of the device than the date the firmware was build.

theFork, 2021-11-06:
The weird thing is, I only see the build dates in the GUI. The version column is empty.

sjorge, 2021-11-06:
Does it show on the about page for the device?

theFork, 2021-11-07:
No, the version is missing here also

felixstorm, 2021-11-07:
@theFork Are you 100% sure that you are on a current version of zigbee2mqtt? Your logs above only show firmware updates that end in .ota.zigbee and none that end in .ota1.zigbee, but the code had been adapted in april this year already to also find ota1 images: https://github.com/Koenkk/zigbee-herdsman-converters/commit/05f1646cbd55b22c2c05ead5f48558cb74001316

I cannot say that I know for sure that the changed code works as intended, but the new regex looks good to me on first sight to also find .ota1.zigbee images and there are some of them available on the ubisys firmware download page that should have been found and shown in the log…

theFork, 2021-11-07:
Good point, it appears as if not. My latest log starts with:
```
➜  ~ head -n 5 /share/zigbee2mqtt/log/2021-11-06.13-47-03/log.txt
debug 2021-11-06 13:47:04: Loaded state from file /share/zigbee2mqtt/state.json
info  2021-11-06 13:47:04: Logging to console and directory: '/share/zigbee2mqtt/log/2021-11-06.13-47-03' filename: log.txt
debug 2021-11-06 13:47:04: Removing old log directory '/share/zigbee2mqtt/log/2021-09-24.10-59-38'
info  2021-11-06 13:47:04: Starting Zigbee2MQTT version 1.18.1 (commit #1.18.1)
info  2021-11-06 13:47:04: Starting zigbee-herdsman (0.13.71)
```
I installed and update Z2M using HomeAssistant. The current version of the addon is `1.18.1-1`. The latest should be `v1.22.0-2`, for some reason the HA Addon-Manager ("supervisor") did/does not offer me updates anymore. I will look into it and give feedback.

-- paused --

felixstorm, 2021-11-07:
You might still use the old version of the HA addin which does not get updates anymore. There is another addin available now which AFAIR is maintained directly by Koen.  
Sorry for not providing more details, I am on my mobile at the moment…

theFork, 2021-11-07:
Again, a good point, thanks. With the new repo I can install the lastest version. I will now figure out a safe way to migrate and switch over to `v1.22.0-2` this evening when I have the peace to do so ;)

rindlerblabla, 2021-11-07:
I am on 1.22-dev-something and I was able to update to 1.9.x just fine.

theFork, 2021-11-08:
Same here, I only had to update to 1.22 :facepalm: 
Thanks for your help.

closed 2021-11-08 by theFork