Koenkk/zigbee2mqtt #9530: Moes BRT-100-TRV - programming schedule : how does it works ?

NebzHB, 2021-11-06:
Hi,

i don't understand how to use the programing part of that valve : 
https://www.zigbee2mqtt.io/devices/BRT-100-TRV.html#programming-mode-composite

it seems the doc have been generated wrongly...

here is what the state looks like : 
[image]

i've tried to provide a schedule object (as @vladi1234 suggested in the device creation issue #7674) in the set topic (instead of program_saturday or the others) but in z2m interface it's said he doesn't find any converter. (and he doesn't find program_saturday or any others). obvi undefined doesn't work either...

how should i use it ?

vladi1234, 2021-11-11:
Hi,
you cannot use this to set the schedule yet, but only for information on how the schedule is set.
It's not completely finished yet, but I'm working on it.

ialtuhov, 2021-11-16:
Hi
I'm trying to connect a Moes BRT-100-TRV device, but I get a response - Not supported.
I use Supervisor HA.
In the logs: A message was received from an unsupported device with the Zigbee model "TS 0601" and the manufacturer's name "_tze200_b6wax7g0".

In the list of supported devices, I see that it should be supported.
I'm asking for help, I don't know what to do.

vladi1234, 2021-11-17:
Hello, I did new PR for schedule today.

With BRT-100-TRV programming mode can now be set directly via fronted.

[image]

NebzHB, 2021-11-18:
great ! what's the topic ?

vladi1234, 2021-11-18:
???

NebzHB, 2021-11-18:
the mqtt topic to publish on ? is it programming_mode ? schedule ? (see my first post)

vladi1234, 2021-11-18:
>Hi,
you cannot use this to set the schedule yet, but only for information on how the schedule is set.
It's not completely finished yet, but I'm working on it.

@NebzHB I have now changed everything, it looks completely different and it works completely differently.

What do you want with (see my first post)?

If you want to know anything, just ask, but be specific!

NebzHB, 2021-11-18:
my specific question is : what do I have to publish in mqtt in order to change the schedule ?

should i publish :
htis object :
{"programming_mode": "06:00/23°C  11:30/21°C  13:30/22°C  17:30/23°C  06:00/24°C  12:00/23°C  14:30/22°C  17:30/21°C  06:00/19°C  12:30/20°C  14:30/21°C  18:30/20°C"}
in the topic zigbee2mqtt/FRIENDLY_NAME/set ?

how to set saturday/sunday/weekday ?

as there is an holiday preset mode, is there an holiday programming mode ? or an holiday temperature (like the eco one) ?

vladi1234, 2021-11-18:
>my specific question is : what do I have to publish in mqtt in order to change the schedule ?

Answer to first question.

You don't have to publish anything in MQTT to be able to change schedules. Such a function was not planned.

> With BRT-100-TRV programming mode can now be set directly via fronted.

The answer to the second question is automatically omitted.

NebzHB, 2021-11-18:
ok, only from the frontend then, not in mqtt.

NebzHB, 2021-11-18:
my other question are still valid in the frontend, how to separate the weekdays/saturday/sunday ? does max 4 stages/periods means 4 stages/periods mandatory ? if not how to do 3 periods, with a 4th 00:00/00°C ? if not how ?

vladi1234, 2021-11-18:
> my other question is still valid in the frontend, how to separate the weekdays/saturday/sunday ?

I have already tried to display the days separately, but it does not work, it does not work with conversion.
I've already tried the separator, it doesn't work either.
Fronted mask optically, is unfortunately not yet so flexible to show separation or periods.
You have to use it as it is, all numbers in a row.

vladi1234, 2021-11-18:
If you only need two or three periods, then the next period must be the same as the last.

`06:00/23°C  11:30/21°C  19:30/22°C  19:30/22°C`

NebzHB, 2021-11-18:
ok thx

closed 2021-11-18 by NebzHB

vladi1234, 2021-11-18:
The important thing is that it works. :)
thx

vladi1234, 2021-11-18:
Oh by the way @NebzHB  , I just tested it.
It also works great with JSON!

Example:

topic zigbee2mqtt/FRIENDLY_NAME/set

```
{
  "programming_mode": "19:00/23°C  11:30/21°C  13:30/22°C  17:30/23°C  06:00/24°C  12:00/23°C  14:30/22°C  17:30/21°C  06:00/19°C  12:30/20°C  14:30/21°C  18:30/20°C"
}
```

NebzHB, 2021-11-18:
great ! even more perfect that way !

tom-ch1, 2022-09-25:
The payload format is not documented on https://www.zigbee2mqtt.io/devices/BRT-100-TRV.html

BeFygo, 2022-12-30:
> Hi, I am new to zigbee2mqtt and want to configure a weekly schedule for my thermostat TS0601. When I use the above example for 2 or 3 periods and paste this into the workdays_schedule field I allway will get an error message like "Publish set holidays_schedule ... Type Error: can not read properties of undefined. Any Idea how to fix this?

Hey erwano1957,

Publishing the following in MQTT works for me:

topic:
`zigbee2mqtt/YOUR DEVICE NAME/set`

Payload:
`{"schedule": {"week_day": "friday","schedule":"05:00/20 06:00/21 22:00/20 24:00/20"}}`

erwano1957, 2023-01-03:
Hey,

Thx a lot for your reply ... I will give it a try later on.