Koenkk/zigbee2mqtt #9531: Energy consumption for AQARA SSM-U01 lumi.switch.n0agl1 does not work correctly

lglw, 2021-11-06:
Hi.
Energy consumption and temperature statistics for **AQARA SSM-U01 lumi.switch.n0agl1** does not work correctly.
I waited, rebooted, added, removed this relay. Other functions of this relay work correctly. 
```
{
    "linkquality": 159,
    "power_outage_memory": true,
    "state": "ON",
    "switch_type": "momentary",
    "device_temperature": null,
    "energy": null,
    "power": null
}
```

## Debug info
Zigbee2MQTT version:1.22.0 
Adapter hardware: Sonoff CC2652P + CP2102N
Adapter firmware version: zStack3x0 20210120

rkuralev, 2021-11-15:
Same for me. Consumption and power are always 0 or null:
```
{
    "device_temperature": 31,
    "energy": 0,
    "linkquality": 90,
    "power": 0,
    "power_outage_memory": false,
    "state": "OFF",
    "switch_type": "toggle"
}
```

Zigbee2Mqtt 1.22.0-2
Adapter: ZZH CC2652R, firmware Z-Stack 3.x.0 20210708

z-x-f, 2021-11-16:
@rkuralev Please try to switch the dev branch, I made some changes to this device in the dev branch, and it works well for me.

Koenkk, 2021-11-16:
Could you provide the debug log when pressing the yellow reconfigure button on the device page in the frontend?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

z-x-f, 2021-11-16:
Maybe you can try this configuration:
```js
    {
        zigbeeModel: ['lumi.switch.n0agl1'],
        model: 'SSM-U01',
        vendor: 'Xiaomi',
        description: 'Aqara single switch module T1 (with neutral)',
        fromZigbee: [fz.on_off, fz.xiaomi_power, fz.xiaomi_switch_type, fz.xiaomi_switch_opple_basic],
        toZigbee: [tz.on_off, tz.xiaomi_power, tz.xiaomi_switch_type, tz.xiaomi_switch_power_outage_memory,],
        exposes: [e.switch(), e.power().withAccess(ea.STATE_GET), e.energy(), e.temperature().withAccess(ea.STATE),
            e.voltage().withAccess(ea.STATE), e.power_outage_memory(), e.switch_type()],
        configure: async (device, coordinatorEndpoint, logger) => {
            await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
        },
    },
```

rkuralev, 2021-11-16:
@Koenkk here you go
`Zigbee2MQTT:debug 2021-11-16 13:51:13: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"shower_ventilator_switch","transaction":"of8hs-1"}'
Zigbee2MQTT:info  2021-11-16 13:51:13: Configuring 'shower_ventilator_switch'
Zigbee2MQTT:debug 2021-11-16 13:51:14: Received Zigbee message from 'shower_ventilator_switch', type 'readResponse', cluster 'haElectricalMeasurement', data '{"acPowerDivisor":10,"acPowerMultiplier":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-11-16 13:51:14: Received Zigbee message from 'shower_ventilator_switch', type 'readResponse', cluster 'seMetering', data '{"divisor":1000,"multiplier":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-11-16 13:51:15: Successfully configured 'shower_ventilator_switch'
Zigbee2MQTT:info  2021-11-16 13:51:15: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"shower_ventilator_switch"},"status":"ok","transaction":"of8hs-1"}'`

@z-x-f thank you for you reply. Would you mind to point me out how to apply this configuration? I have not found anything like JSON config in the frontend...

z-x-f, 2021-11-16:
@rkuralev You can modify <code>/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/xiaomi.js</code>, save and restart z2m, and then re-pair the device.

rkuralev, 2021-11-16:
@z-x-f I don't have this directory

[image]

Koenkk, 2021-11-16:
@z-x-f thanks for the suggestions

Please try with the following converter (and give the measurements some time to come in)

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.switch.n0agl1'],
    model: 'SSM-U01',
    vendor: 'Xiaomi',
    description: 'Aqara single switch module T1 (with neutral)',
    fromZigbee: [fz.on_off, fz.xiaomi_power, fz.device_temperature, fz.xiaomi_switch_type, fz.xiaomi_switch_power_outage_memory, fz.xiaomi_switch_opple_basic],
    exposes: [e.switch(), e.energy(), e.power(), e.device_temperature(), e.power_outage_memory(), e.switch_type(), e.voltage(), e.current()],
    toZigbee: [tz.xiaomi_switch_type, tz.on_off, tz.xiaomi_switch_power_outage_memory, tz.xiaomi_power],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genDeviceTempCfg']);
        await reporting.onOff(endpoint);
        await reporting.deviceTemperature(endpoint);
        device.powerSource = 'Mains (single phase)';
        device.save();
    },
};

module.exports = definition;

```

How to use the external converter: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#instructions

rkuralev, 2021-11-17:
@Koenkk unfortunately does not help. Still see zero-s in power and consumption. Also get errors when trying to do actions, I collected several of them:
```
Failed to configure 'shower_ventilator_switch', attempt 1 (Error: Bind 0x54ef44100009b57b/1 genOnOff from '0x00124b0021600424/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7))

Failed to configure 'shower_ventilator_switch', attempt 2 (Error: ConfigureReporting 0x54ef44100009b57b/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 20831 - 1 - 8 - 6 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7))

Publish 'get' 'state' to 'shower_ventilator_switch' failed: 'Error: Read 0x54ef44100009b57b/1 genOnOff(["onOff"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'

Publish 'get' 'power_outage_memory' to 'shower_ventilator_switch' failed: 'Error: Read 0x54ef44100009b57b/1 aqaraOpple([513], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'

Publish 'get' 'power_outage_memory' to 'shower_ventilator_switch' failed: 'Error: Read 0x54ef44100009b57b/1 aqaraOpple([513], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'

Request 'zigbee2mqtt/bridge/request/device/remove' failed with error: 'Failed to remove device 'shower_ventilator_switch' (block: false, force: false) (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)'
```

Koenkk, 2021-11-17:
@rkuralev looks like communication with the device fails, try to repair it.

closed 2021-12-25 by github-actions[bot]

heikkis, 2023-01-25:
@rkuralev Did you manage to get this working? I have same device and same problem, only temperature is published.

[image]

[image]

heikkis, 2023-01-27:
After updating ZZH! zigbee stick firmware and re-join the device, it worked at least for the power and voltage.