Koenkk/zigbee2mqtt #9541: inconsistent OTA reporting

michalk-k, 2021-11-07:
## What happened
Here are two the same devices. Both reporting the same version (see screenshot below). One of them however requests for OTA update while second one doesn't. Checking updates for the second one results with:
```
Debug Update available for '0xec1bbdfffea28cea': NO
Info No update available for 'ikea_symfonisk2'
```

[image]

Note ikea_symfonisk1 starts downloading the update, but after some progress fails on continue.

## What did you expect to happen
OTA proposals are consistent across the same devices with the same FW installed.

## How to reproduce it (minimal and precise)
N/A

## Debug info
Zigbee2MQTT version: 2.22.0-2
Adapter hardware: Conbee II
Adapter firmware version: 2.12.6

francisp2, 2021-11-07:
> ikea_symfonisk1 starts downloading the update, but after some progress fails on continue.

Put new batteries, and put it very close to your coordinator.

michalk-k, 2021-11-07:
thanks for the idea. I'll try this way if current attempt fail. After about 2 hours there is 90% done :) Fingers crossed

Anyway it doesn't answer main question.

ipeacocks, 2021-11-10:
@michalk-k have you succeed?

michalk-k, 2021-11-12:
Yes, update has been finished with success
Version number nor release date haven't changed.

[image]

So doubts about inconsistency of displayed information remains.

groenmarsmannetje, 2021-11-23:
I also tried to force remove and then pair the 5-button remote again. 
And also after that the firmware information is not updated.

Bascht74, 2021-11-30:
I have the same problem with new HUE dimmers (V2). I paired one and after that I paired it to the HUE bridge and did an firmware update. After repairing to Z2M it shows the old firmware - even after repairing. 
Two other dimmers that i updated before connecting them to Z2M are OK - their firmware is shown right:
[image]

Why won't it update the firmware shown in the OTA tab? Is there a "persistent cache" preventing updates?

Same with some IKEA switches and an HUE lamp: For the lamp in HUE APP the firmware version is OK, but in Z2M it is old (I updated that device via Z2M).

nielsrune, 2021-12-02:
Yesterday, I upgraded my three E1743s. Upgrade was successful and no upgrade is available for them on the OTA tab.

However, two of them are still reporting the old firmware and build date 2.2.010 20190723, while one is reporting the new firmware and build date 2.3.079 20210722.

FWIW, for one of those reporting the incorrect data, I noticed a log entry with a payload something like this (from memory):
`{"from": {builddate: "20190723", version: "2.2.010}, "to": "null"}`
As said it's from memory, the point being that the new data was never received

Bascht74, 2021-12-05:
Maybe I have the same problem. I posted an issue as well, maybe I can close it?
Can somebody tell me if these observations are the same?

It would be really nice to have the correct firmware version viewed...

Seb

ryan-sondgeroth, 2021-12-09:
I posted this over in @Bascht74's issue but figured I would share it here as well since they seem to be related......

I have the same issue as well on a number of my Hue bulbs.    If you read the swBuildId attribute value from the genBasic cluster for your Hue bulb using the Dev Console, it will report the correct firmware value.    The issue seems to stem from the fact that the firmware info is listed in two different places in the /opt/zigbee2mqtt/data/database.db file for each bulb, and when the firmware data is updated or inserted when adding the bulb, it only gets updated in the first of those two locations.  The UI apparently uses the second of these two locations for populating the values in the UI.   

I stopped Zigbee2mqtt, and edited the lines with the incorrect firmware manually, and that resulted in the correct firmware being displayed in the UI when I restarted it.   

While it doesn't really fix the larger issue behind this phenomenon, it may help point someone much smarter than me in the correct direction to do so.

SHxKM, 2021-12-09:
I'm also facing this with some of my Hue lights.

nielsrune, 2021-12-12:
@ryan-sondgeroth I have looked into my database.db
All my Ikea E1743's which has been updated are reporting the same value twice for build date and version. As they are all updated and checks for update "fails", for the ones where z2m are reporting incorrect values, it is impossible for me to know the actual version running on the controls.

I also have two identical Hue bulbs which have been update after connecting them to my zigbee network a year ago.

They are the Bluetooth kind, so in the app I can see that they are both running firmware version 1.90.1.
However, in z2m one of them are still reporting the old version 1.76.10, both in the frontend and in the database.db file

nielsrune, 2021-12-17:
An update:
I have just updated an Ikea bulb (TRADFRI bulb E27 WW 806lm Model: LED1836G9).

The update was succesful but only the firmware version was updated on the device page, not the firmware build date.

This time, I managed to grab the relevant log from z2m from the last stage of the update procedure:
`Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"id":"Skrivebord","transaction":"zbief-1"},"meta":{"device":"Skrivebord","from":{"dateCode":"20181203","softwareBuildID":"2.3.050"},"status":"update_succeeded","to":{"dateCode":"20181203","softwareBuildID":"2.3.086"}},"type":"ota_update"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20181203","software_build_id":"2.3.050"},"id":"Skrivebord","to":{"date_code":"20181203","software_build_id":"2.3.086"}},"status":"ok","transaction":"zbief-1"}'`

The update was from version 2.3.050 to version 2.3.086.
The new build date should be something starting with 202112xx, as it matches the 1.19 release note from IKEA

michalk-k, 2022-01-17:
I'm not aware about any fixes around this. 
Don't understand the approach to close issues not touched by developer. Those are not disappearing by closing the issue.

groenmarsmannetje, 2022-01-17:
In latest version I do see an improvement. The last OTA updates that I run a couple of days ago updated the version number after OTA update was completed. Not sure if there was a fix yet to update the version numbers for the devices that are now reported with wrong version because of earlier OTA updates.

Koenkk, 2022-01-17:
@groenmarsmannetje i've pushed a fix to the latest-dev yesterday (https://github.com/Koenkk/zigbee2mqtt/issues/10666)

closed 2022-01-25 by github-actions[bot]

sbraz, 2025-05-24:
Hi, I ran into the same kind of problem with the recent v2.3.0 version of Z2M and Sonoff SNZB-02D sensors. Out of 5 devices, only 1 was shown as successfully updated, the rest were actually updated but the Web UI still showed them as running the older firmware. If that is of any interest, the coordinator is and EmberZNet-based Sonoff ZBDongle-E with firmware 8.0.2.

I followed the suggestion from #12997 to perform a read on the `swBuildId` and `dateCode` attributes, then restart Z2M. It did fix the firmware version displayed in the Web UI but the OTA tab still showed some sensors as requiring an update. I compared lines from `database.db` for the one sensor which went through the update process successfully to the others. There was quite a lot of difference in the JSON. I used that kind of script to compare them:
```sh
correct_address=0xsomething
for i in $(grep SNZB-02D database.db | grep -oP 'ieeeAddr":"\S+'); do
    diff -U0 <(grep -F "$correct_address" database.db | jq -S ) <(grep -F "$i" database.db | jq -S)
done
```

**I'm sharing what I did to fix the display issue. It's likely not be the best way to fix the problem but it worked:**

* Stop Z2M
* Delete the lines matching the inconsistent devices from `database.db`
* Restart Z2M
* Permit join and pair the devices again

Maybe I could have interviewed the devices again and it would have fixed the database but I didn't want to spend more time on this problem.

Before doing that, I also attempted to remove the devices from the Web UI (without "force remove" or "block from joining again"), however that did not delete the line from `database.db` or fix the issue.

Now, after the cleanup, if I check the contents of `database.db` by running the aforementioned script again, all devices return similar data.