Koenkk/zigbee2mqtt #9543: Expose local_temperature_calibration to Homeassistant from TS0601_thermostat

boyi01, 2021-11-07:
Hey,

i would like to use a script to sync the local_temperature_calibration to an external temperature sensor in Home assistant.

local_temperature_calibration is avaible via MQTT but it's not exposend as value in Homeassistant.

I would like to do this myself... but i have no idea in which file i need to start...

So mybe somebody can give me some advice with this or just add this entity

THX
This project is awesome!

craiq, 2021-11-09:
Hey,

if you want to use it in a script, you could also set the calibration via a service (mqtt publish)
Here are more details about: https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html#local-temperature
FRIENDLY_NAME = name in zigbee2mqtt

boyi01, 2021-11-09:
Yeah i set it via script, but for calculation i also need to know the value...

yeah i can set it to 0 and calculate then... but that's not nice... cause i double up talking to the thermostat..

craiq, 2021-11-09:
For calculation you can access the value via the state_attr template https://www.home-assistant.io/docs/configuration/templating/

boyi01, 2021-11-10:
I tried it... but then it's often an old value... ;(
so i set it from 0 to -2,5 and the in next itteration it calculate with 0

craiq, 2021-11-10:
To understand the problem I would need to know more in detail what you've done. You might need a delay, when you trigger on a state change until all states are up to date

Drafteed, 2021-11-10:
I suggested this in #9269

Drafteed, 2021-11-10:
Will add in #9595

Koenkk, 2021-11-10:
Fixed by https://github.com/Koenkk/zigbee2mqtt/pull/9595

PR #9595 (2021-11-10): Expose local_temperature_calibration in HA as numeric
Hello! 

This PR will enable discovery a climate `local_temperature_calibration` in Home Assistant as `Number` entity.

This will close: #9543 and #9269
closed 2021-11-10 by Koenkk