Koenkk/zigbee2mqtt #9544: HEIMAN HS2WD-E : Triggering impossible

Pauliakov, 2021-11-07:
Hello to all,

First of all, I wanted to thank you for this great tool that you have made available and that works perfectly for all my other devices.

I have a problem with the Heiman HS2WD-E alarm that I just bought. 
My zigbee2MQTT is running on Jeedom, not Home Assistant, but work perfectly on all my other devices.
My sniffer detects the object well but I only have the information below :

[image]

And I can't publish to get the other information.
When I want to publish the order below:
`{"warning": {"duration": 10, "mode": "emergency", "strobe": false}}`

I have an error feedback in the logs that I can't resolve INVALID JSON

Below are the pairing logs
`info  2021-11-07 13:44:10: Device '0x842e14fffexxxxxx' joined
info  2021-11-07 13:44:10: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842e14fffexxxxxx","ieee_address":"0x842e14fffexxxxxx"},"type":"device_joined"}'
info  2021-11-07 13:44:10: Starting interview of '0x842e14fffexxxxxx'
info  2021-11-07 13:44:10: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842e14fffexxxxxx","ieee_address":"0x842e14fffexxxxxx","status":"started"},"type":"device_interview"}'
info  2021-11-07 13:44:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x842e14fffexxxxxx"},"type":"device_connected"}'
info  2021-11-07 13:44:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x842e14fffexxxxxx"},"type":"pairing"}'
info  2021-11-07 13:44:11: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842e14fffexxxxxx","ieee_address":"0x842e14fffexxxxxx"},"type":"device_announce"}'
info  2021-11-07 13:44:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x842e14fffexxxxxx"},"type":"device_announced"}'
info  2021-11-07 13:44:14: Successfully interviewed '0x842e14fffexxxxxx', device has successfully been paired
info  2021-11-07 13:44:14: Device '0x842e14fffexxxxxx' is supported, identified as: HEIMAN Smart siren (HS2WD-E)
info  2021-11-07 13:44:14: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart siren","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"features":[{"access":2,"description":"Mode of the warning (sound effect)","name":"mode","property":"mode","type":"enum","values":["stop","burglar","fire","emergency","police_panic","fire_panic","emergency_panic"]},{"access":2,"description":"Sound level","name":"level","property":"level","type":"enum","values":["low","medium","high","very_high"]},{"access":2,"description":"Turn on/off the strobe (light) during warning","name":"strobe","property":"strobe","type":"binary","value_off":false,"value_on":true},{"access":2,"description":"Duration in seconds of the alarm","name":"duration","property":"duration","type":"numeric","unit":"s"}],"name":"warning","property":"warning","type":"composite"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"HS2WD-E","supports_ota":false,"vendor":"HEIMAN"},"friendly_name":"0x842e14fffexxxxxx","ieee_address":"0x842e14fffexxxxxx","status":"successful","supported":true},"type":"device_interview"}'
info  2021-11-07 13:44:14: Configuring '0x842e14fffexxxxxx'
info  2021-11-07 13:44:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Smart siren","friendly_name":"0x842e14fffexxxxxx","model":"HS2WD-E","supported":true,"vendor":"HEIMAN"},"type":"pairing"}'
info  2021-11-07 13:44:15: MQTT publish: topic 'zigbee2mqtt/0x842e14fffexxxxxx', payload '{"battery":25.5,"linkquality":39}'
info  2021-11-07 13:44:15: Successfully configured '0x842e14fffexxxxxx'`

Thank you very much for your help

Bartekn86, 2021-11-07:
similar to https://github.com/Koenkk/zigbee2mqtt/issues/9380

Bartekn86, 2021-11-25:
hi, could you look @ problem please ?

Bartekn86, 2021-12-25:
Hi,  could sb look at the problem please ?

closed 2022-02-02 by github-actions[bot]