Koenkk/zigbee2mqtt #9545: Smart Light Strip Controller RGBW Model: YSR-MINI-01

Martin004, 2021-11-07:
I am very confused. I bought a Smart Light Strip Controller with the model number: YSR-MINI-01 Edition: RGBW.
I assumed it was a ysrsai product.

After integrating it, I was able to control the RGB LEDs.
Since it is an RGBW stripe, I would also like to control the separate white LEDs.
This is currently not possible.

The Debug Info shows as Device Info Nue / 3A Smart 7W E27 light bulb (HGZB-06A).

Do you have any idea what's going wrong here?

Greatings Martin 

## Information about the device + link
debug 2021-11-07 15:05:48: Skipping message, definition is undefined and still interviewing
info  2021-11-07 15:05:48: Successfully interviewed '0x00158d0004775299', device has successfully been paired
info  2021-11-07 15:05:48: Device '0x00158d0004775299' is supported, identified as: Nue / 3A Smart 7W E27 light bulb (HGZB-06A)
info  2021-11-07 15:05:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart 7W E27 light bulb","exposes":[{"features":[{"access":7,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0},{"access":7,"description":"Color temperature of this light","name":"color_temp","presets":[{"description":"Coolest temperature supported","name":"coolest","value":150},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500}],"property":"color_temp","type":"numeric","unit":"mired","value_max":500,"value_min":150},{"access":7,"description":"Color temperature after cold power on of this light","name":"color_temp_startup","presets":[{"description":"Coolest temperature supported","name":"coolest","value":150},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500},{"description":"Restore previous color_temp on cold power on","name":"previous","value":65535}],"property":"color_temp_startup","type":"numeric","unit":"mired","value_max":500,"value_min":150},{"description":"Color of this light in the CIE 1931 color space (x/y)","features":[{"access":7,"name":"x","property":"x","type":"numeric"},{"access":7,"name":"y","property":"y","type":"numeric"}],"name":"color_xy","property":"color","type":"composite"}],"type":"light"},{"access":2,"description":"Triggers an effect on the light (e.g. make light blink for a few seconds)","name":"effect","property":"effect","type":"enum","values":["blink","breathe","okay","channel_change","finish_effect","stop_effect"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"HGZB-06A","options":[{"access":2,"description":"Controls the transition time (in seconds) of on/off, brightness, color temperature (if applicable) and color (if applicable) changes. Defaults to `0` (no transition).","name":"transition","property":"transition","type":"numeric","value_min":0},{"access":2,"description":"When enabled colors will be synced, e.g. if the light supports both color x/y and color temperature a conversion from color x/y to color temperature will be done when setting the x/y color (default true).","name":"color_sync","property":"color_sync","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"Nue / 3A"},"friendly_name":"0x00158d0004775299","ieee_address":"0x00158d0004775299","status":"successful","supported":true},"type":"device_interview"}'
info  2021-11-07 15:05:49: Configuring '0x00158d0004775299'
info  2021-11-07 15:05:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Smart 7W E27 light bulb","friendly_name":"0x00158d0004775299","model":"HGZB-06A","supported":true,"vendor":"Nue / 3A"},"type":"pairing"}'
debug 2021-11-07 15:05:49: Received Zigbee message from '0x00158d0004775299', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorCapabilities":31}' from endpoint 11 with groupID 0
info  2021-11-07 15:05:49: MQTT publish: topic 'zigbee2mqtt/0x00158d0004775299', payload '{"color_mode":"xy","linkquality":171}'
debug 2021-11-07 15:05:49: Received Zigbee message from '0x00158d0004775299', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorTempPhysicalMax":370,"colorTempPhysicalMin":153}' from endpoint 11 with groupID 0
info  2021-11-07 15:05:49: MQTT publish: topic 'zigbee2mqtt/0x00158d0004775299', payload '{"color_mode":"xy","linkquality":168}'
info  2021-11-07 15:05:49: Successfully configured '0x00158d0004775299'

## data/database.db entry of the device
{"id":65,"type":"Router","ieeeAddr":"0x00158d0004775299","nwkAddr":22851,"manufId":4478,"manufName":"Feibit","powerSource":"Mains (single phase)","modelId":"FNB56-ZCW25FB2.1","epList":[11],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,4,3,6,8,5,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"FNB56-ZCW25FB2.1","manufacturerName":"Feibit","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20171209","swBuildId":"V1.5"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20171209","swBuildId":"V1.5","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1636293949437}

Martin004, 2021-11-14:
Does anyone have any ideas?

csszokolai, 2021-11-15:
Same issue with rgbcct controller. The controller recognized as a bulb

Martin004, 2021-11-22:
Hello I solved the problem.

Device file changed:
File : ysrsai.js
 
`const extend = require('../lib/extend');

module.exports = [
    {
        fingerprint: [{modelID: 'ZB-CL01', manufacturerName: 'YSRSAI'}],
        zigbeeModel: ['ZB-CL03', 'FB56-ZCW20FB1.2'],
        model: 'YSR-MINI-01_rgbcct',
        vendor: 'YSRSAI',
        description: 'Zigbee LED controller (RGB+CCT)',
        extend: extend.light_onoff_brightness_colortemp_color({disableColorTempStartup: true, colorTempRange: [160, 370]}),
    },
    {
		**_fingerprint: [{inClusterList: [0,4,3,6,8,5,768,4096]}],_**
		zigbeeModel: ['ZB-CT01'_**, 'FNB56-ZCW25FB2.1'**_],
        model: 'YSR-MINI-01_wwcw',
        vendor: 'YSRSAI',
        description: 'Zigbee LED controller (WW/CW)',
        extend: extend.light_onoff_brightness_colortemp_color({disableColorTempStartup: true}),
    },
    {
        zigbeeModel: ['ZB-DL01'],
        model: 'YSR-MINI-01_dimmer',
        vendor: 'YSRSAI',
        description: 'Zigbee LED controller (Dimmer)',
        extend: extend.light_onoff_brightness(),
    },
];
`

Martin004, 2021-11-25:
the info log shows now:
`info  2021-11-25 11:01:33: Led_Stripe (0x00158d00047752d8): YSR-MINI-01_wwcw - YSRSAI Zigbee LED controller (WW/CW) (Router)`

Martin004, 2021-11-29:
Koenkk please add , 'FNB56-ZCW25FB2.1'

like 
zigbeeModel: ['ZB-CT01', 'FNB56-ZCW25FB2.1'],
model: 'YSR-MINI-01_wwcw',
vendor: 'YSRSAI',
to your dev. 

it will support my device.

Thanks a lot.

Koenkk, 2021-11-30:
We cannot distinguish these device from each other (since they use the same modelID), if it is really a problem for you change the `modelId` in your data/database.db to `ZB-CT01`

Cirrus94, 2022-01-02:
Hi. 

I am using also the YSRSAI RGBW LED Controller. At first it was recognized as Lonsonho Bulb. I changed the manufName to „YSRSAI“ ind database.db as @Koenkk mentioned in another issue. That solved the manufacturer issue. Another issue is, that the controller is now recognized as RGBCCT. This results in a wrong Home Assistant UI for controlling the LED strip. I can change the color, the brightness and the color temperature of the white channel. Since i only have a warm white channel and RGB that should be for my understanding only brightness for each RGB and the Warm White Channel and the Color for the RGB channel. Is there anything what can be set to change the device to RGBW?

Here is the database.db entry for my device:
`{"id":20,"type":"Router","ieeeAddr":"0xa4c1386611884196","nwkAddr":41646,"manufId":4107,"manufName":"YSRSAI","powerSource":"Mains (single phase)","modelId":"ZB-CL01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"ZB-CL01","manufacturerName":"eWeLight","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":0,"dateCode":"20210907","swBuildId":"2.0.1"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":250,"colorTempPhysicalMax":454}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":0,"dateCode":"20210907","swBuildId":"2.0.1","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1641111024684,"useImplicitCheckin":true} `

Here are some infos from the manual:
[image]

Thanks!

closed 2022-02-09 by github-actions[bot]